newracom/nrc7292_sw_pkg #1: Error whenn loading driver

wzab, 2020-09-18:
When I try to load driver compiled from the newest version of the sources - 1.3.2, I get the following error:

```
[  662.495486] nrc: loading out-of-tree module taints kernel.
[  662.509860] ------------[ cut here ]------------
[  662.509983] WARNING: CPU: 0 PID: 1755 at /home/pi/build/nrc7292_sw_pkg/package/host/nrc_driver/source/nrc_driver/nrc/nrc-hif-cspi.c:535 _c_spi_read_regs+0x1d8/0x1f0 [nrc]
[  662.509989] Modules linked in: nrc(O+) iptable_filter iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack fuse snd_bcm2835(C) snd_pcm snd_timer snd spi_bcm2835 evdev joydev fixed uio_pdrv_genirq uio mac80211 cfg80211 rfkill i2c_dev ip_tables x_tables ipv6
[  662.510168] CPU: 0 PID: 1755 Comm: insmod Tainted: G         C O    4.14.70-v7+ #1144
[  662.510173] Hardware name: BCM2835
[  662.510204] [<8010ffd8>] (unwind_backtrace) from [<8010c240>] (show_stack+0x20/0x24)
[  662.510221] [<8010c240>] (show_stack) from [<80787b24>] (dump_stack+0xd4/0x118)
[  662.510238] [<80787b24>] (dump_stack) from [<8011dacc>] (__warn+0xf8/0x110)
[  662.510252] [<8011dacc>] (__warn) from [<8011dbb4>] (warn_slowpath_null+0x30/0x38)
[  662.510351] [<8011dbb4>] (warn_slowpath_null) from [<7f378f3c>] (_c_spi_read_regs+0x1d8/0x1f0 [nrc])
[  662.510549] [<7f378f3c>] (_c_spi_read_regs [nrc]) from [<7f378398>] (c_spi_read_regs+0x4c/0xb0 [nrc])
[  662.510732] [<7f378398>] (c_spi_read_regs [nrc]) from [<7f37a82c>] (spi_update_status+0x60/0x224 [nrc])
[  662.510922] [<7f37a82c>] (spi_update_status [nrc]) from [<7f37aadc>] (spi_start+0xec/0x148 [nrc])
[  662.511100] [<7f37aadc>] (spi_start [nrc]) from [<7f372598>] (nrc_hif_init+0xb4/0xc0 [nrc])
[  662.511274] [<7f372598>] (nrc_hif_init [nrc]) from [<7f3712f4>] (nrc_platform_probe+0x68/0x130 [nrc])
[  662.511373] [<7f3712f4>] (nrc_platform_probe [nrc]) from [<80545fd0>] (platform_drv_probe+0x5c/0xc0)
[  662.511394] [<80545fd0>] (platform_drv_probe) from [<805445b0>] (driver_probe_device+0x244/0x300)
[  662.511411] [<805445b0>] (driver_probe_device) from [<80544728>] (__driver_attach+0xbc/0xc0)
[  662.511428] [<80544728>] (__driver_attach) from [<805425cc>] (bus_for_each_dev+0x7c/0xb0)
[  662.511444] [<805425cc>] (bus_for_each_dev) from [<80543ef4>] (driver_attach+0x2c/0x30)
[  662.511458] [<80543ef4>] (driver_attach) from [<805439d0>] (bus_add_driver+0x1ac/0x224)
[  662.511472] [<805439d0>] (bus_add_driver) from [<80544e9c>] (driver_register+0x88/0x108)
[  662.511497] [<80544e9c>] (driver_register) from [<80545f20>] (__platform_driver_register+0x50/0x58)
[  662.511595] [<80545f20>] (__platform_driver_register) from [<7f396038>] (nrc_init+0x38/0x1000 [nrc])
[  662.511713] [<7f396038>] (nrc_init [nrc]) from [<80101bf0>] (do_one_initcall+0x50/0x17c)
[  662.511742] [<80101bf0>] (do_one_initcall) from [<801a97c8>] (do_init_module+0x74/0x20c)
[  662.511767] [<801a97c8>] (do_init_module) from [<801a87ec>] (load_module+0x1c28/0x23a0)
[  662.511792] [<801a87ec>] (load_module) from [<801a9140>] (SyS_finit_module+0x9c/0xac)
[  662.511821] [<801a9140>] (SyS_finit_module) from [<80108060>] (ret_fast_syscall+0x0/0x28)
[  662.511961] ---[ end trace 55326a9da99a333c ]---
[  662.532586] start FW 297
[  662.533867] ------------[ cut here ]------------
[  662.533986] WARNING: CPU: 0 PID: 29 at /home/pi/build/nrc7292_sw_pkg/package/host/nrc_driver/source/nrc_driver/nrc/nrc-hif-cspi.c:535 _c_spi_read_regs+0x1d8/0x1f0 [nrc]
[  662.534011] Modules linked in: nrc(O) iptable_filter iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack fuse snd_bcm2835(C) snd_pcm snd_timer snd spi_bcm2835 evdev joydev fixed uio_pdrv_genirq uio mac80211 cfg80211 rfkill i2c_dev ip_tables x_tables ipv6
[  662.534163] CPU: 0 PID: 29 Comm: kworker/0:1 Tainted: G        WC O    4.14.70-v7+ #1144
[  662.534168] Hardware name: BCM2835
[  662.534250] Workqueue: events nrc_check_start [nrc]
[  662.534284] [<8010ffd8>] (unwind_backtrace) from [<8010c240>] (show_stack+0x20/0x24)
[  662.534302] [<8010c240>] (show_stack) from [<80787b24>] (dump_stack+0xd4/0x118)
[  662.534318] [<80787b24>] (dump_stack) from [<8011dacc>] (__warn+0xf8/0x110)
[  662.534333] [<8011dacc>] (__warn) from [<8011dbb4>] (warn_slowpath_null+0x30/0x38)
[  662.534419] [<8011dbb4>] (warn_slowpath_null) from [<7f378f3c>] (_c_spi_read_regs+0x1d8/0x1f0 [nrc])
[  662.534589] [<7f378f3c>] (_c_spi_read_regs [nrc]) from [<7f378398>] (c_spi_read_regs+0x4c/0xb0 [nrc])
[  662.534747] [<7f378398>] (c_spi_read_regs [nrc]) from [<7f37850c>] (spi_wait_ack+0x40/0x80 [nrc])
[  662.534904] [<7f37850c>] (spi_wait_ack [nrc]) from [<7f374670>] (nrc_fw_check_next_frag+0x88/0x10c [nrc])
[  662.535061] [<7f374670>] (nrc_fw_check_next_frag [nrc]) from [<7f374984>] (nrc_download_fw+0x170/0x378 [nrc])
[  662.535217] [<7f374984>] (nrc_download_fw [nrc]) from [<7f371954>] (nrc_check_start+0x598/0x610 [nrc])
[  662.535308] [<7f371954>] (nrc_check_start [nrc]) from [<801376e8>] (process_one_work+0x158/0x454)
[  662.535328] [<801376e8>] (process_one_work) from [<80137a48>] (worker_thread+0x64/0x5b8)
[  662.535345] [<80137a48>] (worker_thread) from [<8013dacc>] (kthread+0x13c/0x16c)
[  662.535363] [<8013dacc>] (kthread) from [<8010810c>] (ret_from_fork+0x14/0x28)
[  662.535401] ---[ end trace 55326a9da99a333d ]---
[  662.535434] ------------[ cut here ]------------
[  662.535451] kernel BUG at /home/pi/build/nrc7292_sw_pkg/package/host/nrc_driver/source/nrc_driver/nrc/nrc-fw.c:193!
[  662.535463] Internal error: Oops - BUG: 0 [#1] SMP ARM
[  662.535474] Modules linked in: nrc(O) iptable_filter iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack fuse snd_bcm2835(C) snd_pcm snd_timer snd spi_bcm2835 evdev joydev fixed uio_pdrv_genirq uio mac80211 cfg80211 rfkill i2c_dev ip_tables x_tables ipv6
[  662.535597] CPU: 0 PID: 29 Comm: kworker/0:1 Tainted: G        WC O    4.14.70-v7+ #1144
[  662.535606] Hardware name: BCM2835
[  662.535691] Workqueue: events nrc_check_start [nrc]
[  662.535706] task: b6956900 task.stack: b6a3a000
[  662.535790] PC is at nrc_fw_check_next_frag+0x108/0x10c [nrc]
[  662.535874] LR is at spi_wait_ack+0x64/0x80 [nrc]
[  662.535885] pc : [<7f3746f0>]    lr : [<7f378530>]    psr: a0000013
[  662.535897] sp : b6a3be60  ip : b6a3be28  fp : b6a3be7c
[  662.535907] r10: 00000000  r9 : 00000008  r8 : 00000001
[  662.535918] r7 : 00000000  r6 : aa29d800  r5 : b5e9adc0  r4 : b4cc4c80
[  662.535930] r3 : 00000140  r2 : 00000000  r1 : 7f384eb4  r0 : fffffffb
[  662.535943] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  662.535955] Control: 10c5383d  Table: 35ce006a  DAC: 00000055
[  662.535966] Process kworker/0:1 (pid: 29, stack limit = 0xb6a3a210)
[  662.535978] Stack: (0xb6a3be60 to 0xb6a3c000)
[  662.535995] be60: b4cc4c80 00e9adc0 aa29d800 b4cc4c80 b6a3beb4 b6a3be80 7f374984 7f3745f4
[  662.536012] be80: 92720100 01000000 01020716 b5e9bdc0 b5e9bdc0 b6fc5940 b5e9adc0 00000000
[  662.536029] bea0: 00000008 00000000 b6a3befc b6a3beb8 7f371954 7f374820 8079dad4 80145bf4
[  662.536046] bec0: 80c5f99c b6fc5940 b6fc8900 80137518 b6a3befc b69fe480 b5e9bdc0 b6fc5940
[  662.536063] bee0: b6fc8900 00000000 00000008 00000000 b6a3bf34 b6a3bf00 801376e8 7f3713c8
[  662.536080] bf00: b6fc5958 80c02d00 00000008 b6fc5940 b69fe498 b6fc5940 b6fc5958 80c02d00
[  662.536097] bf20: 00000008 b69fe480 b6a3bf7c b6a3bf38 80137a48 8013759c b6a3bf5c 00000000
[  662.536114] bf40: 80c02d00 80c02d00 80c88262 b6a3a038 b6a3501c b6a35000 00000000 b69ddf00
[  662.536131] bf60: b69fe480 801379e4 b6a3501c b6909e80 b6a3bfac b6a3bf80 8013dacc 801379f0
[  662.536147] bf80: 80102d94 b69ddf00 8013d990 00000000 00000000 00000000 00000000 00000000
[  662.536163] bfa0: 00000000 b6a3bfb0 8010810c 8013d99c 00000000 00000000 00000000 00000000
[  662.536179] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  662.536195] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[  662.536359] [<7f3746f0>] (nrc_fw_check_next_frag [nrc]) from [<7f374984>] (nrc_download_fw+0x170/0x378 [nrc])
[  662.536521] [<7f374984>] (nrc_download_fw [nrc]) from [<7f371954>] (nrc_check_start+0x598/0x610 [nrc])
[  662.536613] [<7f371954>] (nrc_check_start [nrc]) from [<801376e8>] (process_one_work+0x158/0x454)
[  662.536636] [<801376e8>] (process_one_work) from [<80137a48>] (worker_thread+0x64/0x5b8)
[  662.536657] [<80137a48>] (worker_thread) from [<8013dacc>] (kthread+0x13c/0x16c)
[  662.536678] [<8013dacc>] (kthread) from [<8010810c>] (ret_from_fork+0x14/0x28)
[  662.536694] Code: e24bd010 e89da810 e7f001f2 e7f001f2 (e7f001f2) 
[  662.536709] ---[ end trace 55326a9da99a333e ]---

```

The error seems to be associated with the following line:

```
        if (priv->ack) {
                ret = nrc_hif_wait_ack(hdev, &index, 1);
                BUG_ON(ret < 0);
        }

```

wzab, 2020-09-18:
I have checked out the older version - 1.3.0 and there is the same problem.
The driver delivered in the binary form by the manufacturer of the evaluation kit - Airetos, works correctly.

newracom, 2020-09-18:
Thank you for your report.

When I tried to reproduce your situation, I failed.
Could you check below?

- Is your NRC7292 EVK attached to your Raspberry Pi platform?
  NRC7292 SW package requires NRC7292 EVK (https://www.newracom.com/product/nrc7292-evk/) which is the world-first IEEE 802.11ah (a.k.a Wi-Fi HaLow) device.

- If you have, please check the below test procedure.
1. Copy host driver: nrc7292_sw_pkg\package\host\nrc_driver\source\nrc_driver\nrc
2. Host driver build: make clean; make -j4
3. Copy host driver: cp nrc.ko ~/nrc_pkg/sw/driver
4. Copy firmware: cp nrc7292_sw_pkg/package/host/evk/binary/nrc7292_cspi.bin ~/nrc_pkg/sw/firmware
5. Run start.py script: cd ~/nrc_pkg/script; ./start.py 1 0 US

wzab, 2020-09-19:
I use the evaluation kit from Airetos. It is based on Raspberry Pi. Does it mean that your driver does not work with their board?
[image]
The device tree in their original firmware is strictly the same as in your sources.

wzab, 2020-09-19:
OK. I've managed to run your driver on the Silex [SX-NEWAH](https://www.silextechnology.com/connectivity-solutions/embedded-wireless/sx-newah-evaluation) working with Rapsberry Pi 3.

newracom, 2020-09-21:
Thank you for your update.
Host device driver has a dependency on the interface between NRC7292 module and host platform.
I'm afraid there's trivial discrepancy on the host interface configuration of the Airetos module.

closed 2020-09-21 by newracom

hnguyen63, 2022-01-07:
Could we establish a mesh ad-hoc network using the SX-NEWAH modules or any modules integrated with nrc7292?

Many thanks.

newracom, 2022-01-07:
Dear @hnguyen63,

NRC7292 supports IEEE 802.11s mesh network.

Best regards,
Newracom

hnguyen63, 2022-01-08:
Hi,

Thank you for your quick reply.
We've tried to set up a mesh network with SX-NEWAH-EVK, which integrated
with the nrc7292 but found out that the EVK currently does not support mesh
network (
https://www.silextechnology.com/unwired/sx-newah-industrys-first-802.11ah-wi-fi-module-faqs).
From your email, we believe that we can create a new linux system that can
support the mesh network topology.

Would you please give us some related information or instructions to show
how to implement the mesh network structure. We would
greatly appreciate your help and support in this matter.

Thank you very much in advance.

Kind regards,
Nghia

newracom, 2022-01-10:
Dear @hnguyen63,

Please refer to this application note, https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/AN-7292-007-11s_mesh_network.pdf.
And if you have further question, please open a new issue.

Best regards,
Newracom