newracom/nrc7292_sw_pkg #2: Why the driver accessed SPI device simultaneously in two different ways?

wzab, 2020-09-20:
I started to port the driver to 5.8.6 kernel running on RK3399 platform. Trying to compile it and getting running, I have found a strange thing. The driver makes a copy of the structure describing the underlying SPI device:
https://github.com/newracom/nrc7292_sw_pkg/blob/34a691ab5f61c52b2b33d276e24a118152869ec0/package/host/nrc_driver/source/nrc_driver/nrc/nrc-hif-cspi.c#L1662 
And then uses both the copy and the original structure to communicate with the device. For example:
https://github.com/newracom/nrc7292_sw_pkg/blob/34a691ab5f61c52b2b33d276e24a118152869ec0/package/host/nrc_driver/source/nrc_driver/nrc/nrc-hif-cspi.c#L287
and
https://github.com/newracom/nrc7292_sw_pkg/blob/34a691ab5f61c52b2b33d276e24a118152869ec0/package/host/nrc_driver/source/nrc_driver/nrc/nrc-hif-cspi.c#L293

or

https://github.com/newracom/nrc7292_sw_pkg/blob/34a691ab5f61c52b2b33d276e24a118152869ec0/package/host/nrc_driver/source/nrc_driver/nrc/nrc-hif-cspi.c#L305
and
https://github.com/newracom/nrc7292_sw_pkg/blob/34a691ab5f61c52b2b33d276e24a118152869ec0/package/host/nrc_driver/source/nrc_driver/nrc/nrc-hif-cspi.c#L311

The above is used only in the "alternated mode", but anyway it looks a little annoying.
Doesn't it lead to the inconsistent state of the SPI hardware?

newracom, 2020-09-22:
Thank you for your report.
The alternated mode is only valid for the internal test purpose. 
It's obsolete, so you can ignore that now.
We will get rid of that mode in the next patch or release. Thank you.

newracom, 2020-09-25:
[nrc7292-alternated-mode-removal.zip](https://github.com/newracom/nrc7292_sw_pkg/files/5284931/nrc7292-alternated-mode-removal.zip)
This is the patch for the alternated mode code removal and this will be applied to the next official release.
Thank you for your cooperation.

closed 2020-09-25 by newracom