newracom/nrc7292_sw_pkg #3: Porting to kernel 5.4.x or 5.8.x

wzab, 2020-09-21:
I have tried to port your driver to Orange Pi RK3399, running Armbian with kernel 5.8.6.
Unfortunately, my efforts failed and it was difficult to isolate the cause of the problem.
It can be the problem with the hardware, the problem with newer kernel, and the problem with migration from 32-bit to 64-bit platform.

Therefore I have started with porting the driver to newer kernel on Raspberry Pi3. There is Raspbian version running the 5.4.51-v7+ kernel.
I had to introduce a few changes, that can be identified [by that compare](https://github.com/newracom/nrc7292_sw_pkg/compare/master...wzab:b296d2409bd690af0f8a38161c62abc6d260d025).

I also needed to use UART for debugging (connection via SSH didn't allow to capture all crash messages), so I had to modify the dt overlay, disabling the changes done to UART (fragments 3 and 4):
```
/*
 * Device Tree overlay for Newracom
 */

/dts-v1/;
/plugin/;

/* this includes disabling SPI and pi3-disable-bt overlay
*/

/{
        compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

        fragment@0 {
                target = <&spi0>;
                __overlay__ {
                        status = "okay";

                        spidev@0{
                                status = "disabled";
                        };

                        spidev@1{
                                status = "disabled";
                        };
                };
        };

        fragment@1 {
                target = <&uart1>;
                __overlay__ {
                        status = "disabled";
                };
        };

        fragment@2 {
                target = <&uart0>;
                __overlay__ {
                        pinctrl-names = "default";
                        pinctrl-0 = <&uart0_pins>;
                        status = "okay";
                };
        };
/*
        fragment@3 {
                target = <&uart0_pins>;
                __overlay__ {
                        brcm,pins;
                        brcm,function;
                        brcm,pull;
                };
        };

        fragment@4 {
                target-path = "/aliases";
                __overlay__ {
                        serial0 = "/soc/uart@7e201000";
                        serial1 = "/soc/uart@7e215040";
                };
        };
*/
[... 256 lines cut ...]
[  166.736435] [<8010d758>] (show_stack) from [<808cdbe4>] (dump_stack+0xe0/0x124)
[  166.753149] [<808cdb04>] (dump_stack) from [<80110364>] (handle_IPI+0x334/0x3bc)
[  166.769895]  r8:80da6a80 r7:00000000 r6:00000002 r5:80ca42e4 r4:80dac928 r3:10aae77c
[  166.786966] [<80110030>] (handle_IPI) from [<80102268>] (bcm2836_arm_irqchip_handle_irq+0xa0/0xa4)
[  166.805281]  r10:00000000 r9:b653a000 r8:80d9b18d r7:b653bf6c r6:ffffffff r5:60000013
[  166.822471]  r4:00000002
[  166.834191] [<801021c8>] (bcm2836_arm_irqchip_handle_irq) from [<80101a3c>] (__irq_svc+0x5c/0x7c)
[  166.852363] Exception stack(0xb653bf38 to 0xb653bf80)
[  166.866602] bf20:                                                       80109b94 00000000
[  166.884061] bf40: 60000093 8011a520 80d04f68 b653a000 80d04fb0 00000004 80d9b18d 410fd034
[  166.901519] bf60: 00000000 b653bf94 80d0535c b653bf88 00000000 80109b98 60000013 ffffffff
[  166.918959]  r4:80109b98 r3:10aae77c
[  166.931681] [<80109b64>] (arch_cpu_idle) from [<808eea44>] (default_idle_call+0x34/0x48)
[  166.949041] [<808eea10>] (default_idle_call) from [<80153bc0>] (do_idle+0xec/0x170)
[  166.965943] [<80153ad4>] (do_idle) from [<80153f20>] (cpu_startup_entry+0x28/0x2c)
[  166.982752]  r8:0000406a r7:80dac938 r6:10c0387d r5:00000002 r4:0000008a r3:10aae77c
[  166.999743] [<80153ef8>] (cpu_startup_entry) from [<8010fdbc>] (secondary_start_kernel+0x130/0x13c)
[  167.018102] [<8010fc8c>] (secondary_start_kernel) from [<001027ac>] (0x1027ac)
[  167.034618]  r5:00000055 r4:3652806a
[  167.403550] SMP: failed to stop secondary CPUs
[  167.417120] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
```

There are two problems. The first one is - why the firmware download is not detected correctly. I have verified that hijacking UART has nothing to do with that. In Raspbian working with the old kernel the same modification of DT and config.txt does not lead to crash.
The second problem is why unsuccessful loading of the FW ends with incorrect handling of kobjects, which finally leads to memory corruption and delayed crash.

wzab, 2020-09-21:
It looks like the procedure nrc_register_hw needs modification for the newer kernel...

wzab, 2020-09-23:
To enable testing of the driver with different versions of the kernel, and on different hardware platforms, I have prepared [Buildroot](https://buildroot.org) packages with the kernel driver and with the cli_app application. They are located in my [experimental repository](https://github.com/wzab/nrc7292-buildroot). The first tests have shown that the driver works correctly with 32-bit 4.14.114-v7 kernel on Buildroot-compiled Linux. I'll publish further results.

wzab, 2020-09-23:
I have compiled version for 64-bit system on RPi3 with kernel 4.14.114-v7.
```
# uname -a
Linux buildroot 4.14.114-v8 #1 SMP PREEMPT Thu Sep 24 00:55:06 CEST 2020 aarch64 GNU/Linux
```
 The driver works correctly, and I was able to find the AP:
```
# iwlist wlan0 scan
wlan0     Scan completed :
          Cell 01 - Address: 84:25:3F:xx:xx:xx
                    Channel:165
                    Frequency:5.825 GHz
                    Quality=70/70  Signal level=2 dBm  
                    Encryption key:on
                    ESSID:"Wi-Fi"
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000088b2d6a3
                    Extra: Last beacon: 109ms ago
                    IE: Unknown: 000557692D4669
                    IE: Unknown: 01088C129824B048606C
                    IE: Unknown: 0301A5
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : unknown (8)
                    IE: Unknown: 2D1A0000000000000000000000000000000000000000000000000000
                    IE: Unknown: 3D16A5000000000000000000000000000000000000000000
                    IE: Unknown: 7F080000000000000000
                    IE: Unknown: DD180050F2020101000003A4000027A4000042435E0062322F00
```
However the communication between cli_app and the driver fails:
```
# cli_app txpwr 23
FAIL
```

mokajava3, 2020-09-24:
in first, could you get the version './cli_app show version' ?
If you get this correctly,  the 'set' is missed in your command.
Could you run './cli_app set txpwr 23' ?

wzab, 2020-09-24:
The cli_app is just taken from master. Yes, you are right, I've forgotten to add "set". I'll check it in the afternoon.
With best regards,
Wojtek

wzab, 2020-09-24:
Yes, you are right! It was my stupid mistake. The communication with cli_app works perfectly:
```
# cli_app set txpwr 23
Tx power : 23                   Calibration_use : ON
OK
# cli_app set maxagg 1 8
-------------------------------- updated aggregation ----------------------------------
AC       : BK           State    : OFF          Value    : 8            Size     : 0
AC       : BE           State    : ON           Value    : 8            Size     : 0
AC       : VI           State    : OFF          Value    : 8            Size     : 0
AC       : VO           State    : OFF          Value    : 8            Size     : 0
---------------------------------------------------------------------------------------
OK
# cli_app set gi short
guard interval : short
OK
# 
```
So it looks like your driver is 64-bit compatible,
Now I have to check was is the cause of problems with 5.x kernel compatibility...
With best regards,
Wojtek

wzab, 2020-09-24:
I have built a version for kernel 5.3
You can find it in https://github.com/wzab/nrc7292-buildroot/tree/rpi3_5.3 
Unfortunately, an attempt to load the driver leads to the old, known error:

```
[   34.478101] ------------[ cut here ]------------
[   34.491553] WARNING: CPU: 1 PID: 33 at net/wireless/core.c:868 wiphy_register+0x9b0/0xb10 [cfg80211]
[   34.509275] Modules linked in: nrc(O) mac80211 sha256_generic cfg80211 rfkill libarc4 spi_bcm2835
[   34.526780] CPU: 1 PID: 33 Comm: kworker/1:1 Tainted: G           O      5.3.18-v8 #1
[   34.543296] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
[   34.557877] Workqueue: events nrc_check_start [nrc]
[   34.571448] pstate: 00000005 (nzcv daif -PAN -UAO)
[   34.585096] pc : wiphy_register+0x9b0/0xb10 [cfg80211]
[   34.599074] lr : wiphy_register+0x9b0/0xb10 [cfg80211]
[   34.612822] sp : ffffff801013bb90
[   34.624616] x29: ffffff801013bb90 x28: 0000000000000035 
[   34.638339] x27: 0000000000000001 x26: 000000000000000c 
[   34.652018] x25: 0000000000000000 x24: 0000000000000000 
[   34.665574] x23: ffffffd8360b9d60 x22: ffffffd837029d40 
[   34.679074] x21: ffffffd8360b7f00 x20: 00000000000007dc 
[   34.692565] x19: ffffffd8360b8300 x18: 0000000000000010 
[   34.705971] x17: 0000000000000000 x16: 0000000000000000 
[   34.719295] x15: 0000000000001e00 x14: 0000000000000020 
[   34.732556] x13: 0000000000000000 x12: 0000000000000001 
[   34.745767] x11: 00000000ffffffff x10: ffffffd8372da4c0 
[   34.758941] x9 : ffffff80127d8220 x8 : 0000000000000008 
[   34.772184] x7 : 00000000000000da x6 : ffffffa5cefd8000 
[   34.785362] x5 : 0000000000000001 x4 : ffffffd837fa49e8 
[   34.798458] x3 : ffffffd837fb4d48 x2 : c0b02abfe9604900 
[   34.811553] x1 : 0000000000000000 x0 : 0000000000000024 
[   34.824614] Call trace:
[   34.834857]  wiphy_register+0x9b0/0xb10 [cfg80211]
[   34.847567]  ieee80211_register_hw+0x44c/0xba0 [mac80211]
[   34.860842]  nrc_register_hw+0x48c/0x5d0 [nrc]
[   34.873076]  nrc_check_start+0x35c/0x5f0 [nrc]
[   34.885134]  process_one_work+0x1c0/0x430
[   34.896724]  worker_thread+0x50/0x410
[   34.907965]  kthread+0x104/0x130
[   34.918644]  ret_from_fork+0x10/0x1c
[   34.929579] ---[ end trace 996d219cc3ea8b7d ]---
[   34.941780] failed to register hw
[   34.952533] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000
[   34.968944] Mem abort info:
[   34.979223]   ESR = 0x96000005
[   34.989687]   Exception class = DABT (current EL), IL = 32 bits
[   35.003051]   SET = 0, FnV = 0
[   35.013541]   EA = 0, S1PTW = 0
[   35.024093] Data abort info:
[   35.034311]   ISV = 0, ISS = 0x00000005
[   35.045437]   CM = 0, WnR = 0
[   35.055610] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000036461000
[   35.069384] [0000000000000000] pgd=0000000000000000, pud=0000000000000000
[   35.083477] Internal error: Oops: 96000005 [#1] PREEMPT SMP
[   35.096232] Modules linked in: nrc(O) mac80211 sha256_generic cfg80211 rfkill libarc4 spi_bcm2835
[   35.112327] CPU: 1 PID: 33 Comm: kworker/1:1 Tainted: G        W  O      5.3.18-v8 #1
[   35.127322] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
[   35.140321] Workqueue: events nrc_check_start [nrc]
[   35.152232] pstate: 60000005 (nZCv daif -PAN -UAO)
[   35.163983] pc : nested_table_free+0xc/0x60
[   35.175180] lr : bucket_table_free+0x6c/0xa0
[   35.186269] sp : ffffff801013bc20
[   35.196259] x29: ffffff801013bc20 x28: ffffffa5cefd6000 
[   35.208181] x27: ffffff80100b3cb8 x26: ffffffd8360b8d90 
[   35.219977] x25: ffffffa5cf1899d8 x24: ffffffd836a51d80 
[   35.231692] x23: 0000000000000000 x22: 0000000000000036 
[   35.243414] x21: 0000000008000000 x20: 0000000000000008 
[   35.255118] x19: ffffffd836a51d80 x18: 0000000000000010 
[   35.266773] x17: 0000000000000000 x16: 0000000000000000 
[   35.278303] x15: 0000000000001e00 x14: 0000000000000020 
[   35.289653] x13: 0000000000000040 x12: 0000000000000228 
[   35.300820] x11: 0000000000000000 x10: 0000000000000000 
[   35.311897] x9 : 0000000000000000 x8 : 0000000000000000 
[   35.322874] x7 : 00000000000000fd x6 : ffffffa5cefd8000 
[   35.333794] x5 : 0000000000000000 x4 : 0000000000000001 
[   35.344673] x3 : ffffffa5cefd8db0 x2 : ffffffd837029d40 
[   35.355629] x1 : 0000000000000036 x0 : 0000000000000000 
[   35.366553] Call trace:
[   35.374597]  nested_table_free+0xc/0x60
[   35.384052]  bucket_table_free+0x6c/0xa0
[   35.393638]  rhashtable_free_and_destroy+0x5c/0x150
[   35.404419]  sta_info_stop+0x30/0x40 [mac80211]
[   35.414816]  ieee80211_free_hw+0x4c/0xb0 [mac80211]
[   35.425528]  nrc_free_hw+0x20/0x30 [nrc]
[   35.435268]  nrc_check_start+0x470/0x5f0 [nrc]
[   35.445583]  process_one_work+0x1c0/0x430
[   35.455494]  worker_thread+0x50/0x410
[   35.465031]  kthread+0x104/0x130
[   35.474101]  ret_from_fork+0x10/0x1c
[   35.483486] Code: d503201f a9bd7bfd 910003fd a90153f3 (f9400013) 
[   35.495552] ---[ end trace 996d219cc3ea8b7e ]---
```

wzab, 2020-09-24:
I have increased the verbosity of debugging in the `nrc` driver, and added the `WIRELESS_REGDB` package in Buildroot.
Now I get the following detailed error messages:
```
Welcome to Buildroot
buildroot login: root
# 
# 
# modprobe spi-bcm2835
nrc fw_name=nrc7292_cspi.bin bd_name=nrc7292_bd.dat hifspeed=16000000# modprobe nrc fw_name=nrc7292_cspi.bin bd_name=nrc7292_bd.dat hifspeed=16000000
[   32.047614] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   32.388090] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   32.597647] nrc: loading out-of-tree module taints kernel.
# [   32.654463] start FW 297
[   33.253303] end FW
[   36.335321] ------------[ cut here ]------------
[   36.349018] WARNING: CPU: 3 PID: 145 at net/wireless/core.c:868 wiphy_register+0x9b0/0xb10 [cfg80211]
[   36.367048] Modules linked in: nrc(O) mac80211 sha256_generic cfg80211 rfkill libarc4 spi_bcm2835
[   36.384718] CPU: 3 PID: 145 Comm: kworker/3:2 Tainted: G           O      5.3.18-v8 #1
[   36.401446] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
[   36.416099] Workqueue: events nrc_check_start [nrc]
[   36.429735] pstate: 00000005 (nzcv daif -PAN -UAO)
[   36.443434] pc : wiphy_register+0x9b0/0xb10 [cfg80211]
[   36.457495] lr : wiphy_register+0x9b0/0xb10 [cfg80211]
[   36.471360] sp : ffffff801029bb90
[   36.483349] x29: ffffff801029bb90 x28: 0000000000000035 
[   36.497373] x27: 0000000000000001 x26: 000000000000000c 
[   36.511348] x25: 0000000000000000 x24: 0000000000000000 
[   36.525230] x23: ffffffd1f6a99d60 x22: ffffffd1f6589d40 
[   36.539023] x21: ffffffd1f6a97f00 x20: 00000000000007dc 
[   36.552692] x19: ffffffd1f6a98300 x18: 0000000000000010 
[   36.566293] x17: 0000000000000000 x16: 0000000000000001 
[   36.579864] x15: 0000000000001e00 x14: 0000000000000020 
[   36.593326] x13: 0000000000000000 x12: 0000000000000001 
[   36.606680] x11: 00000000ffffffff x10: ffffffd1f732bcf0 
[   36.619966] x9 : ffffff80127d8220 x8 : 0000000000000008 
[   36.633223] x7 : 00000000000000f3 x6 : ffffffa3f93d8000 
[   36.646364] x5 : 0000000000000000 x4 : 0000000000000001 
[   36.659380] x3 : ffffffa3f93d8db0 x2 : 639a5530193f3900 
[   36.672350] x1 : 0000000000000000 x0 : 0000000000000024 
[   36.685272] Call trace:
[   36.695350]  wiphy_register+0x9b0/0xb10 [cfg80211]
[   36.707892]  ieee80211_register_hw+0x44c/0xba0 [mac80211]
[   36.720922]  nrc_register_hw+0x48c/0x5d0 [nrc]
[   36.732932]  nrc_check_start+0x35c/0x5f0 [nrc]
[   36.744794]  process_one_work+0x1c0/0x430
[   36.756139]  worker_thread+0x50/0x410
[   36.767026]  kthread+0x104/0x130
[   36.777326]  ret_from_fork+0x10/0x1c
[   36.787899] ---[ end trace d2bc9264efb0d86a ]---
[   36.799749] failed to register hw
[   36.810695] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000
[   36.826720] Mem abort info:
[   36.836787]   ESR = 0x96000005
[   36.846701]   Exception class = DABT (current EL), IL = 32 bits
[   36.859707]   SET = 0, FnV = 0
[   36.869584]   EA = 0, S1PTW = 0
[   36.879680] Data abort info:
[   36.889349]   ISV = 0, ISS = 0x00000005
[   36.900308]   CM = 0, WnR = 0
[   36.910008] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000036456000
[   36.923458] [0000000000000000] pgd=0000000000000000, pud=0000000000000000
[   36.937029] Internal error: Oops: 96000005 [#1] PREEMPT SMP
[   36.949319] Modules linked in: nrc(O) mac80211 sha256_generic cfg80211 rfkill libarc4 spi_bcm2835
[   36.965122] CPU: 3 PID: 145 Comm: kworker/3:2 Tainted: G        W  O      5.3.18-v8 #1
[   36.979986] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
[   36.992801] Workqueue: events nrc_check_start [nrc]
[   37.004598] pstate: 60000005 (nZCv daif -PAN -UAO)
[   37.016272] pc : nested_table_free+0xc/0x60
[   37.027237] lr : bucket_table_free+0x6c/0xa0
[   37.038282] sp : ffffff801029bc20
[... 350 lines cut ...]
[   37.107373] x19: ffffffd1f60c1580 x18: 0000000000000010 
[   37.119050] x17: 0000000000000000 x16: 0000000000000000 
[   37.130564] x15: 0000000000001e00 x14: 0000000000000020 
[   37.141897] x13: 0000000000000040 x12: 0000000000000228 
[   37.153093] x11: 0000000000000000 x10: 0000000000000000 
[   37.164172] x9 : 0000000000000000 x8 : 0000000000000000 
[   37.175138] x7 : 0000b5fecb5fe5be x6 : ffffffa3f93d8000 
[   37.186096] x5 : 0000000000000000 x4 : 0000000000000001 
[   37.197018] x3 : ffffffa3f93d8db0 x2 : ffffffd1f6589d40 
[   37.207993] x1 : 0000000000007b06 x0 : 0000000000000000 
[   37.218993] Call trace:
[   37.227056]  nested_table_free+0xc/0x60
[   37.236596]  bucket_table_free+0x6c/0xa0
[   37.246214]  rhashtable_free_and_destroy+0x5c/0x150
[   37.257034]  sta_info_stop+0x30/0x40 [mac80211]
[   37.267484]  ieee80211_free_hw+0x4c/0xb0 [mac80211]
[   37.278214]  nrc_free_hw+0x20/0x30 [nrc]
[   37.288010]  nrc_check_start+0x470/0x5f0 [nrc]
[   37.298358]  process_one_work+0x1c0/0x430
[   37.308276]  worker_thread+0x50/0x410
[   37.317862]  kthread+0x104/0x130
[   37.326954]  ret_from_fork+0x10/0x1c
[   37.336399] Code: d503201f a9bd7bfd 910003fd a90153f3 (f9400013) 
[   37.348482] ---[ end trace d2bc9264efb0d86b ]---
```

mokajava3, 2020-09-24:
Hi. Could you apply the attached files? We can run the firmware and associate successfully in v.5.4.51.
If you confirm this patch, we will apply this the next release version.[
[Patch Porting to kernel 5.4.x.zip](https://github.com/newracom/nrc7292_sw_pkg/files/5278573/Patch.Porting.to.kernel.5.4.x.zip)
](url)

wzab, 2020-09-24:
Great! It works on RPi3 with kernel 5.4 . My test Buildroot is [here](https://github.com/wzab/nrc7292-buildroot/tree/rpi3-5.4)
Thanks a lot!
With best regards,
Wojtek

closed 2020-09-25 by newracom

wzab, 2020-09-25:
I confirm that this version of kernel works also on Raspberry Pi 3 with 64-bit kernel 5.8.9-v8 
The Buildroot configuration used for testing is available at https://github.com/wzab/nrc7292-buildroot/tree/rpi3-5.8 
```
Welcome to Buildroot
buildroot login: root
# uname -r
5.8.9-v8
# uname -a
Linux buildroot 5.8.9-v8 #1 SMP PREEMPT Fri Sep 25 23:33:27 CEST 2020 aarch64 GNU/Linux
# modprobe spi-bcm2835
#modprobe nrc fw_name=nrc7292_cspi.bin bd_name=nrc7292_bd.dat hifspeed=16000000
[   25.830663] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   25.951710] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   26.120551] nrc: loading out-of-tree module taints kernel.
# [   26.171138] start FW 297
[   26.769972] end FW
# ifconfig wlan0 up
# cli_app set txpwr 23
_app set maxagg 1 8
cli_app set gi shortTx power : 23                       Calibration_use : ON
OK
# cli_app set maxagg 1 8
-------------------------------- updated aggregation ----------------------------------
AC       : BK           State    : OFF          Value    : 8            Size     : 0
AC       : BE           State    : ON           Value    : 8            Size     : 0
AC       : VI           State    : OFF          Value    : 8            Size     : 0
AC       : VO           State    : OFF          Value    : 8            Size     : 0
---------------------------------------------------------------------------------------
OK
# cli_app set gi short
guard interval : short
OK
# iwlist wlan0 scan
wlan0     Scan completed :
          Cell 01 - Address: 84:25:3F:87:FD:5A
                    Channel:165
                    Frequency:5.825 GHz
                    Quality=70/70  Signal level=5 dBm  
                    Encryption key:on
                    ESSID:"Wi-Fi"
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000002e6521a
                    Extra: Last beacon: 141ms ago
                    IE: Unknown: 000557692D4669
                    IE: Unknown: 01088C129824B048606C
                    IE: Unknown: 0301A5
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : unknown (8)
                    IE: Unknown: 2D1A0000000000000000000000000000000000000000000000000000
                    IE: Unknown: 3D16A5000000000000000000000000000000000000000000
                    IE: Unknown: 7F080000000000000000
                    IE: Unknown: DD180050F2020101000003A4000027A4000042435E0062322F00
```