newracom/nrc7292_sw_pkg #4: Incorrect behavior of the driver, when GPIOs are busy

wzab, 2020-09-21:
I have checked the operation of the driver with kernel 4.14.70-v7+ on Rapsberry Pi 3.
I have exported the GPIOs used by the EVK:
root@raspberrypi:/home/pi# echo 5 > /sys/class/gpio/export
root@raspberrypi:/home/pi# echo 5 > /sys/class/gpio/export
Then I have loaded the driver as usual.
Instead of the error message informing about busy GPIO resources, I have got the following errors in the kernel:
```
[   76.463975] nrc: loading out-of-tree module taints kernel.
[   76.469823] nrc-cspi: probe of spi0.0 failed with error -22
[   76.470302] Unable to handle kernel NULL pointer dereference at virtual address 00000004
[   76.470307] Unable to handle kernel NULL pointer dereference at virtual address 00000004
[   76.470320] pgd = 80004000
[   76.470325] [00000004] *pgd=00000000
[   76.470341] Internal error: Oops: 17 [#1] SMP ARM
[   76.470349] Modules linked in: nrc(O+) fuse snd_bcm2835(C) snd_pcm snd_timer snd spi_bcm2835 uio_pdrv_genirq uio fixed mac80211 cfg80211 rfkill i2c_dev ip_tables x_tables ipv6
[   76.470429] CPU: 0 PID: 661 Comm: irq/169-spi-irq Tainted: G         C O    4.14.70-v7+ #1144
[   76.470432] Hardware name: BCM2835
[   76.470439] task: aa891e00 task.stack: aaa1a000
[   76.470589] PC is at spi_update_status+0x1c/0x224 [nrc]
[   76.470679] LR is at spi_irq+0x20/0x3c [nrc]
[   76.470686] pc : [<7f3077e8>]    lr : [<7f307b58>]    psr: 68000013
[   76.470691] sp : aaa1bee0  ip : aaa1bf08  fp : aaa1bf04
[   76.470696] r10: 801775b8  r9 : b2e66b00  r8 : b4641800
[   76.470702] r7 : aa8c3000  r6 : b4641800  r5 : 00000000  r4 : aa8c2c00
[   76.470708] r3 : 7f307b38  r2 : 00000000  r1 : aa8c2c00  r0 : aa8c3000
[   76.470718] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   76.470724] Control: 10c5383d  Table: 3449806a  DAC: 00000055
[   76.470730] Process irq/169-spi-irq (pid: 661, stack limit = 0xaaa1a210)
[   76.470737] Stack: (0xaaa1bee0 to 0xaaa1c000)
[   76.470748] bee0: aa8c2c00 b2e66b24 b4641800 00000001 b4641800 b2e66b00 aaa1bf1c aaa1bf08
[   76.470761] bf00: 7f307b58 7f3077d8 b2e66b00 b2e66b24 aaa1bf3c aaa1bf20 801775e4 7f307b44
[   76.470772] bf20: 00000004 aaa1a000 b2e66b24 00000000 aaa1bf7c aaa1bf40 80177930 801775c4
[   76.470782] bf40: 00000000 80c882c4 00000000 80177724 b5df441c b5df4400 00000000 b2e66e40
[   76.470793] bf60: b2e66b00 801777e8 b5df441c b471bb58 aaa1bfac aaa1bf80 8013dacc 801777f4
[   76.470805] bf80: 80102d94 b2e66e40 8013d990 00000000 00000000 00000000 00000000 00000000
[   76.470815] bfa0: 00000000 aaa1bfb0 8010810c 8013d99c 00000000 00000000 00000000 00000000
[   76.470825] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   76.470835] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[   76.471052] [<7f3077e8>] (spi_update_status [nrc]) from [<7f307b58>] (spi_irq+0x20/0x3c [nrc])
[   76.471160] [<7f307b58>] (spi_irq [nrc]) from [<801775e4>] (irq_thread_fn+0x2c/0x44)
[   76.471179] [<801775e4>] (irq_thread_fn) from [<80177930>] (irq_thread+0x148/0x20c)
[   76.471198] [<80177930>] (irq_thread) from [<8013dacc>] (kthread+0x13c/0x16c)
[   76.471224] [<8013dacc>] (kthread) from [<8010810c>] (ret_from_fork+0x14/0x28)
[   76.471236] Code: e52de004 e8bd4000 e5905054 e1a07000 (e5953004) 
[   76.471247] ---[ end trace da7ce380b3fd4612 ]---
[   76.471284] genirq: exiting task "irq/169-spi-irq" (661) is an active IRQ thread (irq 169)
[   76.717236] irq 169: nobody cared (try booting with the "irqpoll" option)
[   76.717245] CPU: 0 PID: 661 Comm: irq/169-spi-irq Tainted: G      D  C O    4.14.70-v7+ #1144
[   76.717247] Hardware name: BCM2835
[   76.717272] [<8010ffd8>] (unwind_backtrace) from [<8010c240>] (show_stack+0x20/0x24)
[   76.717283] [<8010c240>] (show_stack) from [<80787b24>] (dump_stack+0xd4/0x118)
[   76.717297] [<80787b24>] (dump_stack) from [<80178f5c>] (__report_bad_irq+0x38/0xd0)
[   76.717306] [<80178f5c>] (__report_bad_irq) from [<80179360>] (note_interrupt+0x280/0x2d0)
[   76.717314] [<80179360>] (note_interrupt) from [<80176418>] (handle_irq_event_percpu+0x5c/0x68)
[   76.717322] [<80176418>] (handle_irq_event_percpu) from [<80176478>] (handle_irq_event+0x54/0x78)
[   76.717330] [<80176478>] (handle_irq_event) from [<8017a570>] (handle_level_irq+0xb4/0x140)
[   76.717340] [<8017a570>] (handle_level_irq) from [<80175338>] (generic_handle_irq+0x34/0x44)
[   76.717349] [<80175338>] (generic_handle_irq) from [<804caf9c>] (bcm2835_gpio_irq_handle_bank+0x88/0xac)
[   76.717359] [<804caf9c>] (bcm2835_gpio_irq_handle_bank) from [<804cb04c>] (bcm2835_gpio_irq_handler+0x8c/0x124)
[   76.717367] [<804cb04c>] (bcm2835_gpio_irq_handler) from [<80175338>] (generic_handle_irq+0x34/0x44)
[   76.717376] [<80175338>] (generic_handle_irq) from [<804c44d8>] (bcm2836_chained_handle_irq+0x38/0x50)
[   76.717384] [<804c44d8>] (bcm2836_chained_handle_irq) from [<80175338>] (generic_handle_irq+0x34/0x44)
[   76.717391] [<80175338>] (generic_handle_irq) from [<80175978>] (__handle_domain_irq+0x6c/0xc4)
[   76.717400] [<80175978>] (__handle_domain_irq) from [<80101504>] (bcm2836_arm_irqchip_handle_irq+0xa8/0xac)
[   76.717408] [<80101504>] (bcm2836_arm_irqchip_handle_irq) from [<807a36bc>] (__irq_svc+0x5c/0x7c)
[   76.717412] Exception stack(0xaaa1bc48 to 0xaaa1bc90)
[   76.717417] bc40:                   b6b81ba8 00000000 b4641800 00000000 b4641800 b2e66b00
[   76.717423] bc60: b464185c b4641810 80c97e04 aaa1bf48 aa891e00 aaa1bcb4 80c04174 aaa1bc98
[   76.717427] bc80: 80177514 80177518 68000113 ffffffff
[   76.717434] [<807a36bc>] (__irq_svc) from [<80177518>] (irq_finalize_oneshot+0x98/0x138)
[... 66 lines cut ...]
[   77.067838] bf80: 5f660ef0 c0241600 00000005 0003f040 80108204 b471a000 00000000 b471bfa8
[   77.067843] bfa0: 80108060 801a90b0 c0241600 00000005 00000003 00bf6008 00000000 76f64794
[   77.067849] bfc0: c0241600 00000005 0003f040 0000017b 00000000 7e9f4c94 00000005 00bf6008
[   77.067854] bfe0: 7e9f4ab0 7e9f4aa0 0002212c 76ef64f0 60000010 00000003 00000000 00000000
[   77.067951] [<7f3077e8>] (spi_update_status [nrc]) from [<7f307adc>] (spi_start+0xec/0x148 [nrc])
[   77.068039] [<7f307adc>] (spi_start [nrc]) from [<7f2ff598>] (nrc_hif_init+0xb4/0xc0 [nrc])
[   77.068126] [<7f2ff598>] (nrc_hif_init [nrc]) from [<7f2fe2f4>] (nrc_platform_probe+0x68/0x130 [nrc])
[   77.068182] [<7f2fe2f4>] (nrc_platform_probe [nrc]) from [<80545fd0>] (platform_drv_probe+0x5c/0xc0)
[   77.068191] [<80545fd0>] (platform_drv_probe) from [<805445b0>] (driver_probe_device+0x244/0x300)
[   77.068198] [<805445b0>] (driver_probe_device) from [<80544728>] (__driver_attach+0xbc/0xc0)
[   77.068208] [<80544728>] (__driver_attach) from [<805425cc>] (bus_for_each_dev+0x7c/0xb0)
[   77.068216] [<805425cc>] (bus_for_each_dev) from [<80543ef4>] (driver_attach+0x2c/0x30)
[   77.068222] [<80543ef4>] (driver_attach) from [<805439d0>] (bus_add_driver+0x1ac/0x224)
[   77.068228] [<805439d0>] (bus_add_driver) from [<80544e9c>] (driver_register+0x88/0x108)
[   77.068235] [<80544e9c>] (driver_register) from [<80545f20>] (__platform_driver_register+0x50/0x58)
[   77.068283] [<80545f20>] (__platform_driver_register) from [<7f323038>] (nrc_init+0x38/0x1000 [nrc])
[   77.068344] [<7f323038>] (nrc_init [nrc]) from [<80101bf0>] (do_one_initcall+0x50/0x17c)
[   77.068355] [<80101bf0>] (do_one_initcall) from [<801a97c8>] (do_init_module+0x74/0x20c)
[   77.068364] [<801a97c8>] (do_init_module) from [<801a87ec>] (load_module+0x1c28/0x23a0)
[   77.068372] [<801a87ec>] (load_module) from [<801a9140>] (SyS_finit_module+0x9c/0xac)
[   77.068381] [<801a9140>] (SyS_finit_module) from [<80108060>] (ret_fast_syscall+0x0/0x28)
[   77.068387] Code: e52de004 e8bd4000 e5905054 e1a07000 (e5953004) 
[   77.068394] ---[ end trace da7ce380b3fd4613 ]---
root@raspberrypi:/home/pi# 
```

newracom, 2020-09-23:
Thank you for your report.
This comes from the fact that NRC7292 uses the same GPIO 5 in HSPI_EIRQ.
We will handle this decently as you suggested. So please avoid GPIO 5 usage.

closed 2020-09-23 by newracom

wzab, 2020-09-23:
The same is also with GPIO 6. I've made a mistake in my report.
It was:
```
root@raspberrypi:/home/pi# echo 5 > /sys/class/gpio/export
root@raspberrypi:/home/pi# echo 6 > /sys/class/gpio/export
```
With best regards,
Wojtek