newracom/nrc7292_sw_pkg #5: Porting to RK3399-based platform incorrect response to C_SPI_READ command

wzab, 2020-09-27:
When I use the driver on the RK3399-based platform, the board responds for the C_SPI_READ command with:
`0xff,0xff,0xff,0xff,0xff,0xff,0xff,0x5f`, while the last byte is supposed to contain C_SPI_ACK = 0x47.
I have checked the signals with the oscilloscope, and they seem to be correct, however the CS0 and SCK lines have illegal levels (ca. 0.9V) before the first transfer is started.
To avoid problems caused by transients, I have duplicated the attempt to read the register 0. However in both cases the board responds in the same incorrect way.
By the way, there is a problem in the driver because the failure to contact the board triggers the whole cascade of errors.
Below are the messages captured during the test.
The sources used for tests are available in https://github.com/wzab/nrc7292-buildroot in branch  `rpi_n10_5.6.19`.

```
# . ./test1
[   38.435976] WZab STF: 0
[   38.435976]  buffer:ff,ff,ff,ff,ff,ff,ff,5f
[   38.436622] ------------[ cut here ]------------
[   38.437082] WARNING: CPU: 3 PID: 308 at /home/wzab/pados_roboczy_wifi/git/nrc7292-buildroot/buildroot-2020.08/output/build/nrc-module-1.0/./nrc-hif-cspi.c:537 _c_spi_r]
[   38.438593] Modules linked in: nrc(O+) mac80211 libarc4 cfg80211 rfkill ipv6 nf_defrag_ipv6 rockchipdrm realtek analogix_dp dw_mipi_dsi dw_hdmi cec drm_kms_helper dwma]
[   38.440812] CPU: 3 PID: 308 Comm: modprobe Tainted: G           O      5.7.2 #1
[   38.441456] Hardware name: Radxa ROCK Pi N10 (DT)
[   38.441876] pstate: 20000005 (nzCv daif -PAN -UAO)
[   38.442319] pc : _c_spi_read_regs+0x284/0x2a0 [nrc]
[   38.442769] lr : _c_spi_read_regs+0x23c/0x2a0 [nrc]
[   38.443200] sp : ffff8000120eb3c0
[   38.443495] x29: ffff8000120eb3d0 x28: ffff800012185000 
[   38.443967] x27: 0000000000000100 x26: ffff0000b82e33b8 
[   38.444438] x25: ffff800009133e10 x24: ffff0000b6c8a540 
[   38.444909] x23: 0000000050200000 x22: 0000000000000000 
[   38.445379] x21: ffff0000b6c8a540 x20: 0000000000000010 
[   38.445850] x19: ffff0000b9006200 x18: 0000000000000010 
[   38.446321] x17: 0000000000000001 x16: 0000000000000001 
[   38.446792] x15: ffff0000b9006670 x14: 0720072007200720 
[   38.447263] x13: 0720072007200720 x12: 0720072007200720 
[   38.447734] x11: 0720072007200720 x10: 0720072007200720 
[   38.448205] x9 : 0720072007200720 x8 : 0720072007200720 
[   38.448676] x7 : 0720072007200720 x6 : 0000000000000163 
[   38.449146] x5 : 0000000000000001 x4 : 0000000000000000 
[   38.449617] x3 : 0000000000000000 x2 : f45439dc40a32c00 
[   38.450087] x1 : 0000000000000000 x0 : 000000000000005f 
[   38.450559] Call trace:
[   38.450799]  _c_spi_read_regs+0x284/0x2a0 [nrc]
[   38.451219]  c_spi_read_regs+0x2c/0x70 [nrc]
[   38.451616]  c_spi_probe+0xc8/0x360 [nrc]
[   38.451976]  spi_drv_probe+0x80/0xe0
[   38.452298]  really_probe+0xd8/0x320
[   38.452617]  driver_probe_device+0x58/0xf0
[   38.452984]  device_driver_attach+0x74/0x80
[   38.453356]  __driver_attach+0x58/0xe0
[   38.453692]  bus_for_each_dev+0x70/0xc0
[   38.454033]  driver_attach+0x24/0x30
[   38.454353]  bus_add_driver+0x14c/0x1f0
[   38.454694]  driver_register+0x64/0x120
[   38.455037]  __spi_register_driver+0x5c/0x70
[   38.455433]  nrc_hif_cspi_init+0xdc/0x170 [nrc]
[   38.455853]  nrc_hif_init+0x14/0xb0 [nrc]
[   38.456227]  nrc_platform_probe+0x58/0x120 [nrc]
[   38.456639]  platform_drv_probe+0x54/0xb0
[   38.456996]  really_probe+0xd8/0x320
[   38.457316]  driver_probe_device+0x58/0xf0
[   38.457682]  device_driver_attach+0x74/0x80
[   38.458054]  __driver_attach+0x58/0xe0
[   38.458389]  bus_for_each_dev+0x70/0xc0
[   38.458730]  driver_attach+0x24/0x30
[   38.459050]  bus_add_driver+0x14c/0x1f0
[   38.459391]  driver_register+0x64/0x120
[   38.459735]  __platform_driver_register+0x48/0x60
[   38.460169]  nrc_init+0x60/0xc0 [nrc]
[   38.460499]  do_one_initcall+0x54/0x1b0
[   38.460843]  do_init_module+0x54/0x210
[   38.461178]  load_module+0x1d08/0x22c0
[   38.461513]  __do_sys_finit_module+0xd8/0xf0
[   38.461892]  __arm64_sys_finit_module+0x20/0x30
[   38.462297]  el0_svc_common.constprop.0+0x6c/0x170
[... 180 lines cut ...]
[   38.783325]  show_stack+0x18/0x30
[   38.783625]  dump_stack+0xb8/0x104
[   38.783932]  __report_bad_irq+0x4c/0xdc
[   38.784276]  note_interrupt+0x2d4/0x380
[   38.784619]  handle_irq_event_percpu+0x88/0x90
[   38.785015]  handle_irq_event+0x48/0xd0
[   38.785356]  handle_level_irq+0xb8/0x140
[   38.785706]  generic_handle_irq+0x30/0x50
[   38.786065]  rockchip_irq_demux+0xac/0x1d0
[   38.786430]  generic_handle_irq+0x30/0x50
[   38.786788]  __handle_domain_irq+0x64/0xc0
[   38.787153]  gic_handle_irq+0x60/0x160
[   38.787488]  el1_irq+0xb8/0x180
[   38.787772]  cpuidle_enter_state+0x90/0x300
[   38.788144]  cpuidle_enter+0x38/0x50
[   38.788465]  call_cpuidle+0x1c/0x40
[   38.788776]  do_idle+0x1f4/0x290
[   38.789065]  cpu_startup_entry+0x28/0x50
[   38.789416]  rest_init+0xd8/0xe8
[   38.789708]  arch_call_rest_init+0x10/0x1c
[   38.790074]  start_kernel+0x4a4/0x4dc
[   38.790399] handlers:
[   38.790605] [<000000003c5614da>] irq_default_primary_handler threaded [<00000000ca6bdd86>] spi_irq [nrc]
[   38.791474] Disabling IRQ #120
```

wzab, 2020-09-27:
I have noticed one difference between the expected and real behaviour of the SPI.
When I set SPI speed to 16MHz, I get ca. 9.2MHz. When I set it 8 MHz, I get ca. 6MHz.
Is NRC7292 sensitive to using othe SPI clock frequency than 16MHz?

wzab, 2020-09-27:
I changed the requested SCK frequency from 16MHz to 25MHz. Now the real frequency is ca. 18.5MHz and NC7292 responds correctly. I have to remove debug messages to check if the whole initialization works.

wzab, 2020-09-27:
After removal of debug messages, I got another error message:
```
[  720.291067] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[  720.319583] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[  720.372384] nrc: loading out-of-tree module taints kernel.
[  720.378979] !!!!! garbage rx data
```
After some time another error was displayed:
```
[  758.932422] ------------[ cut here ]------------
[  758.932855] WARNING: CPU: 4 PID: 328 at /home/wzab/pados_roboczy_wifi/git/nrc7292-buildroot/buildroot-2020.08/output/build/nr]
[  758.934353] Modules linked in: nrc(O) mac80211 libarc4 cfg80211 rfkill ipv6 nf_defrag_ipv6 rockchipdrm realtek analogix_dp dwl
[  758.936380] CPU: 4 PID: 328 Comm: irq/120-spi-irq Tainted: G           O      5.7.2 #1
[  758.937073] Hardware name: Radxa ROCK Pi N10 (DT)
[  758.937486] pstate: 20000005 (nzCv daif -PAN -UAO)
[  758.937916] pc : _c_spi_read_regs+0x230/0x250 [nrc]
[  758.938351] lr : _c_spi_read_regs+0x1c0/0x250 [nrc]
[  758.938778] sp : ffff800011edba40
[  758.939069] x29: ffff800011edba40 x28: ffff0000b81c1c00 
[  758.939535] x27: ffff0000b81c1c00 x26: ffff800010154b90 
[  758.940001] x25: 0000000000000001 x24: ffff0000b8b34800 
[  758.940466] x23: ffff0000b8b34800 x22: 0000000000000020 
[  758.940931] x21: 0000000000000010 x20: ffff0000b81c1c00 
[  758.941397] x19: ffff0000b8b34550 x18: 0000000000000006 
[  758.941862] x17: 000000000000000b x16: 0000000000000007 
[  758.942327] x15: 0000000000000009 x14: 000000000000003f 
[  758.942792] x13: 0000000000000000 x12: 000000000000002e 
[  758.943257] x11: 0000000000000000 x10: 0000000000000005 
[  758.943722] x9 : ffff0000bd9b1a90 x8 : ffff0000bd9b1100 
[  758.944188] x7 : ffff0000b9578700 x6 : 0000000000000003 
[  758.944653] x5 : 0000000000000000 x4 : 0000000000000000 
[  758.945117] x3 : 0000000000000001 x2 : 0000000000000002 
[  758.945583] x1 : e40033ed08578800 x0 : 00000000000000c7 
[  758.946048] Call trace:
[  758.946274]  _c_spi_read_regs+0x230/0x250 [nrc]
[  758.946680]  c_spi_read_regs+0x2c/0x70 [nrc]
[  758.947063]  spi_update_status+0x58/0x1e0 [nrc]
[  758.947469]  spi_irq+0x1c/0x50 [nrc]
[  758.947789]  irq_thread_fn+0x2c/0xa0
[  758.948104]  irq_thread+0x150/0x250
[  758.948413]  kthread+0x138/0x160
[  758.948700]  ret_from_fork+0x10/0x24
[  758.948703] ---[ end trace 0cf649f6715f3880 ]---
[  811.268431] ------------[ cut here ]------------
[  811.268864] WARNING: CPU: 4 PID: 328 at /home/wzab/pados_roboczy_wifi/git/nrc7292-buildroot/buildroot-2020.08/output/build/nr]
[  811.270361] Modules linked in: nrc(O) mac80211 libarc4 cfg80211 rfkill ipv6 nf_defrag_ipv6 rockchipdrm realtek analogix_dp dwl
[  811.272387] CPU: 4 PID: 328 Comm: irq/120-spi-irq Tainted: G        W  O      5.7.2 #1
[  811.273079] Hardware name: Radxa ROCK Pi N10 (DT)
[  811.273493] pstate: 20000005 (nzCv daif -PAN -UAO)
[  811.273922] pc : _c_spi_read_regs+0x230/0x250 [nrc]
[  811.274358] lr : _c_spi_read_regs+0x1c0/0x250 [nrc]
[  811.274784] sp : ffff800011edba40
[  811.275076] x29: ffff800011edba40 x28: ffff0000b81c1c00 
[  811.275542] x27: ffff0000b81c1c00 x26: ffff800010154b90 
[  811.276007] x25: 0000000000000001 x24: ffff0000b8b34800 
[  811.276472] x23: ffff0000b8b34800 x22: 0000000000000020 
[  811.276938] x21: 0000000000000010 x20: ffff0000b81c1c00 
[  811.277403] x19: ffff0000b8b34550 x18: 0000000000000006 
[  811.277868] x17: 000000000000000b x16: 0000000000000007 
[  811.278333] x15: 0000000000000009 x14: 0000000000000005 
[  811.278798] x13: 0000000000000000 x12: 0000000000000014 
[  811.279264] x11: 0000000000000000 x10: 00000000000009c0 
[  811.279729] x9 : ffff800011edb710 x8 : 0000000000000004 
[  811.280194] x7 : ffff0000b9578700 x6 : 0000000000000003 
[  811.280659] x5 : 0000000000000000 x4 : 0000000000000000 
[  811.281124] x3 : 0000000000000001 x2 : 0000000000000002 
[  811.281589] x1 : e40033ed08578800 x0 : 00000000000000c7 
[  811.282054] Call trace:
[  811.282279]  _c_spi_read_regs+0x230/0x250 [nrc]
[  811.282684]  c_spi_read_regs+0x2c/0x70 [nrc]
[  811.283066]  spi_update_status+0x58/0x1e0 [nrc]
[  811.283470]  spi_irq+0x1c/0x50 [nrc]
[  811.283791]  irq_thread_fn+0x2c/0xa0
[  811.284106]  irq_thread+0x150/0x250
[  811.284414]  kthread+0x138/0x160
[  811.284419]  ret_from_fork+0x10/0x24
[  811.285027] ---[ end trace 0cf649f6715f3881 ]---
```
So it looks like 16MHz clock is really important. Now the question is - how can I enforce availability of such clock in RockpiN10?

wzab, 2020-09-27:
On Rapsberrypi 3 the communication work even with very exotic SPI frequencies (e.g. 11 MHz). It must be another problem with SPI timing...

wzab, 2020-09-27:
In Raspberrypi 3 I was able to establish communication with NRC7292 even at SPI clock frequency as low as 800 kHz.
With Rockpi N10 (based on RK3399), I got the reasonable results (at least commands were confirmed only at frequency of ca 18.5 MHz.
The SPI timing looks correctly.

[image]
The data (yellow) are changing after the falling edge of the clock (blue).
[image]
The transfers (changes of clock - yellow) are active when CS (blue) is low. Clock between transfers is low.

I have added the delay of sampling of the received data: https://github.com/wzab/nrc7292-buildroot/blob/42aeb9441bdbb9a3f0c0486cfbb324cd4c95c92a/br2_kernel_patches/006-spi-rx-sample-delay-ns.patch#L9

I have tried to add delays at the begining and at the end of the transfer: 
https://github.com/wzab/nrc7292-buildroot/commit/fdd22ffeb66574c9a6075ef3e8ebf666aa335825

None of the above has cured the problem :(

newracom, 2020-09-28:
CSPI function of NRC7292 is a slave-only function. 
SPI clock speed depends on SCK clock frequency which is driven by HOST (which is the SPI master function).
The slave can support even low SCK clock speed.
As your description, we suspect that your HOST didn’t drive SCK/nCS output all the time. Sometimes, the host lets Hi-Z state on pins.
If you check the SCK/nCS signal level which is disconnected from NRC7292 with an oscilloscope, It could help to understand.

wzab, 2020-09-28:
Indeed, **before the first transfer**, the SCK and nCS lines are not driven (I have described in the initial report in that issue).
The question is if such event (e.g. spurious generation of incorrect SCK/nCS pulses due to floating line) may affect further operation of the NRC7292? If yes, how can I cure it? Is there any procedure to reset it?
I can change the configuration of those pins from "pcfg_pull_up" to "pcfg_pull_down" in the [devicetree](https://elixir.bootlin.com/linux/v5.8.6/source/arch/arm64/boot/dts/rockchip/rk3399.dtsi#L2380) and check if it helps.

newracom, 2020-09-28:
The below action is required to cure this situation.
- At least 64bit dummy SCK clock with keeping nCS low and MOSI high.

wzab, 2020-09-28:
Changing "pcfg_pull_up" to "pcfg_pull_down" didn't help. I can try tomorrow two possible fixes:

1. Solder stronger pull-up (ca. 1k?) to nCS and stronger pull-down (ca. 1k?) to SCK
2. Add your proposed cleanup (sending more then 8 0xff bytes) before trying to initialize NRC7292.

wzab, 2020-09-29:
Pulling the CS up and SCK down with external resistors didn't help.

wzab, 2020-09-29:
Adding the "reset" procedure also didn't help...

wzab, 2020-09-29:
Well, it seems that the cause of the problem is the level translator: https://www.ti.com/lit/ds/symlink/txb0108.pdf used on the Silex Technology module. The pull-up or pull-down resistors may disturb its operation. However, switching them off (like in the last commit: https://github.com/wzab/nrc7292-buildroot/commit/a9688ef8962e5faa4cc6a77fe73c32e82404d6dc ) also does not help.
The reason is probably the fact, that a similar translator is in the baseboard of the Rock Pi N10: https://dl.radxa.com/rockpin10/docs/hw/rockpi_n10_sch_v1.1_20190909.pdf 
[image]
As the result, The low level never drops below 0.9V, as can be seen in the oscillograms below:

[image]
Transmitted data. The low level reaches 0V.

[image]
Received data. The low level remains at 0.9V

The above picture explains, why the transmission could work at 18.75MHz. After the falling edge, the RX voltage drops below 0.9V for a short time due to "ringing"...

newracom, 2020-09-30:
Probably the level translator seems to be the cause of this problem as you thought.
But it is difficult to help you because we don’t have much information about your hardware environment.
So we need to have a common understanding of your situation in detail. Could you kindly answer the below questions?
- What is your 11ah module which you’re testing? (exact part name)
- What is the board name which was connected with your host board? Could you provide the board schematic of both?
- Could you take a picture of test boards and post them with indicating the probe point of previous oscilloscope captures?
- Do you have a way to bypass the level translator between the host and the 11ah module?

wzab, 2020-10-01:
Please see the photo of the module I use:

[image]

It is SX-NEWAH from Silex Technology. I do not have a schematic diagram, but I could identify the voltage level translator [TXB0108](https://www.ti.com/lit/ds/symlink/txb0108.pdf), denoted as U8 (next to R50, near to GPIO connector).

As the SBC to which I connect the 11ah module, I use the Rock Pi N10:

[image]

I have sent a link to its schematic diagram before:  https://dl.radxa.com/rockpin10/docs/hw/rockpi_n10_sch_v1.1_20190909.pdf
Here is the related part of the schematic diagram:

[image]

The level translators are the small BGA chips located just below the heat-sink, above the pins 34-37 of the GPIO conector.
So we have two voltage level translators with auto direction sensing connected in series. That's not likely to work reliably.
Unfortunately, we can't easily remove them (at least from Rock Pi N10).

yodazhong, 2020-10-02:
As you pointed out, it seems that this issue comes from the two voltage level translators.
You can try to remove the level translator on the Silex board and connect signals directly with your Rock Pi N10 board.
Or you can ask this to the Silex board provider.

newracom, 2020-10-02:
As you pointed out, it seems that this issue comes from the two voltage level translators.
You can try to remove the level translator on the Silex board and connect signals directly with your Rock Pi N10 board.
Or you can ask this to the Silex board provider.

closed 2020-10-02 by newracom

wzab, 2020-10-02:
I have been given the [AHMB7292S](https://www.alfa.com.tw/products_detail/339.htm) module from ALFA Network (many thanks to them!):
[image]
I have connected it simply via breadboard jumper wires:
[image]
In that configuration the network works correctly:
```
# . ./test1                                                                     
[   22.352293] cfg80211: Loading compiled-in X.509 certificates for regulatory e
[   22.380140] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'       
[   22.445530] nrc: loading out-of-tree module taints kernel.                   
# [   22.485843] start FW 297                                                   
[   23.301841] end FW                                                           
                                                                                
# iwconfig                                                                      
lo        no wireless extensions.                                               
                                                                                
eth0      no wireless extensions.                                               
                                                                                
wlan0     IEEE 802.11  ESSID:off/any                                            
          Mode:Managed  Access Point: Not-Associated   Tx-Power=0 dBm           
          Retry short limit:7   RTS thr:off   Fragment thr:off                  
          Encryption key:off                                                    
          Power Management:on                                                   
                                                                                
# ifconfig wlan0 up                                                             
# iwlist wlan0 scan                                                             
wlan0     Scan completed :                                                      
          Cell 01 - Address: 02:00:EB:B9:83:F9                                  
                    Channel:41                                                  
                    Frequency:5.205 GHz (Channel 41)                            
                    Quality=68/70  Signal level=-42 dBm                         
                    Encryption key:off                                          
                    ESSID:"halow_demo"                                          
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s         
                              36 Mb/s; 48 Mb/s; 54 Mb/s                         
                    Mode:Master                                                 
                    Extra:tsf=0000000000000000                                  
                    Extra: Last beacon: 188ms ago                               
                    IE: Unknown: 000A68616C6F775F64656D6F                       
                    IE: Unknown: 01088C129824B048606C                           
                    IE: Unknown: 030129                                         
                    IE: Unknown: 2D1A0000000000000000000000000000000000000000000
                    IE: Unknown: 3D162900000000000000000000000000000000000000000
                    IE: Unknown: 7F080000000000000000                           
                    IE: Unknown: DD180050F2020101000003A4000027A4000042435E00620
                                                                                
# iwconfig wlan0 essid halow_demo                                               
# [  499.790215] wlan0: authenticate with 02:00:eb:b9:83:f9                     
[  499.790746] wlan0: send auth to 02:00:eb:b9:83:f9 (try 1/3)                  
[  499.995552] wlan0: send auth to 02:00:eb:b9:83:f9 (try 2/3)                  
[  500.064692] wlan0: authenticated                                             
[  500.067229] wlan0: associate with 02:00:eb:b9:83:f9 (try 1/3)                
[  500.143341] wlan0: RX AssocResp from 02:00:eb:b9:83:f9 (capab=0x1 status=0 a)
[  500.144197] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready         
[  500.144689] wlan0: associated                                                
                                                                                
# iwconfig                                                                      
lo        no wireless extensions.                                               
                                                                                
eth0      no wireless extensions.                                               
                                                                                
wlan0     IEEE 802.11  ESSID:"halow_demo"                                       
          Mode:Managed  Frequency:5.205 GHz  Access Point: 02:00:EB:B9:83:F9    
          Bit Rate=6 Mb/s   Tx-Power=30 dBm                                     
          Retry short limit:7   RTS thr:off   Fragment thr:off                  
          Encryption key:off                                                    
          Power Management:on                                                   
          Link Quality=67/70  Signal level=-43 dBm                              
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0              
          Tx excessive retries:0  Invalid misc:4   Missed beacon:0              
[... 5 lines cut ...]
[... cut ...]
So the driver works perfectly with Rockchip N10 and with kernel 5.7.2.
The cause of the problem is probably the connection of two voltage level converters with auto direction sensing in series, that is created when one plugs the NEWAH-SX board to the Rock Pi N10 GPIO connectors.
Probably the NEWAH-SX modules will also work perfectly, when connected directly to SPI working at 1,8V.
Anyway, it seems that in case of unidirectional connections, like standard SPI lines, it may be better to use unidirectional voltage level translators.