newracom/nrc7292_sw_pkg #7: Driver crashes the system, when there is SPI communication problem

wzab, 2021-02-01:
If somebody loads the driver, while the modul is inaccessible, it crashes the kernel.
I have added diagnostic messages to see what is going on.
My driver with added debugging messages is available at https://github.com/wzab/nrc7292_sw_pkg/commit/4c0431eb69b9c3a7e9ec8e3b4a063c2d490e94be .
However similar problem happens also with standard driver.
I attach the dump from the console.
The driver reads from the SPI all 0xff bytes (because the module is not available):
`[   49.545244] nrc80211 nrc80211: read_regs:ff,ff,ff,ff,ff,ff,ff,ff`
The c_spi_probe function fails:
`[   49.701466] nrc-cspi: probe of spi5.0 failed with error -22`
But the spi_driver is registered, tries to communicate with the hardware and tries to dereference the non-initialized pointer.
```
[   49.701466] nrc-cspi: probe of spi5.0 failed with error -22
[   49.702099] nrc80211 nrc80211: Registered driver = 0
[   49.702099] 
[   49.702672] nrc80211 nrc80211: Successfully registered driver = 0
[   49.702672] 
[   49.703346] nrc80211 nrc80211: start()
[   49.706981] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[   49.707097] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[   49.707103] pgd = ffffff80095c8000
[   49.707119] [00000000] *pgd=00000000bfffe003, *pud=00000000bfffe003, *pmd=0000000000000000
```
Which crashes the kernel.

wzab, 2021-02-02:
Here I attach the log dump from the crash that was generated by the original driver on the standard platform - Raspberry Pi.
It was achieved by loading the driver without connected NRC7292 module.
[p.cap.gz](https://github.com/newracom/nrc7292_sw_pkg/files/5907576/p.cap.gz)

newracom, 2021-02-18:
Thank you for your report and the patch will be applied to this repository soon.
In the meantime, please be advised the following note on this issue.

When insmod command triggered, the function call flow in nrc driver is like below.

```
nrc_platform_probe()
 └ nrc_hif_init()
    ├ nrc_hif_cspi_init() ================> (3) this function cannot have a chance to handle any errors because of the kernel operation.
    │  ├ spi_busnum_to_master()
    │  ├ spi_new_device()
    │  └ spi_register_driver() ===========> (2) kernel ignores internal error from c_spi_probe() and returns 0
    │     └ c_spi_probe() ================> (1) detect error here and return error value.
    └ nrc_hif_start() ====================> (4) execution without any doubt.
       └ spi_start()
          └ spi_update_status() ==========> (5) invalid access error. this is the reason of the kernel crash.
```

So, we can detect the error situation in (1), but the Linux kernel ignores this kind of error internally as follows (drivers/based/dd.c).

```
	switch (ret) {
	case -EPROBE_DEFER:
		/* Driver requested deferred probing */
		dev_dbg(dev, "Driver %s requests probe deferral\n", drv->name);
		driver_deferred_probe_add(dev);
		/* Did a trigger occur while probing? Need to re-trigger if yes */
		if (local_trigger_count != atomic_read(&deferred_trigger_count))
			driver_deferred_probe_trigger();
		break;
	case -ENODEV:
	case -ENXIO:
		pr_debug("%s: probe of %s rejects match %d\n",
			 drv->name, dev_name(dev), ret);
		break;
	default:
		/* driver matched but the probe failed */
		printk(KERN_WARNING
		       "%s: probe of %s failed with error %d\n",
		       drv->name, dev_name(dev), ret);
	}
	/*
	 * Ignore errors returned by ->probe so that the next driver can try
	 * its luck.
	 */
	ret = 0;
```

As a matter of fact, this process should have stopped in (3) with some appropriate error handling from detecting the error between SPI master and slave. So, we modified the error checking logic from the verification of the real SPI connection in c_spi_probe().

If there's an error while probing SPI connection, 'spi' in 'struct nrc_spi_priv' will be updated to NULL and this can be used to detect error later. Please refer to the attached patch file. So, we can prevent the current invalid access error and kernel crash.

And, if you apply the modification of 'start.py' from the diff file, you can stop the execution of 'start.py' instantly when it encounters an error as you mentioned.

[patch.txt](https://github.com/newracom/nrc7292_sw_pkg/files/6005189/patch.txt)

closed 2021-02-18 by newracom