newracom/nrc7292_sw_pkg #8: After reset, it is not clear when it is safe to load the driver

wzab, 2021-02-04:
The driver crashes if the NRC7292 is unresponsive (as reported in https://github.com/newracom/nrc7292_sw_pkg/issues/7 ). However, after reset (enforced by the reset pin of the module) it takes some significant time until the module is ready. Is there any way to detect reliably when the module is ready?
Waiting always 15 seconds after deasserting reset is suboptimal.

newracom, 2021-02-18:
Thank you for your report. Please be advised of the following note on this issue.

The kernel module is ready to download firmware from the host after power-on or reset. And the module cannot operate as a Wi-Fi device yet before downloading firmware.

After downloading the firmware, it is required to enable the wireless interface with the command 'ifconfig up' and configure many things to make it operate as a WLAN device including IP configuration via DHCP.

All of this work is done by 'start.py' and currently, it takes about 15 seconds as you know.
But, registering nrc driver and downloading firmware are finished at the moment of the insmod command.

As per your inquiry, we should look into the kernel logs (driver's log exactly) to check if the module downloads the firmware correctly and is ready to run the firmware. If the firmware downloading has been done successfully, then you can see the logs as below. Please make sure that you add 'NRC_DBG_HIF' into 'DEFAULT_NRC_DBG_MASK' which is defined in nrc-debug.h when building nrc.ko.

```
2021-02-16 10:15:27.8152 kernel:[ 6997.930081] nrc80211 nrc80211: max_speed_hz = 16000000
2021-02-16 10:15:27.8153 kernel:[ 6997.930081]
2021-02-16 10:15:27.8153 kernel:[ 6997.930464] nrc80211 nrc80211: Newracom IEEE802.11 C-SPI: chipid=7292, sw_id=1020716, board_id=5797B0DE
2021-02-16 10:15:27.8153 kernel:[ 6997.930464]
2021-02-16 10:15:27.8194 kernel:[ 6997.930471] nrc80211 nrc80211: Boot loader
2021-02-16 10:15:27.8194 kernel:[ 6997.930471]
2021-02-16 10:15:27.8194 kernel:[ 6997.932514] nrc80211 nrc80211: start()
2021-02-16 10:15:27.8352 kernel:[ 6997.948765] nrc80211 nrc80211: [nrc_check_fw_file, 327] Checking firmware... (uni_s1g.bin)
2021-02-16 10:15:27.8353 kernel:[ 6997.948765]
2021-02-16 10:15:27.8353 kernel:[ 6997.948782] nrc80211 nrc80211: [nrc_check_fw_file, 336] OK...(b11a347c, 0x46958)
2021-02-16 10:15:27.8353 kernel:[ 6997.948782]
2021-02-16 10:15:27.8353 kernel:[ 6997.948790] nrc80211 nrc80211: [spi_check_fw++]
2021-02-16 10:15:27.8353 kernel:[ 6997.948790]
2021-02-16 10:15:27.8353 kernel:[ 6997.948855] nrc80211 nrc80211: [spi_check_fw, 961] sys.sw_id = 0x1020716
2021-02-16 10:15:27.8353 kernel:[ 6997.948855]
2021-02-16 10:15:27.8354 kernel:[ 6997.948860] nrc80211 nrc80211: [spi_check_fw--]
2021-02-16 10:15:27.8354 kernel:[ 6997.948860]
2021-02-16 10:15:27.8354 kernel:[ 6997.948866] nrc80211 nrc80211: FW download....uni_s1g.bin
2021-02-16 10:15:27.8354 kernel:[ 6997.948866]
2021-02-16 10:15:27.8354 kernel:[ 6997.948871] nrc80211 nrc80211: suspend()
2021-02-16 10:15:27.8354 kernel:[ 6997.948877] start FW 287
2021-02-16 10:15:28.1951 kernel:[ 6998.316734] end FW
2021-02-16 10:15:31.2651 kernel:[ 7001.378128] nrc80211 nrc80211: [spi_check_fw++]
2021-02-16 10:15:31.2652 kernel:[ 7001.378128]
2021-02-16 10:15:31.2652 kernel:[ 7001.378198] nrc80211 nrc80211: [spi_check_fw, 961] sys.sw_id = 0x1ec01c8
2021-02-16 10:15:31.2652 kernel:[ 7001.378198]
2021-02-16 10:15:31.2652 kernel:[ 7001.378203] nrc80211 nrc80211: [spi_check_fw--]
2021-02-16 10:15:31.2652 kernel:[ 7001.378203]
2021-02-16 10:15:31.2926 kernel:[ 7001.401430] nrc80211 nrc80211: system ready
2021-02-16 10:15:31.2927 kernel:[ 7001.401444] nrc80211 nrc80211: ready
2021-02-16 10:15:31.2927 kernel:[ 7001.401451] nrc80211 nrc80211:   -- type: 19
2021-02-16 10:15:31.2927 kernel:[ 7001.401457] nrc80211 nrc80211:   -- length: 48
2021-02-16 10:15:31.2927 kernel:[ 7001.401462] nrc80211 nrc80211:   -- version: 0x00000002
2021-02-16 10:15:31.2927 kernel:[ 7001.401468] nrc80211 nrc80211:   -- tx_head_size: 12
2021-02-16 10:15:31.2927 kernel:[ 7001.401473] nrc80211 nrc80211:   -- rx_head_size: 36
2021-02-16 10:15:31.2927 kernel:[ 7001.401477] nrc80211 nrc80211:   -- payload_align: 0
2021-02-16 10:15:31.2927 kernel:[ 7001.401483] nrc80211 nrc80211:   -- cap_mask: 0x39
2021-02-16 10:15:31.2927 kernel:[ 7001.401488] nrc80211 nrc80211:   -- cap_li: 100, 0
2021-02-16 10:15:31.2927 kernel:[ 7001.401493] nrc80211 nrc80211:   -- cap_idle: 100, 0
2021-02-16 10:15:31.2928 kernel:[ 7001.401501] nrc80211 nrc80211:   -- mac_addr[0]: 8c:0f:fa:00:2a:a8
2021-02-16 10:15:31.2928 kernel:[ 7001.401506] nrc80211 nrc80211:   -- mac_addr[1]: 8c:0f:fa:00:51:b8
2021-02-16 10:15:31.2928 kernel:[ 7001.401512] nrc80211 nrc80211:   -- cap_mask[0]: 0x0
2021-02-16 10:15:31.2928 kernel:[ 7001.401517] nrc80211 nrc80211:   -- cap_mask[1]: 0x0
```
This the best way to detect the readiness of the module system so far.

closed 2021-02-18 by newracom

wzab, 2021-02-18:
What happens in case if the module must be reset due to hardware issue?
Should we reload the driver? Up to now, removing the driver without switching off the interface led to crash (maybe it is fixed with patch for issue #7 ).

newracom, 2021-02-19:
> What happens in case if the module must be reset due to hardware issue?
> Should we reload the driver? Up to now, removing the driver without switching off the interface led to crash (maybe it is fixed with patch for issue #7 ).

Could you explain the details of the case you mentioned?
It will be helpful to understand your situation correctly.

wzab, 2021-02-24:
The procedure you propose relies on the output of the driver. However an attempt to lad the driver too soon after deaserting the RESET line, leads to error in communication with the module and kernel crash.
Please try to load the driver immediately after the reset line is deaserted.

newracom, 2021-02-27:
Please let me explain your situation in terms of insmod/rmmod handling procedure of nrc host driver.

1. Run insmod to load the module (driver) into Linux kernel.
   1. Initialize the driver (internal variables, data structures, verifying SPI connection, and so on)
   1. If there’s no problem with step i, then download FW binary from driver to the module via SPI interface
   1. If there’s no problem with step ii, then start FW code in the module
1. Configure module via SPI communication
1. Reset HW while FW is operating: The FW codes on memory will be lost and the bootloader on ROM will start.
1. Wait for the FW downloading from the nrc driver: This is the bootloader operation.
1. Run rmmod: Free the driver resources and remove the driver module from the Linux kernel.
1. Go to step 1 to start the nrc driver and module operation of FW code.

If the rmmod/insmod sequence (step 4-6) didn't work after HW reset, you need to check the module status by using lsmod command. If there’s any problem during rmmod and/or the ‘Used by’ field of nrc module at the result of lsmod shows -1, this means rmmod failed. So you cannot insmod again in this situation.

We’d like to know what kind of problem you encounter and where it is among the steps above.
Please describe the situation in more detail and it will be very helpful if you will provide system logs together.