newracom/nrc7292_sw_pkg #9: MAC frame PV1 of 802.11ah

edgaralb, 2021-02-09:
Dear

How can I configure an NRC7292 from sx-newah, to work with 802.11ah header reduction. Since after capturing with the Sniffer a PV0 frame is displayed and not a PV1.
[image]
Best regards.

newracom, 2021-02-09:
Thank you for your inquiry.

PV1 MAC header was supported in SDK v1.1.0.
But WFA HaLow TG found a PV1 security ambiguity issue in IEEE 802.11ah standard and reported this issue to IEEE 802.11 WG.
And WFA HaLow TG also decided to postpone the PV1 interoperability test until the clarification of this issue.
Since then, PV1 was disabled in the SDK.

If you have any specific requirements for PV1 in your product, please let us know.

closed 2021-02-18 by newracom

modez-s, 2022-02-02:
Hello,
I am currently testing the connection between two of your equipment. Everything is working fine but I would like to test the use of PV1 headers for a research project.
You seem to indicate in the message that it is possible to make particular request concerning this mechanism.
> If you have any specific requirements for PV1 in your product, please let us know.
Is it possible to integrate the PV1 short header mechanism in the SDK 1.3.4 despite the security flaws found ? Would you have a solution to propose me ?
Thank you very much