newracom/nrc7292_sw_pkg #13: error when loading the newracom driver (nrc7292) on nvidia jetson nano developer kit.

Amitrk619, 2021-08-03:
I am triying to port newracom driver (nrc7292) on nvidia jetson nano developer kit. I am able to build the NRC7292 host driver on jetson nano but when trying to load the Newracom NRC7292 kernel module. I get following errors!

https://i.postimg.cc/QtStc8v8/Screenshot-from-2021-08-03-12-19-12.png

gtnano2@gtnano2:~/nrc7292_sw_pkg/package/host/nrc_driver/source/nrc_driver/nrc$ dmesg

[   86.621601] nrc: loading out-of-tree module taints kernel.
[   86.644299] spi-tegra114 7000d400.spi: chipselect 0 already in use
[   86.650508] failed to initialize hif
[   86.653917] nrc80211: probe of nrc80211 failed with error -22

gtnano2@gtnano2:~/nrc7292_sw_pkg/package/host/nrc_driver/source/nrc_driver/nrc$ 

gtnano2@gtnano2:~/nrc7292_sw_pkg$ uname -a
Linux gtnano2 4.9.201-tegra #1 SMP PREEMPT Fri Feb 19 08:40:32 PST 2021 aarch64 aarch64 aarch64 GNU/Linux

 I am using the silex SX-newah 802.11ah module.

@newracom @droidifi @wzab``

droidifi, 2021-08-03:
The DTS file probably needs to be modified. 

The Jetson SPI interface must run as a SPI slave instead of a SPI master.

Amitrk619, 2021-08-04:
what changes do I have to make in the DTS file and what does this error means " nrc80211: probe of nrc80211 failed with error -22 "

> The DTS file probably needs to be modified.
> 
> The Jetson SPI interface must run as a SPI slave instead of a SPI master.

newracom, 2021-08-04:
Dear Amitrk619,

Thank you for your asking.
Unfortunately, NRC7292 operates as an SPI slave and only supports this.
So SPI of the host platform should operate as a master.
The current host driver doesn't support slave mode and we don't have a plan to support this at the moment.

Best regards,
Newracom

Amitrk619, 2021-08-04:
From what I know the host platform jetson nano operate as a spi master.

newracom, 2021-08-05:
Dear Amitrk619,

When the bus number of the SPI master is 0, the SPI device using chipselect 0 can be found by opening the /sys/class/spi_master/spi0/spi0.0/modalias file.
 
	$ cat /sys/class/spi_master/spi0/spi0.0/modalias

If the spidev driver is using chipselect 0, the modalias is “spi:spidev”.
But the modalias of the nrc7292 driver is “spi:nrc-cspi”.

If you need to use chipselect 0 for the nrc7292, you should remove the SPI device using chipselect 0 from the kernel. 

If you can rebuild the kernel, you should disable it in the kernel configuration menu (menuconfig).
The menu path for the spidev driver is “Device Drivers -> SPI support ->User mode SPI device driver support”.
Alternatively, if your kernel supports the device tree, it may be possible by modifying the device tree source file.

If you cannot rebuild the kernel and the SPI device driver was already built as a module, the driver module should not be loaded into the kernel.
If the spidev driver is using chipselect 0, you will be able to disable it with one of the following:

	- Kernel Module Blacklisting
		$ echo "blacklist spidev" >> /etc/modprobe.d/blacklist-spidev.conf

	- Module Renaming
		$ cd /lib/modules/${uname -r}/kernel/drivers/spi
		$ sudo mv spidev.ko spidev.ko~

	- “rmmod” command
		$ sudo rmmod spidev

	- Device Tree Overlay

Best regards,
Newracom

closed 2021-08-30 by newracom