newracom/nrc7292_sw_pkg #15: Hostapd + nl80211: Could not configure driver mode

modez-s, 2021-10-30:
Hello Newracom,
First of all, I would like to thank you for the excellent work you have done from a technical point of view but also for the documentation.
I followed your GitHub tutorial to install the NRC7292 driver on a Raspberry Pi 3B+ board in Host mode. When I run the script to start an Access Point `./start 1 0 EU`, I get the following error :

```
[6] Start hostapd on wlan0
Configuration file: /home/pi/nrc_pkg/script/conf/EU/ap_halow_open.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
```
```
$ sudo systemctl status hostapd.service
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/lib/systemd/system/hostapd.service; enabled; vendor preset: enabled)
   Active: activating (start) since Sat 2021-10-30 13:54:51 CEST; 18ms ago
Cntrl PID: 1611
   CGroup: /system.slice/hostapd.service

oct. 30 13:54:51 raspberrypi systemd[1]: Starting Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticato
oct. 30 13:54:51 raspberrypi hostapd[1611]: Configuration file:
oct. 30 13:54:51 raspberrypi hostapd[1611]: Could not open configuration file '' for reading.
oct. 30 13:54:51 raspberrypi hostapd[1611]: Failed to set up interface with
oct. 30 13:54:51 raspberrypi hostapd[1611]: Failed to initialize interface
oct. 30 13:54:51 raspberrypi systemd[1]: hostapd.service: Control process exited, code=exited status=1
oct. 30 13:54:51 raspberrypi systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authe
```

I checked that the SPI link and the drivers were correctly used :
```
$ lsmod
nrc                   192512  0
mac80211              659456  1 nrc
cfg80211              573440  2 mac80211,nrc
```
```
$ cat /sys/class/spi_master/spi0/spi0.0/modalias 
spi:nrc-cspi
```
but I can't fix this problem and start the hostapd service correctly.
Have you already experienced this type of problem and would you have a solution for me?
Respectful greetings

newracom, 2021-11-01:
Dear @modez-s,

Based on your test command and its result below, 
it seems that you started hostapd as a daemon and failed to initialize it. 

***
oct. 30 13:54:51 raspberrypi hostapd[1611]: Could not open configuration file '' for reading.
oct. 30 13:54:51 raspberrypi hostapd[1611]: Failed to set up interface with
oct. 30 13:54:51 raspberrypi hostapd[1611]: Failed to initialize interface
***

Our start.py script has its own way of starting hostapd inside.
Please refer to the attached a sample host log: [v1.3.4_sample_host_log.txt](https://github.com/newracom/nrc7292_sw_pkg/files/7455100/v1.3.4_sample_host_log.txt)

The daemon startup requires a proper conf file in /etc/hostapd directory.
Could you share your conf file?

And please let us know your EVK information.

Best regards,
Newracom

modez-s, 2021-11-02:
Hello @newracom 
The solution you proposed works.
I had tried to implement the 1.3.3 solution at first, before switching last week to the 1.3.4 version. So I didn't take the time to read the documentation UG-7292-018-Raspberry_Pi_setup.pdf.
Thanks for your help

closed 2021-11-02 by modez-s