newracom/nrc7292_sw_pkg #20: Undefined 4 bytes before ATComm?

Dromopoda, 2021-11-30:
Hi everyone,

I've been using the nrc7292 with raspberry pi 3 b+, and I've noticed using a logic analyzer, that the raspberry (and also the module) Send 4 bytes before an AT Command, besides the command period of 48 bits. 

I've realized that it sends "HS" (In ascii) and the length of the AT command, plus another byte of a value in hex being incremented for each command I've send, but I don't know what is this for, or what is this pointing to. Taking care of that the raspberry is constantly resetting the irq, (reading the 12th register) and those "actions" does sum to the value incrementing, I discard the idea in where the 4th byte Is some sort of "slot pointer" or something like that

So... I don't know what is "HS" and why is it sending it, and why does it send the length 2 times (1 in the command period and other in the AT command) and what is that 4th byte, why does it increment and why isn't this information in the documentations.

I hope someone could help me,

Best regards, 

Dromopoda.

Dromopoda, 2021-11-30:
[image]

I add this foto for those who didn't understand.
This is the log of the logic analyzer.

nrc-sbkim, 2021-12-02:
Dear @Dromopoda,

If you have seen the following user guide, 

[https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/UG-7292-002-Host%20driver%20porting.pdf](url)
	
it's for the nrc7292 driver running on a Linux host.

[https://github.com/newracom/nrc7292_sw_pkg/tree/master/package/host/src/nrc](url)

Please see only to the following chapters for AT Command.

	chapter 4. Host SPI
	chapter 8. HSPI Register map

You can find the raspi-atcmd-cli application with the following files.

[https://github.com/newracom/nrc7292_sdk/blob/master/package/standalone/sdk/apps/atcmd/host/raspi-atcmd-cli/nrc-hspi.c](url)
[https://github.com/newracom/nrc7292_sdk/blob/master/package/standalone/sdk/apps/atcmd/host/raspi-atcmd-cli/nrc-hspi.h](url)

These files are needed to communicate via SPI with the NRC7292 running AT Command firmware.
You will be able to implement the host application by including these files with little or no modification. 

AT Command firmware has slot memory queues for TX and RX.
The slot size is 512 bytes and the queue depth is 32.
You can see it in the firmware log.

[image]

The slot information for the current firmware is checked by hspi_read() in nrc-hspi.c. 

A slot has a 4-byte header.
The len field has the length of valid data.
The seq field is incremented by 1 each time data is written to the slot and reset from 63 to 0.

[image]

[image]

The maximum length of data that can be written to a slot is 508 bytes.
Even if the length of the data is less than 508 bytes, the host must read or write in units of 512 bytes because the slot size is 512 bytes.

[image]

The HSPI_EIRQ interrupt is not used for AT Command.
So, the following registers should be read to update the available slot count before a read or write operation.

[image]

You can see it in hspi_read() and hspi_write() in nrc-hspi.c.

[image]

[image]

Please also refer to the following logs.

[image]

Best regards,
@Newracom

Dromopoda, 2021-12-02:
Hi @nrc-sbkim,

Thanks for the answer...

U've clarified some things for me, but I still don't get some...

I'm doing a little CLI App and If I send in the seq, a value being incremented by 1 for each write transaction, I only get the answer from the chip every 4 writted command... (i.e. seq: 0, 4, 8 and so on) but if I left the seq byte on 0 I do get answer every command I send...

I still don't get the usage of the RX and TX queue status registers, When I read TXQ status reg. I do get some like this:
[image]

In the last image u sended, what is the "read/write slot" value? I don't understand the usage of that value, sorry If I'm kinda fool.

The last issue that I'm confronting (at least for now) is that I don't know how to handle the AT+WSCAN command. I did try reading consecutively after sending the command but I don't get any response. I also did try to wait some seconds and then read the register but still the same... you may guide me here.

Dromopoda, 2021-12-02:
I've also found that I must send 0x20 (" " backspace, in ascii) before the CRLF (i.e. AT= 41 54 20 0D 0A) The length should be...? 5?

I see that you put 4 in the last image u send, but I tried and works only If I put 5 in length.

nrc-sbkim, 2021-12-02:
Dear @Dromopoda,

If the CLI App you are testing is not the raspi-atcmd-cli App we provide, I can't figure out the cause of your issues. 
Your issues do not appear in the raspi-atcmd-cli App.

The TX queue is for sending data to the host, and the RX queue is for receiving data from the host.
When sending data from the host to the firmware, the RX QUEUE status registers should be read. 

[image]

The following logs are output from hspi_read_slot() and hspi_write_slot() in nrc-hspi.c. 

[image]

Please refer to the following files for how to handle AT commands.

https://github.com/newracom/nrc7292_sdk/blob/master/package/standalone/sdk/apps/atcmd/host/raspi-atcmd-cli/nrc-atcmd.c
https://github.com/newracom/nrc7292_sdk/blob/master/package/standalone/sdk/apps/atcmd/host/raspi-atcmd-cli/nrc-atcmd.h

Best regards,
@newracom

Dromopoda, 2021-12-02:
Hi again @nrc-sbkim,

I'm using the cli app and trying to understand it to make one that fulfill my needs, cause I need an app more specific.

I know the fact that the TXQ reg is where to find the answer from the halow module, and RXQ reg where to put the AT comm, but I don't know how to interpret the values of those registers... each bit is a slot or are values in decimal, or what?

nrc-sbkim, 2021-12-03:
Dear @Dromopoda,

Please see to the following functions in nrc-hspi.c.

[image]

[image]

The hspi_status_t struct is in a previous comment.
And you can find it in nrc-hspi.h. 

Best regards,
@newracom

closed 2021-12-03 by Dromopoda