newracom/nrc7292_sw_pkg #22: Bugs related to print function in start.py

huhanwj, 2021-12-20:
Hi everyone,

I am trying to start an AP on my Raspberry Pi 4B with and I get something like this

```bash
pi@raspberrypi:~/nrc_pkg/script $ ./start.py 1 0 EU
  File "/home/pi/nrc_pkg/script/./start.py", line 351
    print "country: " + country +", prefer_bw: " + str(prefer_bw)+ ", dwell_time: " + str(dwell_time)
          ^
SyntaxError: invalid syntax
pi@raspberrypi:~/nrc_pkg/script $ vi start.py 
pi@raspberrypi:~/nrc_pkg/script $ ./start.py 1 0 EU
  File "/home/pi/nrc_pkg/script/./start.py", line 624
    print "[*] Self configuration start!"
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("[*] Self configuration start!")?

```
I'm running Python 3.9.2 on my Pi and I am not sure what caused this and whether it is the version that made this bug as I just allow the system to do the update by simply execute `sudo apt update`. I also do not follow closely on what Python updates from 3.7 to 3.8 to 3.9.

I believe that this is not a fatal bug. Just report to you guys and hope it will get fixed for better user experience.

newracom, 2021-12-20:
Dear @huhanwj,

I'm afraid you're using old version package. 
Could you provide me the result of following commands?

(1) cat /etc/os-release
(2) uname -a
(3) cat ~/nrc_pkg/VERSION-SDK.txt

Best regards,
Newracom

newracom, 2021-12-21:
Dear @huhanwj.

The recent commit, https://github.com/newracom/nrc7292_sw_pkg/commit/961bacbda2cdc2cf2be7027a2d4c302a9cd0e52f, includes the bug fix for this issue.
Thank you again for your report.

Best regards,
Newracom

closed 2021-12-21 by newracom