newracom/nrc7292_sw_pkg #23: Segmentation fault when trying to load nrc.ko

huhanwj, 2021-12-21:
Hi, me again.

Trying to use *start.py* to create an AP and this time I got segmentation fault during sudo insmod progress.

I am using a Raspberry Pi 4B

`uname -r` gives:

`5.10.63-v7l+`

Driver info:

```bash
pi@raspberrypi:~ $ modinfo /home/pi/nrc_pkg/sw/driver/nrc.ko 
filename:       /home/pi/nrc_pkg/sw/driver/nrc.ko
description:    Newracom 802.11 driver
license:        Dual BSD/GPL
author:         Newracom, Inc.(http://www.newracom.com)
srcversion:     6D4D1721FD73A419C8AA3D5
depends:        mac80211,cfg80211
name:           nrc
vermagic:       5.10.63-v7l+ SMP mod_unload modversions ARMv7 p2v8 
parm:           fw_name:Firmware file name (charp)
parm:           bd_name:Board Data file name (charp)
parm:           hifport:HIF port device name (charp)
parm:           hifspeed:HIF port speed (int)
parm:           spi_bus_num:SPI controller bus number (int)
parm:           spi_cs_num:SPI chip select number (int)
parm:           spi_gpio_irq:SPI gpio irq (int)
parm:           spi_gpio_poll:SPI gpio polling interval (msec) (int)
parm:           spi_gdma_irq:SPI gdma irq (int)
parm:           loopback:HIF loopback (bool)
parm:           lb_count:HIF loopback Buffer count (int)
parm:           disable_cqm:Disable CQM (0: enable, 1: disable) (int)
parm:           listen_interval:Listen Interval (int)
parm:           bss_max_idle:BSS Max Idle (int)
parm:           bss_max_idle_usf_format:BSS Max Idle specified in units of usf (bool)
parm:           enable_short_bi:Enable Short BI (bool)
parm:           enable_monitor:Enable Monitor (bool)
parm:           bss_max_idle_offset:BSS Max Idle Offset (int)
parm:           macaddr:MAC Address (charp)
parm:           power_save:power save (int)
parm:           sleep_duration:deepsleep duration of non-TIM mode power save (array of int)
parm:           wlantest:wlantest (bool)
parm:           ndp_preq:Enable NDP Probe Request (bool)
parm:           ndp_ack_1m:Enable 1M NDP ACK (bool)
parm:           enable_hspi_init:Enable HSPI Initialization (bool)
parm:           nullfunc_enable:Enable null func on mac80211 (bool)
parm:           auto_ba:Enable auto ba session setup on connection / QoS data Tx (bool)
parm:           sw_enc:Use SW Encryption instead of HW Encryption (bool)
parm:           signal_monitor:Enable SIGNAL(RSSI/SNR) Monitor (bool)
parm:           enable_usn:Use configuration of KR USN (Same ac between data and beacon) (bool)
parm:           debug_level_all:Driver debug level all (bool)
parm:           credit_ac_be:credit number for AC_BE (int)

```

syslog after `sudo insmod /home/pi/nrc_pkg/sw/driver/nrc.ko` (This gives an output as Segmentation Fault):

```bash
[   62.803181] nrc: loading out-of-tree module taints kernel.
[   62.814816] ------------[ cut here ]------------
[   62.814940] WARNING: CPU: 2 PID: 1074 at /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-hif-cspi.c:347 _c_spi_read_regs+0x200/0x220 [nrc]
[   62.814956] Modules linked in: nrc(O+) mac80211 libarc4 snd_soc_hdmi_codec 8021q garp stp llc joydev v3d gpu_sched brcmfmac brcmutil sha256_generic raspberrypi_hwmon cfg80211 rfkill i2c_brcmstb vc4 cec i2c_bcm2835 bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_isp(C) videobuf2_vmalloc bcm2835_mmal_vchiq(C) vc_sm_cma(C) videobuf2_dma_contig spi_bcm2835 videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc snd_bcm2835(C) drm_kms_helper snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer rpivid_mem snd syscopyarea sysfillrect sysimgblt fb_sys_fops w1_gpio wire cn uio_pdrv_genirq uio nvmem_rmem i2c_dev drm drm_panel_orientation_quirks backlight fuse ip_tables x_tables ipv6
[   62.816155] CPU: 2 PID: 1074 Comm: insmod Tainted: G         C O      5.10.63-v7l+ #1488
[   62.816165] Hardware name: BCM2711
[   62.816175] Backtrace: 
[   62.816210] [<c0b83534>] (dump_backtrace) from [<c0b838c8>] (show_stack+0x20/0x24)
[   62.816226]  r7:ffffffff r6:00000000 r5:60000013 r4:c12e6b3c
[   62.816245] [<c0b838a8>] (show_stack) from [<c0b87cb4>] (dump_stack+0xcc/0xf8)
[... 194 lines cut ...]
[   62.823072] 7f00: c54dd600 c567db54 c4d97f3c c4d97f18 bf4c7054 bf4c6d0c bf4c704c bf4c7038
[   62.823089] 7f20: c529e300 c4d96000 c56c17a0 00000000 c4d97f54 c4d97f40 c0b93864 c032f004
[   62.823106] 7f40: ffffe000 4f649745 c4d97f74 c529e300 c529ee80 00000000 c4d96000 bf4be7ac
[   62.823123] 7f60: c54dd600 c567db54 c4d97fac c4d97f78 c02459f8 bf4be7b8 c529e324 c529e324
[   62.823140] 7f80: ffffe000 c529ee80 c0245888 00000000 00000000 00000000 00000000 00000000
[   62.823157] 7fa0: 00000000 c4d97fb0 c02000ec c0245894 00000000 00000000 00000000 00000000
[   62.823175] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   62.823191] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[   62.823218] Backtrace: 
[   62.823350] [<bf4be7ac>] (spi_rx_thread [nrc]) from [<c02459f8>] (kthread+0x170/0x174)
[   62.823373]  r10:c567db54 r9:c54dd600 r8:bf4be7ac r7:c4d96000 r6:00000000 r5:c529ee80
[   62.823386]  r4:c529e300
[   62.823408] [<c0245888>] (kthread) from [<c02000ec>] (ret_from_fork+0x14/0x28)
[   62.823432] Exception stack(0xc4d97fb0 to 0xc4d97ff8)
[   62.823455] 7fa0:                                     00000000 00000000 00000000 00000000
[   62.823478] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   62.823494] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   62.823513]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0245888
[   62.823533]  r4:c529ee80
[   62.823562] Code: 1a00003d e51b3048 e1a01101 e3a02ea2 (e5935040) 
[   62.823584] ---[ end trace b1e1c50376dc55a2 ]---

```

Any suggestions?

newracom, 2021-12-21:
Dear @huhanwj,

Did you build your own nrc.ko with https://github.com/newracom/nrc7292_sw_pkg/tree/master/package/host/src/nrc?
The nrc.ko binary of https://github.com/newracom/nrc7292_sw_pkg/tree/master/package/host/evk/sw_pkg/nrc_pkg/sw/driver is for Linux kernel 4.14.70 as mentioned in the README.txt.
If you used nrc.ko binary directly, please build your own and try again.

Best regards,
Newracom

huhanwj, 2021-12-22:
> Dear @huhanwj,
>
> Did you build your own nrc.ko with https://github.com/newracom/nrc7292_sw_pkg/tree/master/package/host/src/nrc? The nrc.ko binary of [...]

Yes, I do build my own nrc.ko otherwise it will not have `vermagic:       5.10.63-v7l+ SMP mod_unload modversions ARMv7 p2v8 ` in modinfo. However, I build the host driver without building firmware and the cli_app, does this matters?

newracom, 2021-12-22:
Dear @huhanwj,

You don't need to touch firmware and cli_app. I'm afraid there's something missing on your RPi4.
Could you check your RPi4 with https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/UG-7292-018-Raspberry_Pi_setup.pdf?

Best regards,
Newracom

huhanwj, 2021-12-23:
> Dear @huhanwj,
> 
> You don't need to touch firmware and cli_app. I'm afraid there's something missing on your RPi4. Could you check your RPi4 with https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/UG-7292-018-Raspberry_Pi_setup.pdf?
> 
> Best regards, Newracom

Thanks, I seemed missing this document. I will check this out and follow up here once I can get it to work or not.

huhanwj, 2021-12-23:
> Dear @huhanwj,
> 
> You don't need to touch firmware and cli_app. I'm afraid there's something missing on your RPi4. Could you check your RPi4 with https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/UG-7292-018-Raspberry_Pi_setup.pdf?
> 
> Best regards, Newracom

Found missing disabling broadcom wifi and user mode SPI driver. However after full reinstallation, still get segmentation fault when doing `sudo insmod`.

Should I return to an older RPi OS version and have a try?

huhanwj, 2021-12-23:
Just switch to A ready-to-go SD card image for a Raspberry Pi model 3 or model 4 from https://github.com/droidifi/newracom-s1g

Loading nrc.ko will gives

```
nrc-cspi: probe of spi0.0 failed with error -22
failed to initialize hif
nrc80211: probe of nrc80211 failed with error -22

```
in `dmesg`

Is this problem related to the ALFA card?

Will have a try on 3B+ on next step.

enganguish, 2021-12-23:
@huhanwj How did you use 'insmod' exactly? Can you share your cli when you have done?

If you have just entered only 'sudo insmod /home/pi/nrc_pkg/sw/driver/nrc.ko' which was mentioned above, the module loading would be fail. Because some essential module parameters were missed.
(It will be helpful,  [https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/UG-7292-002-Host%20driver%20porting.pdf] - Appendix A)

Or, you may easily catch  what's necessary from /home/pi/nrc_pkg/script/start.py.
(please look for 'insmod' in start.py.)

huhanwj, 2021-12-23:
> 

I was using the start.py and it returns a segmentation fault. 

I believe that the key part of this bug is related to `[   62.818737] nrc-cspi: probe of spi0.0 failed with error -22` this probe error, but I have no idea how it was caused.

enganguish, 2021-12-27:
@huhanwj 

I have a question. Does your [AHPI7292S](https://www.alfa.com.tw/products/ahpi7292s?variant=36473961283656) have DIP switch for the mode selection? In the link of ALFA, it seems to be shown empty on that DIP switch position of the module PCB.

huhanwj, 2021-12-27:
> @huhanwj
> 
> I have a question. Does your [AHPI7292S](https://www.alfa.com.tw/products/ahpi7292s?variant=36473961283656) have DIP switch for the mode selection? In the link of ALFA, it seems to be shown empty on that DIP switch position of the module PCB.

No, it's empty there. However, it does have a mode switch for you to switch among Standalone, Download and Host mode.

enganguish, 2021-12-27:
> No, it's empty there. However, it does have a mode switch for you to switch among Standalone, Download and Host mode.

Then, Did you configure your AHPI7292S with 'Host mode'? This is necessary...

enganguish, 2021-12-27:
> No, it's empty there. However, it does have a mode switch for you to switch among Standalone, Download and Host mode.

And then, the switch was 'Host mode' already. right?

huhanwj, 2021-12-27:
> And then, the switch was 'Host mode' already. right?

Yes, the default setting is host mode

huhanwj, 2021-12-27:
Just tested with RPi 3B+ with kernel version 5.10.63-v7+. Everything is fine (no bug when executing `insmod`) and hostapd gives successful message when setting up the AP. I haven't tested the STA/Client as I only bring one 3B+ from the lab back home during Christmas. I will further test it tomorrow.

huhanwj, 2021-12-28:
> Just tested with RPi 3B+ with kernel version 5.10.63-v7+. Everything is fine (no bug when executing `insmod`) and hostapd gives successful message when setting up the AP. I haven't tested the connections as I only bring one 3B+ from the lab back home during Christmas. I will further test it tomorrow.

Update: Successfully connection between AP-STA when both sides are using 3B+ as host machine.

So, I believe that my board is somehow incompatible with RPi 4B or there are some errors in the spi communication between the board and the RPi 4B's SoC.

newracom, 2022-01-05:
Dear @huhanwj,

If you don't have any further question or issue, I'd like to close this issue.
Please leave your opinion.

Best regards,
Newracom

closed 2022-01-06 by huhanwj