newracom/nrc7292_sw_pkg #24: How to change from  -bw :2 Mhz (NRC Auto)

numbatboy, 2022-01-20:
Hi all,

How can I change bandwith to 1 or 4 MHz?  Defaults to 2 MHz?

Setup: 2x Raspberry Pi 3 B+ and 2x AHPI7292S from ALFA Network Inc. Operating in host mode 
with start.py - AP and STA. Hoping someone can help.

Thanks,
Tristan

newracom, 2022-01-20:
Dear @numbatboy,

Please refer to chapter 6 of https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/UG-7292-001-EVK%20User%20Guide%20(Host%20Mode).pdf.

Best regards,
Newracom

newracom, 2022-01-22:
Dear @numbatboy,

I'd like to close this issue next week if your issue was resolved.

Best regards,
Newracom

numbatboy, 2022-01-22:
Dear @newracom,

Chapter 6, set appropriate channel in US/ap_halow_open.conf file.
RTM - everything working well now - problem solved.

Thank you very much,
numbatboy

closed 2022-01-24 by newracom