newracom/nrc7292_sw_pkg #26: PV1 Headers

modez-s, 2022-02-02:
Hello Newracom,
I am currently testing the connection between two of your equipment. Everything is working fine but I would like to test the use of PV1 headers for a research project. You seem to indicate in a message that it is possible to make particular request concerning this mechanism.

> Thank you for your inquiry.
> PV1 MAC header was supported in SDK v1.1.0.
> But WFA HaLow TG found a PV1 security ambiguity issue in IEEE 802.11ah standard and reported this issue to IEEE 802.11 WG.
> And WFA HaLow TG also decided to postpone the PV1 interoperability test until the clarification of this issue.
> Since then, PV1 was disabled in the SDK.
> If you have any specific requirements for PV1 in your product, please let us know.

Is it possible to integrate the PV1 short header mechanism in the SDK 1.3.4 despite the security flaws found ? Would you have a solution to propose me ?
Thank you very much

newracom, 2022-02-02:
Dear @modez-s,

Thank you for your inquiry about PV1 frame support.
As you know, PV1 is excluded from the HaLow R1 certification program due to the IEEE 802.11ah standard issue.
And there's been no progress on that issue, so HaLow TG didn't include PV1 as candidate feature of HaLow R2 program.
Under these circumstances, we have no plan to support PV1 at the moment.
If you have any PV1-specific requirement or use case, please let us know.

Best regards,
Newracom

closed 2022-02-08 by newracom

modez-s, 2022-02-08:
Thank you for your answer
Best regards