newracom/nrc7292_sw_pkg #28: Can't get NRC7292 working on Pi3

Ankush-Pathak, 2022-02-15:
I am trying to get the module up and running on a Pi 3. I get an `Oops` on `insmod`. I am attaching the kernel trace. Could you please check and see if you can help me out?  
[kernel_trace.txt](https://github.com/newracom/nrc7292_sw_pkg/files/8074265/kernel_trace.txt)

newracom, 2022-02-15:
Dear @Ankush-Pathak,

Thank you for your question.
Could you check this user guide first?
https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/UG-7292-018-Raspberry_Pi_setup.pdf

Best regards,
Newracom

Ankush-Pathak, 2022-02-15:
Yes, I went through this, and I am doing everything exactly as it says.

enganguish, 2022-02-17:
@Ankush-Pathak ,

I have some questions.

- What type of product do you have? can you tell me the model name of your NRC7292 product? Is it an EVK of NRC7292 or other module?
- What module parameters did you use when insmod?
- Have you checked [UG-7292-002-Host driver porting.pdf](https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/UG-7292-002-Host%20driver%20porting.pdf) and [UG-7292-001-EVK User Guide (Host Mode).pdf](https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/UG-7292-001-EVK%20User%20Guide%20(Host%20Mode).pdf)?

newracom, 2022-02-17:
Dear @Ankush-Pathak,

Could you update dtoverlay definitions in /boot/config.txt as below and try again?
[image]
The prefix "pi3-" is required for Pi3.
I'm afraid our user guide missed that point. We will update the guide doc.
Thank you for your report.

Best regards,
Newracom

newracom, 2022-02-24:
Dear @Ankush-Pathak,

Did you try /boot/config.txt update?
If you tried and solved your problem, I'd like to close this issue.

Best regards,
Newracom

newracom, 2022-03-11:
Dear @Ankush-Pathak,

Per the disabled onboard Wi-Fi, there's no special reason for that.
The RPi is used for the host platform for HaLow, so we disabled all other wireless interfaces, Wi-Fi and bluetooth.
If you want, you can use them by updating /boot/config.txt and /etc/modprobe.d/raspi-blacklist.conf.

And please let me know and close this issue when you verify the resolution I mentioned above.

Best regards,
Newracom

closed 2022-08-11 by newracom