newracom/nrc7292_sw_pkg #29: Are there any examples for mesh networking?

synaption, 2022-03-03:
I notice mesh is listed as a feature added is the release description.  I also found mesh.py, but it's not immediately obvious to me how to use it.  Are there any examples of using the mesh capabilities of the module?

newracom, 2022-03-04:
Dear @synaption,

Thank you for your question.
The application note for mesh network is provided here, https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/AN-7292-007-11s_mesh_network.pdf

Best regards,
Newracom

newracom, 2022-03-08:
Dear @synaption,

If you find mesh information on application note above and there's no further question, I'd like to close this issue.

Best regards,
Newracom

synaption, 2022-03-08:
You can close it.  I have not tried it yet, but it looks like everything I need.  Thanks.