newracom/nrc7292_sw_pkg #33: cannot compile against >5.15

ZeroChaos-, 2022-05-09:
https://github.com/Gateworks/nrc7292/issues/1 has additional details, but I get this built failure on kernel >5.15
```
  if x86_64-pc-linux-gnu-objdump -h /var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/nrc-hif-uart.o | grep -q __ksymtab; then x86_64-pc-linux-gnu-gcc -E -D__GENKSYMS__ -Wp,-MMD,/var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/.nrc-hif-uart.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -DCONFIG_X86_X32_ABI -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=1024 -fstack-protector-strong -Wimplicit-fallthrough=5 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -fomit-frame-pointer -fno-stack-clash-protection -fzero-call-used-regs=used-gpr -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-truncation -Wno-zero-length-bounds -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -Wno-alloc-size-larger-than -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -fplugin=./scripts/gcc-plugins/structleak_plugin.so -fplugin=./scripts/gcc-plugins/randomize_layout_plugin.so -fplugin=./scripts/gcc-plugins/stackleak_plugin.so -fplugin-arg-structleak_plugin-byref-all -DSTRUCTLEAK_PLUGIN -DRANDSTRUCT_PLUGIN -DSTACKLEAK_PLUGIN -fplugin-arg-stackleak_plugin-track-min-size=100 -fplugin-arg-stackleak_plugin-arch=x86 -DDEBUG -DBUILD_UMAC_DRIVER -DCONFIG_NRC_HIF_CSPI -DENABLE_DYNAMIC_PS -Wall  -DMODULE  -DKBUILD_BASENAME='"nrc_hif_uart"' -DKBUILD_MODNAME='"nrc"' -D__KBUILD_MODNAME=kmod_nrc /var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/nrc-hif-uart.c | scripts/genksyms/genksyms    -r /dev/null > /var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/.tmp_nrc-hif-uart.ver; x86_64-pc-linux-gnu-ld -m elf_x86_64 -r -o /var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/.tmp_nrc-hif-uart.o /var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/nrc-hif-uart.o -T /var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/.tmp_nrc-hif-uart.ver; mv -f /var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/.tmp_nrc-hif-uart.o /var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/nrc-hif-uart.o; rm -f /var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/.tmp_nrc-hif-uart.ver; fi
/var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/nrc-ssp.c: In function ‘nrc_ssp_register’:
/var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/nrc-ssp.c:591:18: error: implicit declaration of function ‘spi_busnum_to_master’ [-Werror=implicit-function-declaration]
  591 |         master = spi_busnum_to_master(spi_bus_num);
      |                  ^~~~~~~~~~~~~~~~~~~~
/var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/nrc-ssp.c:591:16: warning: assignment to ‘struct spi_controller *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
  591 |         master = spi_busnum_to_master(spi_bus_num);
      |                ^
  x86_64-pc-linux-gnu-gcc -Wp,-MMD,/var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/.nrc-hif-cspi.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -DCONFIG_X86_X32_ABI -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=1024 -fstack-protector-strong -Wimplicit-fallthrough=5 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -fomit-frame-pointer -fno-stack-clash-protection -fzero-call-used-regs=used-gpr -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-truncation -Wno-zero-length-bounds -Wno-array-bounds -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -Wno-alloc-size-larger-than -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -fplugin=./scripts/gcc-plugins/structleak_plugin.so -fplugin=./scripts/gcc-plugins/randomize_layout_plugin.so -fplugin=./scripts/gcc-plugins/stackleak_plugin.so -fplugin-arg-structleak_plugin-byref-all -DSTRUCTLEAK_PLUGIN -DRANDSTRUCT_PLUGIN -DSTACKLEAK_PLUGIN -fplugin-arg-stackleak_plugin-track-min-size=100 -fplugin-arg-stackleak_plugin-arch=x86 -DDEBUG -DBUILD_UMAC_DRIVER -DCONFIG_NRC_HIF_CSPI -DENABLE_DYNAMIC_PS -Wall  -DMODULE  -DKBUILD_BASENAME='"nrc_hif_cspi"' -DKBUILD_MODNAME='"nrc"' -D__KBUILD_MODNAME=kmod_nrc -c -o /var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/nrc-hif-cspi.o /var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/nrc-hif-cspi.c  ; ./tools/objtool/objtool orc generate  --module  --no-fp   --retpoline  --uaccess   /var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/nrc-hif-cspi.o
   { echo ; echo '/var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/nrc-hif-ssp.o: $(wildcard ./tools/objtool/objtool)' ; } >> /var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/.nrc-hif-ssp.o.cmd
  if x86_64-pc-linux-gnu-objdump -h /var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/nrc-hif-ssp.o | grep -q __ksymtab; then x86_64-pc-linux-gnu-gcc -E -D__GENKSYMS__ -Wp,-MMD,/var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/.nrc-hif-ssp.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -DCONFIG_X86_X32_ABI -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk [... cut ...]
/var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/nrc-hif-cspi.c:2123:16: warning: assignment to ‘struct spi_controller *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 2123 |         master = spi_busnum_to_master(spi_bus_num);
      |                ^
   { echo ; echo '/var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/nrc-mac80211.o: $(wildcard ./tools/objtool/objtool)' ; } >> /var/tmp/portage/net-wireless/nrc7292-9999/work/nrc7292-9999/package/host/src/nrc/.nrc-mac80211.o.cmd
```

newracom, 2022-05-20:
Dear @ZeroChaos- 

Thank you for your inquiry.
Unfortunately, NRC host driver supports Kernel version up to 5.15.
Please share your product information and we will take your system requirements into our SW on the next release.

Best regards,
Newracom

jrddunbr, 2022-07-10:
Unfortunately, even on Linux 5.15, there is an issue where some Linux distributions don't have the deprecated `CONFIG_GPIO_SYSFS` option set to `y`, which means that it doesn't even work on that kernel version either for those operating systems because the `/sys/class/gpio` interface is not in the kernel builds. This means that one would need to build their own kernel with the support for that interface, or use an even older LTS kernel that isn't in the official package repositories.

Tested on Arch Linux using `linux-lts` (Linux version 5.15.53), it does not include the above config since the `linux-lts` package was on Linux 5.10 back in January of this year at the latest.

https://archlinux.org/packages/core/x86_64/linux-lts/

All Linux kernel development moved to using the new character buffer based GPIO functionality back in 2015 when they deprecated the sysfs-based interface. Please update your driver to use the new interface and/or support newer kernels.

jrddunbr, 2022-07-10:
For what it's worth, my frustration with this platform is making me really want to use other 802.11ah products on the market; this device is too hard to get working for anyone that has but the most advanced Linux kernel skillset. I've spent over 5 days on this project compiling your source code and the Linux kernel in several different ways, and I still cannot get your simple deployment examples to work.

closed 2022-10-06 by newracom