newracom/nrc7292_sw_pkg #34: Driver compile failed on latest release(v1.3.4_rev03)

matlinuxer2, 2022-05-25:
Just tried latest release and found the driver compiled failed. Information collected below:

Error message:

```
pi@raspberrypi:~/nrc7292_sw_pkg/package/host/src/nrc $ make

make[1]: Entering directory '/usr/src/linux-headers-5.15.32-v7l+'
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-trx.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-init.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-debug.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/hif.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/wim.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-hif-debug.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-hif-uart.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-hif-ssp.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-fw.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-netlink.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-ssp.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-hif-cspi.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/mac80211-ext.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-stats.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-pm.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-dump.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-hif-sdio.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-bd.o
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-bd.c: In function ‘nrc_check_bd’:
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-bd.c:921:11: error: implicit declaration of function ‘get_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
  921 |  old_fs = get_fs();
      |           ^~~~~~
      |           sget_fc
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-bd.c:921:11: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-bd.c:922:2: error: implicit declaration of function ‘set_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
  922 |  set_fs(KERNEL_DS);
      |  ^~~~~~
      |  sget_fc
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-bd.c:922:9: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
  922 |  set_fs(KERNEL_DS);
      |         ^~~~~~~~~
      |         KERNFS_NS
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-bd.c:922:9: note: each undeclared identifier is reported only once for each function it appears in
cc1: all warnings being treated as errors
make[2]: *** [scripts/Makefile.build:277: /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-bd.o] Error 1
make[1]: *** [Makefile:1868: /home/pi/nrc7292_sw_pkg/package/host/src/nrc] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.32-v7l+'
make: *** [Makefile:61: modules] Error 2
pi@raspberrypi:~/nrc7292_sw_pkg/package/host/src/nrc $ 
```

nrc7292_sw_pkg commit: https://github.com/newracom/nrc7292_sw_pkg/commit/5f11f25abd0080b33919bf05e835a7bf601779f2

OS: Raspberry Pi 4 (32-bit)
```
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```

kernel version: 5.15.32
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux
```

newracom, 2022-05-26:
Dear @matlinuxer2,

Thank you for your report.
I just pushed a patch, https://github.com/newracom/nrc7292_sw_pkg/commit/c04aaf5020b2358bf551bfd52283b6d363cd8e61
Please try again with it and let me know the result.
Thank you for your cooperation.

Best regards,
Newracom

matlinuxer2, 2022-05-26:
@newracom  It works now, thanks!

```
pi@raspberrypi:~/nrc7292_sw_pkg/package/host/src/nrc $ make

make[1]: Entering directory '/usr/src/linux-headers-5.15.32-v7l+'
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-trx.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-init.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-debug.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/hif.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/wim.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-hif-debug.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-hif-uart.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-hif-ssp.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-fw.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-netlink.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-ssp.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-hif-cspi.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/mac80211-ext.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-stats.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-pm.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-dump.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-hif-sdio.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-bd.o
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-recovery.o
  LD [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc.o
  MODPOST /home/pi/nrc7292_sw_pkg/package/host/src/nrc/Module.symvers
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc.mod.o
  LD [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.15.32-v7l+'
pi@raspberrypi:~/nrc7292_sw_pkg/package/host/src/nrc $ 
```

closed 2022-05-26 by matlinuxer2