newracom/nrc7292_sw_pkg #35: cannot insmod nrc module

fjydpf, 2022-05-30:
Dear newracom:
I am trying to running the module on a Pi CM4.   I get an Oops on insmod.   I am attaching the kernel trace, and the kernel version is  5.10.   Could you please check and see if you can help me out? I have checked UG-7292-018-Raspberry_Pi_setup.pdf. Thank you
[nrc-log.txt](https://github.com/newracom/nrc7292_sw_pkg/files/8795278/nrc-log.txt)
.

enganguish, 2022-05-31:
Hi @fjydpf ,

Did you wire your NRC7292 module or EVB to the Pi CM4 correctly?
NRC7292 module/EVB should be connected via HSPI host interface physically first.
Could you check this first?

fjydpf, 2022-05-31:
Hi@enganguish, thank you for your reply first. 

The above problem should be the mode error on the NRC7292 module, now it has a new error on insmod. the error message is '-nrc_check_start:error!!',  I am attaching the kernel trace, Could you please check and see if you can help me out? thank you!

[nrc-log.txt](https://github.com/newracom/nrc7292_sw_pkg/files/8802545/nrc-log.txt)

newracom, 2022-05-31:
Dear @fjydpf,

Could you share your results for below commands?

ls /dev
cat /boot/config.txt
uname -a
cat /etc/os-release

Best regards,
Newracom

fjydpf, 2022-06-02:
Dear @newracom
 
I am attaching the messages in 1.txt,please accept and help to check, thank you!

[1.txt](https://github.com/newracom/nrc7292_sw_pkg/files/8819788/1.txt)

newracom, 2022-06-24:
Dear @fjydpf,

Sorry for the late reply.
Based on your logs, I found one problem.
As guided on 3.4 of https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/UG-7292-018-Raspberry_Pi_setup.pdf, you need to disable user mode SPI device driver.
Please check this.

Best regards,
Newracom

newracom, 2022-08-11:
Dear @fjydpf,

I'd like to close this because there's no further update from you.
Please create a new one if you have any further issue.

Best regards,
Newracom

closed 2022-08-11 by newracom