newracom/nrc7292_sw_pkg #36: Failed to initialize hostapd interface for mesh.

synaption, 2022-06-23:
I'm trying to run start.py to set up a mesh portal.  Things look like they are going well until wpa_supplicant tries to initialize initialize the interface.  I am also not sure if the channel pair not allowed message matters.  Any idea where I'm going wrong?  

```
python nrc7292_sw_pkg/package/host/evk/sw_pkg/nrc_pkg/script/start.py 4 3 US 0$ bash mesh_ap_start.sh 
Done.
Done.
------------------------------
Model            : 7292
STA Type         : MESH
Country          : US
Security Mode    : WPA3-SAE
CAL. USE         : ON
AMPDU            : ON
Download FW      : uni_s1g.bin
TX Power         : 17
Mesh Mode        : Mesh Portal
------------------------------
NRC MESH setting for HaLow...
[*] Set Max CPU Clock on RPi
[0] Clear
wpa_supplicant: no process found
hostapd: no process found
wireshark-gtk: no process found
rmmod: ERROR: Module nrc is not currently loaded
rm: can't remove '/home/br2rauc/nrc7292_sw_pkg/package/host/evk/sw_pkg/nrc_pkg/script/conf/temp_self_config.conf': No such file or directory
[ 5731.442686] audit: type=1325 audit(1655868925.759:8): table=nat family=2 entries=0 op=xt_register pid=1270926 comm="modprobe"
[ 5731.454669] audit: type=1325 audit(1655868925.771:9): table=nat family=2 entries=5 op=xt_replace pid=1270916 comm="iptables"
[ 5731.491099] audit: type=1325 audit(1655868925.807:10): table=filter family=2 entries=4 op=xt_replace pid=1270939 comm="iptables"
[1] Copy
total 648
drwxr-xr-x    2 user     user          4096 Jun 22 03:35 .
drwxr-xr-x    4 user     user          4096 Jun 11 03:18 ..
-rwxr-xr-x    1 user     user           491 Jun 17 18:43 copy
-rwxr-xr-x    1 user     user           562 Jun 11 03:18 nrc7292_bd.dat
-rwxr-xr-x    1 user     user        323076 Jun 11 03:18 nrc7292_cspi.bin
-rwxr-xr-x    1 root     root        323076 Jun 22 03:35 uni_s1g.bin
-rwxr-xr-x    1 root     root        323076 Jun 22 03:35 /lib/firmware/uni_s1g.bin
====================================
 MESH INTERFACE   : wlan0 
 USE DHCP Client 
====================================
Config for Mesh Portal is done!
IP and DHCP config done
[2] Set Module Parameters
[3] Loading module
sudo insmod /lib/modules/5.10.92-v8/extra/nrc.ko  hifspeed=2000000 spi_bus_num=0 spi_cs_num=0 spi_gpio_irq=5 spi_gpio_poll=-1 fw_name=uni_s1g.bin power_save=0 ndp_preq=1 auto_ba=1 sw_enc=1 listen_interval=1000 credit_ac_be=40
x
[ 5733.039142] start FW 321
[ 5734.821509] end FW
spi_gpio_poll
[4] Set tx power
Tx power : 17                   Calibration_use : ON
OK
Board Data use                   : off
OK
[5] Set guard interval
guard interval : long
OK
[6] Set cal_use
Calibration_use : on                    Country  : US
OK
[*] Start DHCPCD and DNSMASQ
Job for dnsmasq.service failed because the control process exited with error code.
See "systemctl status dnsmasq.service" and "journalctl -xeu dnsmasq.service" for details.
wpa_supplicant: no process found
[7] Start wpa_supplicant on wlan0
Successfully initialized wpa_supplicant
[ 5744.942164] Failed to load board data, error:1
wlan0: CTRL-EVENT-REGDOM-CHANGE init=USER type=COUNTRY alpha2=US
HT40 channel pair (159, 155) not allowed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
Failed to initialize hostapd interface for mesh
wlan0: Failed to init mesh
wlan0: leaving mesh
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
wlan0: mesh leave error=-107
wlan0: Could not join mesh
[8] Start NAT
[ 5745.939412] audit: type=1325 audit(1655868940.255:11): table=filter family=2 entries=4 op=xt_replace pid=1274181 comm="iptables"
[ 5745.978409] audit: type=1325 audit(1655868940.295:12): table=nat family=2 entries=5 op=xt_replace pid=1274189 comm="iptables"
[ 5746.018299] audit: type=1325 audit(1655868940.335:13): table=filter family=2 entries=4 op=xt_replace pid=1274201 comm="iptables"
[ 5746.068107] audit: type=1325 audit(1655868940.383:14): table=nat family=2 entries=5 op=xt_replace pid=1274209 comm="iptables"
[ 5746.105864] audit: type=1325 audit(1655868940.419:15): table=filter family=2 entries=5 op=xt_replace pid=1274221 comm="iptables"
[ 5746.145032] audit: type=1325 audit(1655868940.459:16): table=nat family=2 entries=6 op=xt_replace pid=1274228 comm="iptables"
Waiting for IP
Waiting for IP
Waiting for IP
Waiting for IP
wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Traceback (most recent call last):
  File "/home/br2rauc/nrc7292_sw_pkg/package/host/evk/sw_pkg/nrc_pkg/script/start.py", line 725, in <module>
    run_mpp('wlan0', str(sys.argv[3]), strSecurity(), supplicant_debug, peer, static_ip)
  File "/home/br2rauc/nrc7292_sw_pkg/package/host/evk/sw_pkg/nrc_pkg/script/mesh.py", line 134, in run_mpp
    time.sleep(1)
KeyboardInterrupt
```

I'm using buildroot with the 64 bit 5.10 kernel on a pi4 with the sx-newah US dev board.

synaption, 2022-06-24:
I found some additional debugging info for the wpa_supplicant command.
```
$ sudo wpa_supplicant -dd -i wlan0 -c /home/br2rauc/nrc7292_sw_pkg/package/host/evk/sw_pkg/nrc_pkg/script/conf/US/mp_halow_sae.conf 1 
evk/sw_pkg/nrc_pkg/script/conf/US/mp_halow_sae.conf 1 
wpa_supplicant v2.10
random: Trying to read entropy from /dev/random
Successfully initialized wpa_supplicant
Initializing interface 'wlan0' conf '/home/br2rauc/nrc7292_sw_pkg/package/host/evk/sw_pkg/nrc_pkg/script/conf/US/mp_halow_sae.conf' driver 'default' ctrl_interface 'N/A' bridge 'N/A'
Configuration file '/home/br2rauc/nrc7292_sw_pkg/package/host/evk/sw_pkg/nrc_pkg/script/conf/US/mp_halow_sae.conf' -> '/home/br2rauc/nrc7292_sw_pkg/package/host/evk/sw_pkg/nrc_pkg/script/conf/US/mp_halow_sae.conf'
Reading configuration file '/home/br2rauc/nrc7292_sw_pkg/package/host/evk/sw_pkg/nrc_pkg/script/conf/US/mp_halow_sae.conf'
ctrl_interface='/var/run/wpa_supplicant'
country='US'
update_config=1
Line: 5 - start of a new network block
ssid - hexdump_ascii(len=8):
     6e 72 63 5f 6d 65 73 68                           nrc_mesh        
mode=5 (0x5)
scan_ssid=1 (0x1)
proto: 0x2
key_mgmt: 0x400
pairwise: 0x10
group: 0x10
PSK (ASCII passphrase) - hexdump_ascii(len=8): [REMOVED]
ieee80211w=2 (0x2)
beacon_int=100 (0x64)
frequency=5795 (0x16a3)
dot11MeshRetryTimeout=1000 (0x3e8)
dot11MeshHoldingTimeout=400 (0x190)
dot11MeshMaxRetries=4 (0x4)
mesh_rssi_threshold=0 (0x0)
PSK (from passphrase) - hexdump(len=32): [REMOVED]
p2p_disabled=1
ignore_old_scan_res=1
Priority group 0
   id=0 ssid='nrc_mesh'
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:10
nl80211: Supported cipher 00-0f-ac:6
nl80211: Supports Probe Response offload in AP mode
nl80211: Using driver-based off-channel TX
nl80211: Driver-advertised extended capabilities (default) - hexdump(len=8): 00 00 00 00 00 00 00 40
nl80211: Driver-advertised extended capabilities mask (default) - hexdump(len=8): 00 00 00 00 00 00 00 40
nl80211: Supported vendor command: vendor_id=0xfcffaa subcmd=176
nl80211: Supported vendor command: vendor_id=0xfcffaa subcmd=240
nl80211: Supported vendor command: vendor_id=0xfcffaa subcmd=241
nl80211: Supported vendor command: vendor_id=0xfcffaa subcmd=242
nl80211: Supported vendor command: vendor_id=0xfcffaa subcmd=243
nl80211: Supported vendor command: vendor_id=0xfcffaa subcmd=244
nl80211: Supported vendor command: vendor_id=0xfcffaa subcmd=224
nl80211: Supported vendor command: vendor_id=0xfcffaa subcmd=225
nl80211: Supported vendor command: vendor_id=0xfcffaa subcmd=226
nl80211: Supported vendor command: vendor_id=0xfcffaa subcmd=227
nl80211: Supported vendor command: vendor_id=0xfcffaa subcmd=228
nl80211: Supported vendor event: vendor_id=0xfcffaa subcmd=0
nl80211: Supported vendor event: vendor_id=0xfcffaa subcmd=1
nl80211: use P2P_DEVICE support
nl80211: key_mgmt=0x1ff0f enc=0x18f auth=0x7 flags=0x4000511f3e3d8e0 rrm_flags=0x10 probe_resp_offloads=0x6 max_stations=0 max_remain_on_chan=1000 max_scan_ssids=2
nl80211: interface wlan0 in phy nrc80211
nl80211: Set mode ifindex 3 iftype 2 (STATION)
nl80211: Subscribe to mgmt frames with non-AP handle 0x558ca773c0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x558ca773c0 match=040a multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x558ca773c0 match=040b multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x558ca773c0 match=040c multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x558ca773c0 match=040d multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x558ca773c0 match=090a multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x558ca773c0 match=090b multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x558ca773c0 match=090c multicast=0
[... 252 lines cut ...]
wlan0: Request to deauthenticate - bssid=00:00:00:00:00:00 pending_bssid=00:00:00:00:00:00 reason=3 (DEAUTH_LEAVING) state=SCANNING
TDLS: Tear down peers
wlan0: State: SCANNING -> DISCONNECTED
nl80211: Set wlan0 operstate 1->0 (DORMANT)
netlink: Operstate: ifindex=3 linkmode=-1 (no change), operstate=5 (IF_OPER_DORMANT)
EAPOL: External notification - portEnabled=0
EAPOL: External notification - portValid=0
QM: Clear all active DSCP policies
wlan0: CTRL-EVENT-DSCP-POLICY clear_all
wlan0: WPA: Clear old PMK and PTK
wlan0: Cancelling scan request
wlan0: Cancelling authentication timeout
Off-channel: Clear pending Action frame TX (pending_action_tx=(nil)
Off-channel: Action frame sequence done notification: pending_action_tx=(nil) drv_offchan_tx=1 action_tx_wait_time=0 off_channel_freq=0 roc_waiting_drv_freq=0
QM: Clear all active DSCP policies
wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Remove interface wlan0 from radio nrc80211
Remove radio nrc80211
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
netlink: Operstate: ifindex=3 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 3 iftype 2 (STATION)
nl80211: Unsubscribe mgmt frames handle 0x888888dd042ffb49 (mode change)
wlan0: CTRL-EVENT-TERMINATING 
```

enganguish, 2022-06-30:
Hi @synaption ,

This channel related issue occured by the changes in wpa_supplicant v2.10.

> HT40: control channel: 159 (5795 MHz), secondary channel: 155 (5775 MHz)
HT40 channel pair (159, 155) not allowed
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED

Please just comment out the line below. Then, you can solve the problem.
[https://github.com/newracom/nrc7292_sw_pkg/blob/e35a67a8d359f2ad2cf0099aa14dd1e68c89fb63/package/host/src/nrc/nrc-mac80211.c#L3543](https://github.com/newracom/nrc7292_sw_pkg/blob/e35a67a8d359f2ad2cf0099aa14dd1e68c89fb63/package/host/src/nrc/nrc-mac80211.c#L3543)

A next revision(v1.3.4_rev04) including this would be released soon.

synaption, 2022-12-05:
Seems like the US setting works now.

closed 2022-12-05 by synaption