newracom/nrc7292_sw_pkg #37: spi_master spi0: failed to transfer one message from queue

synaption, 2022-06-27:
I have seen other people with this issue a few times with different solutions, but I am still having problems initializing the driver reliably.  I have an sxnewah evaluation board, and a custom board with the sxnewah module.  I was hoping all I would have to do is hook up the reset pin, but I'm not sure that does anything at all.  A full power cycle seems to make successful driver loading much more likely, but not perfect.  How do I solve this problem once and for all?  Is this a software problem, or a hardware problem?

```
[28839.019397] nrc: unknown parameter 'spi_gpio_poll' ignored
[28839.051225] start FW 300
[28839.257613] nrc-cspi spi0.0: SPI transfer timed out
[28839.262664] spi_master spi0: failed to transfer one message from queue
[28839.269780] ------------[ cut here ]------------
[28839.274527] WARNING: CPU: 0 PID: 166688 at ../newah_driver-custom/nrc-hif-cspi.c:364 _c_spi_read_regs+0x208/0x230 [nrc]
[28839.285322] Modules linked in: nrc(O) mac80211 libarc4 tun 8021q garp stp llc snd_soc_hdmi_codec bcm2835_codec(C) v3d bcm2835_isp(C) bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) gpu_sched v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 raspberrypi_hwmon videobuf2_common brcmfmac i2c_brcmstb brcmutil videodev spi_bcm2835 vc_sm_cma(C) mc snd_bcm2835(C) snd_soc_bcm2835_i2s vc4 cec drm_kms_helper drm rpivid_mem cfg80211 drm_panel_orientation_quirks syscopyarea sysfillrect sysimgblt rfkill fb_sys_fops pps_gpio backlight pps_core uio_pdrv_genirq nvmem_rmem uio snd_soc_simple_card snd_soc_simple_card_utils snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd sch_fq_codel snd_i2smic_rpi(O) fuse ipv6
[28839.350567] CPU: 0 PID: 166688 Comm: kworker/0:0 Tainted: G         C O      5.10.92-v8 #1
[28839.358839] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[28839.365322] Workqueue: events nrc_check_start [nrc]
[28839.370216] pstate: 20000005 (nzCv daif -PAN -UAO -TCO BTYPE=--)
[28839.376260] pc : _c_spi_read_regs+0x208/0x230 [nrc]
[28839.381174] lr : _c_spi_read_regs+0x19c/0x230 [nrc]
[28839.386056] sp : ffffffc017083940
[28839.389373] x29: ffffffc017083940 x28: 0000000000000000 
[28839.394699] x27: ffffff80463c54c8 x26: ffffffc0114ce788 
[28839.400022] x25: 0000000000000000 x24: 0000000000000001 
[28839.405344] x23: 0000000000000000 x22: ffffff804764f800 
[28839.410667] x21: 0000000000000020 x20: 0000000000000010 
[28839.415989] x19: ffffffc017083c78 x18: ffffffc0970834a7 
[28839.421310] x17: 000000000000000c x16: 000000000000000c 
[28839.426632] x15: 0000000000000006 x14: 0720072007200720 
[28839.431955] x13: 0720072007200765 x12: 0775076507750771 
[28839.437276] x11: 0720076d076f0772 x10: ffffffc01135d990 
[28839.442598] x9 : ffffffc0107e348c x8 : ffffffc017083c2c 
[28839.447920] x7 : ffffffc0112e9000 x6 : ffffffc0112e9210 
[28839.453242] x5 : 0000000000000001 x4 : 0000000000000000 
[28839.458565] x3 : ffffff8043884320 x2 : 0000000000000000 
[28839.463886] x1 : ffffff8040999e40 x0 : 00000000000000ff 
[28839.469209] Call trace:
[28839.471690]  _c_spi_read_regs+0x208/0x230 [nrc]
[28839.476255]  c_spi_read_regs+0x34/0x70 [nrc]
[28839.480557]  spi_wait_ack+0x54/0xb0 [nrc]
[28839.484599]  nrc_fw_check_next_frag+0x108/0x140 [nrc]
[28839.489683]  nrc_download_fw+0x2fc/0x380 [nrc]
[28839.494160]  nrc_check_start+0x630/0x780 [nrc]
[28839.498615]  process_one_work+0x1d8/0x4c0
[28839.502628]  worker_thread+0x144/0x470
[28839.506385]  kthread+0x14c/0x160
[28839.509618]  ret_from_fork+0x10/0x38
[28839.513197] ---[ end trace 04658f747fb730c6 ]---
[28839.517981] ------------[ cut here ]------------
[28839.522619] kernel BUG at ../newah_driver-custom/nrc-fw.c:193!
[28839.528459] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP
[28839.533948] Modules linked in: nrc(O) mac80211 libarc4 tun 8021q garp stp llc snd_soc_hdmi_codec bcm2835_codec(C) v3d bcm2835_isp(C) bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) gpu_sched v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 raspberrypi_hwmon videobuf2_common brcmfmac i2c_brcmstb brcmutil videodev spi_bcm2835 vc_sm_cma(C) mc snd_bcm2835(C) snd_soc_bcm2835_i2s vc4 cec drm_kms_helper drm rpivid_mem cfg80211 drm_panel_orientation_quirks syscopyarea sysfillrect sysimgblt rfkill fb_sys_fops pps_gpio backlight pps_core uio_pdrv_genirq nvmem_rmem uio snd_soc_simple_card snd_soc_simple_card_utils snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd sch_fq_codel snd_i2smic_rpi(O) fuse ipv6
[28839.599169] CPU: 0 PID: 166688 Comm: kworker/0:0 Tainted: G        WC O      5.10.92-v8 #1
[28839.607440] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[28839.613923] Workqueue: events nrc_check_start [nrc]
[28839.618816] pstate: 60000005 (nZCv daif -PAN -UAO -TCO BTYPE=--)
[28839.624856] pc : nrc_fw_check_next_frag+0x12c/0x140 [nrc]
[28839.630289] lr : nrc_fw_check_next_frag+0x108/0x140 [nrc]
[28839.635693] sp : ffffffc017083ca0
[28839.639009] x29: ffffffc017083ca0 x28: 0000000000000000 
[28839.644333] x27: ffffff80463c54c8 x26: ffffffc0114ce788 
[28839.649656] x25: 0000000000000000 x24: 0000000000000001 
[28839.654978] x23: 0000000000000000 x22: ffffff8043851400 
[28839.660301] x21: ffffff8047671ec0 x20: ffffff8042ff5400 
[28839.665624] x19: ffffff80463bf880 x18: ffffffc0970834a7 
[28839.670946] x17: 000000000000000c x16: 000000000000000c 
[28839.676267] x15: 0000000000000006 x14: 0720072007200720 
[28839.681589] x13: 0720072007200765 x12: 0775076507750771 
[28839.686912] x11: 0720076d076f0772 x10: ffffffc01135d990 
[28839.692235] x9 : ffffffc00906b6f8 x8 : ffffff8040999e40 
[28839.697558] x7 : ffffffc0112e9000 x6 : ffffffc0112e9210 
[28839.702880] x5 : 0000000000000001 x4 : 0000000000000000 
[28839.708202] x3 : 00000000000004f5 x2 : 0000000000000000 
[28839.713523] x1 : 0000000000000000 x0 : 00000000fffffffb 
[28839.718846] Call trace:
[28839.721327]  nrc_fw_check_next_frag+0x12c/0x140 [nrc]
[28839.726412]  nrc_download_fw+0x2fc/0x380 [nrc]
[28839.730887]  nrc_check_start+0x630/0x780 [nrc]
[28839.735342]  process_one_work+0x1d8/0x4c0
[28839.739357]  worker_thread+0x144/0x470
[28839.743113]  kthread+0x14c/0x160
[28839.746348]  ret_from_fork+0x10/0x38
[28839.749933] Code: d4210000 52800000 17ffffd8 d4210000 (d4210000) 
[28839.756036] ---[ end trace 04658f747fb730c7 ]---
[28839.760660] note: kworker/0:0[166688] exited with preempt_count 1
[28839.767119] ------------[ cut here ]------------
[28839.771775] WARNING: CPU: 0 PID: 0 at kernel/rcu/tree.c:624 rcu_eqs_enter.constprop.0+0x138/0x140
[28839.780654] Modules linked in: nrc(O) mac80211 libarc4 tun 8021q garp stp llc snd_soc_hdmi_codec bcm2835_codec(C) v3d bcm2835_isp(C) bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) gpu_sched v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 raspberrypi_hwmon videobuf2_common brcmfmac i2c_brcmstb brcmutil videodev spi_bcm2835 vc_sm_cma(C) mc snd_bcm2835(C) snd_soc_bcm2835_i2s vc4 cec drm_kms_helper drm rpivid_mem cfg80211 drm_panel_orientation_quirks syscopyarea sysfillrect sysimgblt rfkill fb_sys_fops pps_gpio backlight pps_core uio_pdrv_genirq nvmem_rmem uio snd_soc_simple_card snd_soc_simple_card_utils snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd sch_fq_codel snd_i2smic_rpi(O) fuse ipv6
[28839.845875] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G      D WC O      5.10.92-v8 #1
[28839.853537] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[28839.859984] pstate: 20000085 (nzCv daIf -PAN -UAO -TCO BTYPE=--)
[28839.866001] pc : rcu_eqs_enter.constprop.0+0x138/0x140
[28839.871146] lr : rcu_eqs_enter.constprop.0+0x24/0x140
[28839.876200] sp : ffffffc0112e3ea0
[28839.879517] x29: ffffffc0112e3ea0 x28: 0000000001150018 
[28839.884842] x27: 000000003b357834 x26: ffffffc0112eff40 
[28839.890164] x25: 0000000000000000 x24: 0000000000000000 
[28839.895486] x23: ffffffc010e3ca60 x22: ffffffc011039138 
[28839.900808] x21: 0000000000000000 x20: ffffffc0112e97c8 
[28839.906131] x19: ffffffc01103af80 x18: 000000000000000e 
[28839.911453] x17: 0000000000000001 x16: 0000000000000001 
[28839.916775] x15: 0000001d5b6ef58c x14: 0000000eadb77ac6 
[28839.922098] x13: 0000000000000029 x12: 0000000000000040 
[28839.927420] x11: ffffff8040400000 x10: ffffff8040400002 
[28839.932742] x9 : ffffffc010b42e08 x8 : ffffff8040400028 
[28839.938063] x7 : ffffffc0112e9000 x6 : ffffffc0112e9210 
[28839.943385] x5 : 0000000000000001 x4 : 0000000000000000 
[28839.948706] x3 : ffffffc011025000 x2 : 4000000000000000 
[28839.954028] x1 : 4000000000000002 x0 : ffffff80fb794f80 
[28839.959350] Call trace:
[28839.961804]  rcu_eqs_enter.constprop.0+0x138/0x140
[28839.966602]  rcu_idle_enter+0x1c/0x30
[28839.970273]  default_idle_call+0x38/0x1d4
[28839.974289]  do_idle+0x260/0x2b0
[28839.977522]  cpu_startup_entry+0x34/0x80
[28839.981450]  rest_init+0xe0/0xf0
[28839.984685]  arch_call_rest_init+0x1c/0x28
[28839.988787]  start_kernel+0x540/0x578
[28839.992451] ---[ end trace 04658f747fb730c8 ]---
```

enganguish, 2022-07-01:
Hi @synaption ,

As I see, your [nrc_pkg](https://github.com/newracom/nrc7292_sw_pkg/tree/master/package/host/evk/sw_pkg/nrc_pkg) and the nrc driver doesn't seem to be matched. right?
(You might use an old nrc_pkg(maybe 1.3.4 rev01) with a newer nrc driver.)

Then, could you please update your nrc7292_sw_pkg to the latest version(v1.3.4 rev04)?

synaption, 2022-07-03:
I have made sure nrc_pkg and the driver are correct while upgraded to the newest revision, and I am still having this issue about 50% of the time.

enganguish, 2022-07-04:
Could you show me the syslog with v1.3.4_rev04?

And, please let me know the cli you used when insmod as well.

synaption, 2022-07-04:
I am using bash.  v1.3.4_rev4 and kernel 5.15.
```
sudo insmod /lib/modules/$(uname -r)/extra/nrc.ko  hifspeed=10000000 spi_bus_num=0 spi_cs_num=0 spi_gpio_irq=5 spi_polling_interval=0 fw_name=uni_s1g.bin auto_ba=1 sw_enc=1 listen_interval=1000
[   57.032406] start FW 330
[   57.238227] nrc-cspi spi0.0: SPI transfer timed out
[   57.243265] spi_master spi0: failed to transfer one message from queue
[   57.250102] ------------[ cut here ]------------
[   57.254756] WARNING: CPU: 0 PID: 34 at ../newah_driver-custom/nrc-hif-cspi.c:364 _c_spi_read_regs+0x208/0x230 [nrc]
[   57.265317] Modules linked in: nrc(O) iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 8021q garp stp llc snd_soc_hdmi_codec v3d raspberrypi_hwmon gpu_sched brcmfmac brcmutil bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) i2c_brcmstb bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 spi_bcm2835 videobuf2_common vc4 snd_soc_bcm2835_i2s snd_bcm2835(C) cec vc_sm_cma(C) videodev mc drm_kms_helper drm rpivid_mem snd_soc_simple_card drm_panel_orientation_quirks snd_soc_simple_card_utils pps_gpio snd_soc_core syscopyarea sysfillrect sysimgblt fb_sys_fops backlight snd_compress snd_pcm_dmaengine nvmem_rmem uio_pdrv_genirq uio snd_pcm snd_timer snd sch_fq_codel snd_i2smic_rpi(O) mac80211 libarc4 cfg80211 rfkill fuse ipv6
[   57.334815] CPU: 0 PID: 34 Comm: kworker/0:1 Tainted: G         C O      5.15.32-v8 #1
[   57.342749] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[   57.349196] Workqueue: events nrc_check_start [nrc]
[   57.354145] pstate: 20000005 (nzCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   57.361118] pc : _c_spi_read_regs+0x208/0x230 [nrc]
[   57.366064] lr : _c_spi_read_regs+0x19c/0x230 [nrc]
[   57.371004] sp : ffffffc00978b950
[   57.374320] x29: ffffffc00978b950 x28: 0000000000000000 x27: 0000000000000000
[   57.381476] x26: ffffffc009563848 x25: ffffff80fb78b305 x24: 0000000000000001
[   57.388630] x23: 0000000000000000 x22: ffffff8045646000 x21: 0000000000000020
[   57.395783] x20: 0000000000000010 x19: ffffffc00978bc88 x18: ffffffc08978b4d7
[   57.402935] x17: 0000000000000001 x16: 0000000000000001 x15: 076d07200765076e
[   57.410088] x14: 076f072007720765 x13: 076d07200765076e x12: 076f072007720765
[   57.417240] x11: 0720072007200720 x10: 0000000000000000 x9 : ffffffc00881029c
[   57.424393] x8 : 0000000000000004 x7 : 0000000000000003 x6 : ffffffc009379000
[   57.431545] x5 : ffffffc009379238 x4 : 0000000000000000 x3 : ffffff8045641b38
[   57.438697] x2 : 0000000000000000 x1 : ffffff8040328000 x0 : 00000000000000ff
[   57.445849] Call trace:
[   57.448297]  _c_spi_read_regs+0x208/0x230 [nrc]
[   57.452892]  c_spi_read_regs+0x34/0x70 [nrc]
[   57.457225]  spi_wait_ack+0x54/0xb0 [nrc]
[   57.461295]  nrc_fw_check_next_frag+0x108/0x140 [nrc]
[   57.466407]  nrc_download_fw+0x2fc/0x380 [nrc]
[   57.470909]  nrc_check_start+0x63c/0x790 [nrc]
[   57.475412]  process_one_work+0x1dc/0x490
[   57.479436]  worker_thread+0x144/0x470
[   57.483192]  kthread+0x144/0x160
[   57.486430]  ret_from_fork+0x10/0x20
[   57.490014] ---[ end trace 5f765400c22e14eb ]---
[   57.494784] ------------[ cut here ]------------
[   57.499422] kernel BUG at ../newah_driver-custom/nrc-fw.c:193!
[   57.505261] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP
[   57.510756] Modules linked in: nrc(O) iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 8021q garp stp llc snd_soc_hdmi_codec v3d raspberrypi_hwmon gpu_sched brcmfmac brcmutil bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) i2c_brcmstb bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 spi_bcm2835 videobuf2_common vc4 snd_soc_bcm2835_i2s snd_bcm2835(C) cec vc_sm_cma(C) videodev mc drm_kms_helper drm rpivid_mem snd_soc_simple_card drm_panel_orientation_quirks snd_soc_simple_card_utils pps_gpio snd_soc_core syscopyarea sysfillrect sysimgblt fb_sys_fops backlight snd_compress snd_pcm_dmaengine nvmem_rmem uio_pdrv_genirq uio snd_pcm snd_timer snd sch_fq_codel snd_i2smic_rpi(O) mac80211 libarc4 cfg80211 rfkill fuse ipv6
[   57.580222] CPU: 0 PID: 34 Comm: kworker/0:1 Tainted: G        WC O      5.15.32-v8 #1
[   57.588152] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[   57.594598] Workqueue: events nrc_check_start [nrc]
[   57.599556] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   57.606529] pc : nrc_fw_check_next_frag+0x12c/0x140 [nrc]
[   57.611992] lr : nrc_fw_check_next_frag+0x108/0x140 [nrc]
[   57.617452] sp : ffffffc00978bcb0
[   57.620768] x29: ffffffc00978bcb0 x28: 0000000000000000 x27: 0000000000000000
[   57.627922] x26: ffffffc009563848 x25: ffffff80fb78b305 x24: 0000000000000001
[   57.635075] x23: 0000000000000000 x22: ffffff804541f680 x21: ffffff8045791fc0
[   57.642228] x20: ffffff804560d200 x19: ffffff804541f580 x18: ffffffc08978b4d7
[   57.649380] x17: 0000000000000001 x16: 0000000000000001 x15: 076d07200765076e
[   57.656533] x14: 076f072007720765 x13: 076d07200765076e x12: 076f072007720765
[   57.663686] x11: 0720072007200720 x10: 0000000000000000 x9 : ffffffc001254908
[   57.670838] x8 : ffffff8040328000 x7 : 0000000000000003 x6 : ffffffc009379000
[   57.677990] x5 : ffffffc009379238 x4 : 0000000000000000 x3 : 00000000000004f6
[   57.685142] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 00000000fffffffb
[   57.692295] Call trace:
[   57.694742]  nrc_fw_check_next_frag+0x12c/0x140 [nrc]
[... cut ...]
I am curious why there is a reference to newah_driver-custom.  This is the name of the build directory that my buildroot uses to crossbuild the driver.  It does not exist on the target.  
```
[   57.254756] WARNING: CPU: 0 PID: 34 at ../newah_driver-custom/nrc-hif-cspi.c:364 _c_spi_read_regs+0x208/0x230 [nrc]
````
It's worth noting that I have been making sure to properly rmmod the driver before rebooting.  Still the success rate is about 50% and does not seem to depend on which pi I am using, or which power supply.  I have tried a few.

synaption, 2022-07-05:
I may have solved this problem.  It looks like running clock_config.sh is absolutely necessary.  I could be wrong but so far I have loaded the driver 20 times in a row.  Perhaps the driver was starting correctly when the cpu happened to be automatically clocking itself at a high rate, but then failing when in clocked slower.  My understanding is that the script forces a high rate without automatic scaling.

synaption, 2022-07-05:
It looks like I do still need to run rmmod before reboot in order for the driver to start correctly.  The problem is that I loose the connection once the driver get removed so sending the reboot command becomes tricky.  I'm working on a hack to have systemd remove the driver, but this is more of a workaround than a solution.

enganguish, 2022-07-11:
We have tried to reproduce your problem on our Raspberry Pi 3 and 4, not RPi Compute Module 4 (we don't have this yet), but we were not able to see the same symptom with v5.10.43 and v5.10.63 on that two Pies with both 32-bit and 64-bit images.

Do you have RPi 4(not compute module)and can you see the same problem on it? If so, please let us have your driver logs or syslogs.

synaption, 2022-07-12:
It looks like it does work with a regular pi 4.

newracom, 2022-10-06:
Dear @synaption,

I close this because CM4 is out of our scope at the moment.

Best regards,
Newracom

closed 2022-10-06 by newracom