newracom/nrc7292_sw_pkg #38: failed to register hw

synaption, 2022-06-29:
I have been trying out the latest master branch.  I'm not sure why it is not loading the driver.  I am using a pi4 on 64 bit kernel 5.15.32-v8.  Is this kernel to new?  
```
$sudo insmod /lib/modules/$(uname -r)/extra/nrc.ko  hifspeed=10000000 spi_bus_num=0 spi_cs_num=0 spi_gpio_irq=5 spi_gpio_poll=-1 fw_name=uni_s1g.bin power_save=0 ndp_preq=1 auto_ba=1 sw_enc=1 listen_interval=1000 credit_ac_be=40
[  122.130680] nrc: unknown parameter 'spi_gpio_poll' ignored
[  122.163484] start FW 300
[  122.719501] end FW
[  125.927391] ------------[ cut here ]------------
[  125.932045] WARNING: CPU: 0 PID: 128 at net/wireless/reg.c:2593 wiphy_apply_custom_regulatory+0x114/0x120 [cfg80211]
[  125.942812] Modules linked in: nrc(O) mac80211 libarc4 8021q garp stp llc snd_soc_hdmi_codec bcm2835_codec(C) raspberrypi_hwmon bcm2835_isp(C) v3d gpu_sched bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem brcmfmac videobuf2_vmalloc videobuf2_dma_contig brcmutil videobuf2_memops i2c_brcmstb videobuf2_v4l2 videobuf2_common vc_sm_cma(C) snd_soc_bcm2835_i2s spi_bcm2835 vc4 videodev snd_bcm2835(C) cec mc drm_kms_helper cfg80211 rpivid_mem drm rfkill drm_panel_orientation_quirks pps_gpio syscopyarea sysfillrect sysimgblt fb_sys_fops backlight nvmem_rmem uio_pdrv_genirq uio snd_soc_simple_card snd_soc_simple_card_utils snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd sch_fq_codel snd_i2smic_rpi(O) fuse ipv6
[  126.006882] CPU: 0 PID: 128 Comm: kworker/0:3 Tainted: G         C O      5.15.32-v8 #1
[  126.014903] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[  126.021351] Workqueue: events nrc_check_start [nrc]
[  126.026300] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  126.033273] pc : wiphy_apply_custom_regulatory+0x114/0x120 [cfg80211]
[  126.039920] lr : nrc_register_hw+0x3c8/0x6f0 [nrc]
[  126.044774] sp : ffffffc009c5bc80
[  126.048091] x29: ffffffc009c5bc80 x28: ffffff8046e00242 x27: 0000000000000002
[  126.055247] x26: ffffff80464246e8 x25: ffffff8046420460 x24: 0000000000000000
[  126.062402] x23: ffffff8046420460 x22: ffffffc00110fb30 x21: ffffff8046420300
[  126.069555] x20: 0000000000000000 x19: ffffff8046421fc0 x18: 00000000000024d0
[  126.076709] x17: 00000100000000c8 x16: 0000032000000000 x15: ffffffc001117118
[  126.083863] x14: ffffffc0011171a8 x13: ffffffc001117148 x12: 0000000e00000000
[  126.091017] x11: 0000000000000006 x10: 000000000000002c x9 : ffffffc0010f77a8
[  126.098170] x8 : 000000000000000b x7 : ffffffc001117b68 x6 : 0000000000000009
[  126.105322] x5 : ffffffc00110faf8 x4 : ffffffc001117d18 x3 : 0000000000000007
[  126.112473] x2 : 0000000000200001 x1 : ffffffc00110fb30 x0 : 0000000000200001
[  126.119628] Call trace:
[  126.122077]  wiphy_apply_custom_regulatory+0x114/0x120 [cfg80211]
[  126.128378]  nrc_register_hw+0x3c8/0x6f0 [nrc]
[  126.132883]  nrc_check_start+0x4d0/0x780 [nrc]
[  126.137388]  process_one_work+0x1dc/0x490
[  126.141412]  worker_thread+0x144/0x470
[  126.145168]  kthread+0x144/0x160
[  126.148406]  ret_from_fork+0x10/0x20
[  126.151991] ---[ end trace 0a3ee9eceff2caab ]---
[  126.157302] ------------[ cut here ]------------
[  126.161944] WARNING: CPU: 0 PID: 128 at net/wireless/core.c:886 wiphy_register+0x920/0x940 [cfg80211]
[  126.171415] Modules linked in: nrc(O) mac80211 libarc4 8021q garp stp llc snd_soc_hdmi_codec bcm2835_codec(C) raspberrypi_hwmon bcm2835_isp(C) v3d gpu_sched bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem brcmfmac videobuf2_vmalloc videobuf2_dma_contig brcmutil videobuf2_memops i2c_brcmstb videobuf2_v4l2 videobuf2_common vc_sm_cma(C) snd_soc_bcm2835_i2s spi_bcm2835 vc4 videodev snd_bcm2835(C) cec mc drm_kms_helper cfg80211 rpivid_mem drm rfkill drm_panel_orientation_quirks pps_gpio syscopyarea sysfillrect sysimgblt fb_sys_fops backlight nvmem_rmem uio_pdrv_genirq uio snd_soc_simple_card snd_soc_simple_card_utils snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd sch_fq_codel snd_i2smic_rpi(O) fuse ipv6
[  126.235479] CPU: 0 PID: 128 Comm: kworker/0:3 Tainted: G        WC O      5.15.32-v8 #1
[  126.243497] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[  126.249944] Workqueue: events nrc_check_start [nrc]
[  126.254890] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  126.261862] pc : wiphy_register+0x920/0x940 [cfg80211]
[  126.267214] lr : ieee80211_register_hw+0x828/0xbf0 [mac80211]
[  126.273179] sp : ffffffc009c5bba0
[  126.276495] x29: ffffffc009c5bba0 x28: 0000000000000000 x27: 0000000000000000
[  126.283648] x26: 0000000000000000 x25: 0000000000000000 x24: ffffffc0011e2468
[  126.290801] x23: ffffffc0011e2460 x22: ffffff8046421be0 x21: 0000000000000005
[  126.297953] x20: ffffff8046420000 x19: ffffff8046420300 x18: 0000000000000014
[  126.305106] x17: 0000000007fc16f5 x16: 000000004dd9af0c x15: 00000000d24a0006
[  126.312259] x14: 0000000005ea4a86 x13: 00000000ccedeb6e x12: 0000000000000068
[  126.319411] x11: 00000000000000c0 x10: 00000000ffffffbd x9 : ffffffc000ee4280
[  126.326564] x8 : ffffff8046420438 x7 : 000000007fffffff x6 : 0000000000000005
[  126.333716] x5 : 0000000000000004 x4 : 0000000000000000 x3 : 0000000000000008
[  126.340866] x2 : 000000009c3f2584 x1 : 0000000000000000 x0 : 0000000000000000
[  126.348019] Call trace:
[  126.350467]  wiphy_register+0x920/0x940 [cfg80211]
[  126.355465]  ieee80211_register_hw+0x828/0xbf0 [mac80211]
[  126.361074]  nrc_register_hw+0x584/0x6f0 [nrc]
[  126.365581]  nrc_check_start+0x4d0/0x780 [nrc]
[  126.370085]  process_one_work+0x1dc/0x490
[  126.374109]  worker_thread+0x144/0x470
[  126.377864]  kthread+0x144/0x160
[  126.381103]  ret_from_fork+0x10/0x20
[  126.384687] ---[ end trace 0a3ee9eceff2caac ]---
[  126.389636] failed to register hw
[  126.393027] ------------[ cut here ]------------
[  126.397692] refcount_t: underflow; use-after-free.
[  126.402564] WARNING: CPU: 0 PID: 128 at lib/refcount.c:28 refcount_warn_saturate+0xf8/0x150
[  126.410958] Modules linked in: nrc(O) mac80211 libarc4 8021q garp stp llc snd_soc_hdmi_codec bcm2835_codec(C) raspberrypi_hwmon bcm2835_isp(C) v3d gpu_sched bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem brcmfmac videobuf2_vmalloc videobuf2_dma_contig brcmutil videobuf2_memops i2c_brcmstb videobuf2_v4l2 videobuf2_common vc_sm_cma(C) snd_soc_bcm2835_i2s spi_bcm2835 vc4 videodev snd_bcm2835(C) cec mc drm_kms_helper cfg80211 rpivid_mem drm rfkill drm_panel_orientation_quirks pps_gpio syscopyarea sysfillrect sysimgblt fb_sys_fops backlight nvmem_rmem uio_pdrv_genirq uio snd_soc_simple_card snd_soc_simple_card_utils snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd sch_fq_codel snd_i2smic_rpi(O) fuse ipv6
[... 124 lines cut ...]
[  218.653241]  __netif_receive_skb_core+0x4fc/0xa10
[  218.657955]  __netif_receive_skb_one_core+0x40/0x90
[  218.662845]  __netif_receive_skb+0x20/0x70
[  218.666950]  netif_receive_skb+0x48/0x1c0
[  218.670967]  tun_get_user+0xbf8/0xd80 [tun]
[  218.675176]  tun_chr_write_iter+0x68/0xb0 [tun]
[  218.679724]  new_sync_write+0xec/0x190
[  218.683484]  vfs_write+0x244/0x2b0
[  218.686893]  ksys_write+0x74/0x100
[  218.690301]  __arm64_sys_write+0x28/0x40
[  218.694233]  invoke_syscall+0x50/0x120
[  218.697994]  el0_svc_common.constprop.0+0x104/0x130
[  218.702882]  do_el0_svc+0x30/0xa0
[  218.706205]  el0_svc+0x24/0x70
[  218.709270]  el0t_64_sync_handler+0x1a4/0x1b0
[  218.713638]  el0t_64_sync+0x1a0/0x1a4
[  218.717312] Code: f9401e94 eb1402ff d100e294 54000140 (79400280) 
[  218.723414] ---[ end trace 0a3ee9eceff2cab0 ]---
[  218.728038] Kernel panic - not syncing: Oops: Fatal exception in interrupt
[  218.734920] SMP: stopping secondary CPUs
[  218.738851] Kernel Offset: disabled
[  218.742340] CPU features: 0x200000d1,00000846
[  218.746700] Memory Limit: none
[  218.749760] ---[ end Kernel panic - not syncing: Oops: Fatal exception in interrupt ]---
```

enganguish, 2022-07-01:
Hi @synaption ,

As I see, your [nrc_pkg](https://github.com/newracom/nrc7292_sw_pkg/tree/master/package/host/evk/sw_pkg/nrc_pkg) and the nrc driver doesn't seem to be matched. right?
(You might use an old nrc_pkg(maybe 1.3.4 rev01) with a newer nrc driver.)

Then, could you please update your nrc7292_sw_pkg to the latest version(v1.3.4 rev04)?

synaption, 2022-07-02:
Where are you able to see the mismatch?  I thought I was using the right combination, but I don't understand what nrc_pkg has to do with using insmod to directly load the driver.  In any case, I will give the new revision a try.

synaption, 2022-07-03:
I think using the correct version of uni_s1g.bin fixed this particular issue, but I still am only able to start from a full power cycle bout 50% of the time without running into an spi timeout.

closed 2022-07-03 by synaption