newracom/nrc7292_sw_pkg #39: Bridging eth0 works with AP Mode, not with STA Mode (4addr on, pi4)

bhjelstrom, 2022-06-29:
I have modified start.py to bridge eth0 to the NRC7292 wlan0.

in run_ap, I have added the following for the AP-mode pi 4:

```
    os.system("sudo brctl addif br0 " + "eth0")
    os.system("sudo brctl addif br0 " + "wlan0")
    os.system("sudo ifconfig br0 " + getIpAddr("wlan0"))
    os.system("sudo ifconfig " + "wlan0" + " 0.0.0.0")
    os.system("sudo ifconfig " + "eth0" + " 0.0.0.0")
    os.system("sudo ip route add default via " + getNetAddr("br0") + ".1 dev br0")
```

This works perfectly. the computer wired to the AP-PI4 over ethernet can ping the PI4 as well as connected STA-Mode Pi 4 units. The STA Pi4 units can also ping the computer and AP-mode pi 4.

I would also like to bridge eth0 of one of the STA Pi4s. In run_sta, I have added the following:
```
    os.system("sudo iw dev wlan0 set 4addr on")
    os.system("sudo brctl addif br0 " + "eth0")
    os.system("sudo brctl addif br0 " + "wlan0")
    os.system("sudo ifconfig br0 " + getIpAddr("wlan0"))
    os.system("sudo ifconfig " + "wlan0" + " 0.0.0.0")
    os.system("sudo ifconfig " + "eth0" + " 0.0.0.0")
    os.system("sudo ip route add default via " + getNetAddr("br0") + ".1 dev br0")
```

With 4addr enabled, there are no error messages while making the bridge. Still, the STA loses the ability to ping devices on the network as soon as the bridge is established. Is it possible to bridge STA-Mode wlan0 with eth0? Or are we limited to bridging STA-mode wlan0 to the RPI4s wlan1?

Thanks

enganguish, 2022-06-30:
Hi @bhjelstrom ,

Could you share your results for below commands?
============================
uname -a
cat /etc/os-release
============================

bhjelstrom, 2022-06-30:
uname -a
`Linux raspberrypi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux
`

cat /etc/os-release
`PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
`

bhjelstrom, 2022-06-30:
Update: The STA-mode Pi4 loses all connectivity to the AP-mode pi4 as soon as 4addr is enabled. I can break connectivity with `sudo iw dev wlan0 set 4addr on` and fix connectivity with `sudo iw dev wlan0 set 4addr on`

enganguish, 2022-06-30:
Thank you for the answer.

Could you share the contents of your hostapd configuration file for your AP?

bhjelstrom, 2022-06-30:
For the AP, I'm simply using start.py with wpa2. Bridging with br0 works perfectly on the AP.

```

ctrl_interface=/var/run/hostapd
country_code=US
interface=wlan0
ssid=halow_demo
hw_mode=a

#1MHz
############
#Below channel Must be set hw_mode=g in line 7
############
#902.5MHz
#channel=1
#903.5MHz
#channel=3
#904.5MHz
#channel=5
#905.5MHz
#channel=7
#906.5MHz
#channel=9
#907.5MHz
#channel=11
#############
#908.5MHz
#channel=36
#909.5MHz
#channel=37
#910.5MHz
#channel=38
#911.5MHz
#channel=39
#912.5MHz
#channel=40
#913.5MHz
#channel=41
#914.5MHz
#channel=42
#915.5MHz
#channel=43
#916.5MHz
#channel=44
#917.5MHz
#channel=45
#918.5MHz
#channel=46
#919.5MHz
#channel=47
#920.5MHz
#channel=48
#921.5MHz
#channel=149
#922.5MHz
#channel=150
#923.5MHz
#channel=151
#924.5MHz
#channel=152
#925.5MHz
#channel=100
#926.5MHz
#channel=104
#927.5MHz
#channel=108

#2MHz
############
#Below channel Must be set hw_mode=g in line 7
############
#903MHz
#channel=2
#905MHz
#channel=6
#907MHz
#channel=10
############
#909MHz
#channel=153
#911MHz
#channel=154
#913MHz
#channel=155
#915MHz
#channel=156
#917MHz
#channel=157
#919MHz
#channel=158
#921MHz
#channel=159
#923MHz
#channel=160
#925MHz
channel=161
#927MHz
#channel=112

#4MHz
############
#Below channel Must be set hw_mode=g in line 7
############
#906MHa
#channel=8
############
#910MHz
#channel=162
#914MHz
#channel=163
#918MHz
#channel=164
#922MHz
#channel=165
#926MHz
#channel=116

ieee80211h=1
ieee80211d=1
ieee80211n=1
#wmm_enabled=1
macaddr_acl=0
driver=nl80211
beacon_int=100
#disable_sgi=0
#ap_max_inactivity=3
# USF=0, UI=2 : 3
# USF=1, UI=2 : 16780
# USF=0, UI=10: 11
# USF=1, UI=10: 16788
#listen_interval=2

#WPA2-PSK
#auth_alg=1 #1=wpa, 2=wep, 3=open
wpa=2      # wpa2 only
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP
rsn_pairwise=CCMP
wpa_passphrase=12345678

```

enganguish, 2022-06-30:
Then, Please add 'wds_sta=1' into your configuration file. this is necessary for the STA supported 4-address.

And, I have also added 'bridge=br0' when I tested. but, I'm not sure about this whether it should be applied for you or not.
Because, when I tried to config a bridge on AP, I have referred [this](https://wikidocs.net/78533).

bhjelstrom, 2022-06-30:
That was it.
Thanks for helping so quickly!

enganguish, 2022-06-30:
Oh, it's good!!

Then, could you please close this issue if there's no problem?

Thank you.

closed 2022-07-01 by bhjelstrom