newracom/nrc7292_sw_pkg #40: how t configure to acces internet

spereg53, 2022-07-05:
how to configure halow ap for access to internet by ethernet on raspberry,thanks

spereg53, 2022-07-06:
> Dear @spereg53 <https://github.com/spereg53>,
>
> Could you share your detailed requirements of Internet access?
> If you mean to use embedded WiFi (802.11n) of Raspberry Pi with HaLow,
> it's possible with script changes.
>
> For now, all provided scripts are made for wlan0 interface.
> So, if you enable embedded WiFi of Raspberry Pi, embedded WiFi will use
> wlan0 and HaLow will use wlan1 interface.
> And this configuration requires the corresponding scripts update.
>
> Best,
> Aaron

in one of the raspberry pi 3 ,,is connected to eth0 to internet,but i can
not get work like AP ,i follow all instructions and can´t get nrc´s
interfaces and don´t see SSID halow-demo,can you help me ,thanks
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

newracom, 2022-07-06:
Dear @spereg53,

Could you check your RPi 3 if it comply with [UG-7292-018-Raspberry_Pi_setup.pdf](https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/UG-7292-018-Raspberry_Pi_setup.pdf)?

Best regards,
Newracom

newracom, 2022-07-06:
Dear @spereg53,

If you want to enable NAT on AP, please follow steps.

(1) Edit start.py and change eth0 into your internet interface.
[RPi3#1] vi start.py
def startNAT():
    os.system('sudo sh -c "echo 1 > /proc/sys/net/ipv4/ip_forward"')
    os.system("sudo iptables -t nat -A POSTROUTING -o **eth0** -j MASQUERADE")
    os.system("sudo iptables -A FORWARD -i **eth0** -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT")
    os.system("sudo iptables -A FORWARD -i wlan0 -o **eth0** -j ACCEPT")
(2) Start AP: start.py will call startNAT(), so there's no need to run any additional commands.
[RPi3#1] ./start.py 1 3 US 161
(3) Start STA
[RPi3#2] ./start.py 0 3 US
(4) Check internet connection
[RPi3#2] ping 8.8.8.8

If you have any further question, please let me know.

Best regards,
Newracom

spereg53, 2022-07-06:
if i want to connect one pc to raspberry3(2)STA,how can get to use the
internet of RPI3(1)AP???how can configure eth0 on rpi3(2) for static
ip??,thanks
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

spereg53, 2022-07-09:
how can i configure the nrc7292 for get more tx and can get more
distance???is recommendable??,with your help,i can get to work two modules
and i got with wireguard vpn,thanks ....
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

newracom, 2022-07-12:
Dear @spereg53,

Could you clarify your requirements?
If you mean following configuration, AP-STA mode doesn't work due to the Linux kernel limitation.

PC <---ethernet---> NRC7292 EVK (STA) <---HaLow---> NRC7292 EVK (AP) <---ethernet---> Internet

On Linux, the bridge setup between managed wireless (STA) and wired interface (ethernet) is impossible. 
And there’s WDS feature enabling this, but this is a non-standard extension to the IEEE 802.11 standard, a.k.a. Wi Fi standard. So, that is disabled on the Linux. If user wants this feature, Linux kernel should be updated.
However, WDS is not supported on NRC7292 EVK.
Instead, NRC7292 EVK support 802.11s mesh and this feature supports above setup.
Please refer to https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/AN-7292-007-11s_mesh_network.pdf.

Best regards,
Newracom

spereg53, 2022-07-21:
but how can i use nrc7292 like an repeater or extender my network???
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

newracom, 2022-07-21:
Dear @spereg53,

Please try with following commands.

[image]

Best regards,
Newracom

spereg53, 2022-07-21:
sorry,don´t worked,i give the command ./start.py 4 3 US 0 and is waiting
ip.....,and i try again with the command ./start.py 4 3 US 0 192.168.222.1
and is still waiting ip.....,can you helpme
thanks
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

spereg53, 2022-07-23:
can not get work with mesh portal ,this is the display:
./start.py 4 3 US 0
Done.
Done.
------------------------------
Model            : 7292
STA Type         : MESH
Country          : US
Security Mode    : WPA3-SAE
CAL. USE         : ON
AMPDU            : ON
Download FW      : uni_s1g.bin
TX Power         : 17
Mesh Mode        : Mesh Portal
------------------------------
NRC MESH setting for HaLow...
[*] Set Max CPU Clock on RPi
1200000
1200000
1200000
1200000
Done
[0] Clear
hostapd: ningún proceso encontrado
wireshark-gtk: ningún proceso encontrado
rmmod: ERROR: Module nrc is not currently loaded
rm: no se puede borrar '/home/pi/nrc_pkg/script/conf/temp_self_config.conf': No existe el fichero o el directorio
rm: no se puede borrar '/home/pi/nrc_pkg/script/conf/temp_hostapd_config.conf': No existe el fichero o el directorio
[1] Copy
total 336
drwxr-xr-x 2 pi pi   4096 jul  6 09:41 .
drwxr-xr-x 4 pi pi   4096 jul  6 09:41 ..
-rwxr-xr-x 1 pi pi    362 may 25 05:40 copy
-rwxr-xr-x 1 pi pi    562 may 25 05:40 nrc7292_bd.dat
-rwxr-xr-x 1 pi pi 325628 may 25 05:40 nrc7292_cspi.bin
-rwxr-xr-x 1 root root 325628 may 25 05:40 /lib/firmware/uni_s1g.bin
====================================
 MESH INTERFACE   : wlan0
 USE DHCP Client
====================================
Config for Mesh Portal is done!
IP and DHCP config done
[2] Set Module Parameters
[3] Loading module
sudo modprobe nrc  hifspeed=20000000 spi_bus_num=0 spi_cs_num=0 spi_gpio_irq=5 spi_polling_interval=0 fw_name=uni_s1g.bin power_save=0 auto_ba=1 sw_enc=1 listen_interval=1000 credit_ac_be=40
[4] Set tx power
Tx power : 17                   Calibration_use : ON
OK
Board Data use                   : off
OK
[5] Set guard interval
guard interval : long
OK
[6] Set cal_use
Calibration_use : on                    Country  : US
OK
[*] Start DHCPCD and DNSMASQ
[7] Start wpa_supplicant on wlan0
Successfully initialized wpa_supplicant
Line 20: unknown network field 'dot11MeshRetryTimeout'.
Line 21: unknown network field 'dot11MeshHoldingTimeout'.
Line 22: unknown network field 'dot11MeshMaxRetries'.
Line 23: unknown network field 'mesh_rssi_threshold'.
Line 24: unknown network field 'mesh_basic_rates'.
Line 26: failed to parse network block.
Failed to read or parse configuration '/home/pi/nrc_pkg/script/conf/US/mp_halow_sae.conf'.
[8] Start NAT
Waiting for IP
Waiting for IP
Waiting for IP
Waiting for IP
Waiting for IP
Waiting for IP
Waiting for IP
^CTraceback (most recent call last):
  File "/home/pi/nrc_pkg/script/./start.py", line 762, in <module>
    run_mpp('wlan0', str(sys.argv[3]), strSecurity(), supplicant_debug, peer, static_ip)
  File "/home/pi/nrc_pkg/script/mesh.py", line 134, in run_mpp
    time.sleep(1)
KeyboardInterrupt

can you helpme,thanks in advanced

spereg53, 2022-08-11:
> Dear @spereg53 <https://github.com/spereg53>,
>
> I'm afraid you missed 2.1 of
> https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/AN-7292-007-11s_mesh_network.pdf
> Please follow the every steps of this application note.
>
> Best regards,
> Newracom

i did all the steps and is still waiting ip;
***@***.***:~/nrc_pkg/script $ ./start.py 4 3 US 0*
*Done.*
*Done.*
*------------------------------*
*Model            : 7292*
*STA Type         : MESH*
*Country          : US*
*Security Mode    : WPA3-SAE*
*CAL. USE         : ON*
*AMPDU            : ON*
*Download FW      : uni_s1g.bin*
*TX Power         : 17*
*Mesh Mode        : Mesh Portal*
*------------------------------*
*NRC MESH setting for HaLow...*
*[*] Set Max CPU Clock on RPi*
*1200000*
*1200000*
*1200000*
*1200000*
*Done*
*[0] Clear*
*wpa_supplicant: ningún proceso encontrado*
*hostapd: ningún proceso encontrado*
*wireshark-gtk: ningún proceso encontrado*
*rmmod: ERROR: Module nrc is not currently loaded*
*rm: no se puede borrar
'/home/pi/nrc_pkg/script/conf/temp_self_config.conf': No existe el fichero
o el directorio*
*rm: no se puede borrar
'/home/pi/nrc_pkg/script/conf/temp_hostapd_config.conf': No existe el
fichero o el directorio*
*[1] Copy*
*total 336*
*drwxr-xr-x 2 pi pi   4096 jul  6 09:41 .*
*drwxr-xr-x 4 pi pi   4096 jul  6 09:41 ..*
*-rwxr-xr-x 1 pi pi    362 may 25 05:40 copy*
*-rwxr-xr-x 1 pi pi    562 may 25 05:40 nrc7292_bd.dat*
*-rwxr-xr-x 1 pi pi 325628 may 25 05:40 nrc7292_cspi.bin*
*-rwxr-xr-x 1 root root 325628 may 25 05:40 /lib/firmware/uni_s1g.bin*
*====================================*
* MESH INTERFACE   : wlan0*
* USE DHCP Client*
*====================================*
*Config for Mesh Portal is done!*
*IP and DHCP config done*
*[2] Set Module Parameters*
*[3] Loading module*
*sudo modprobe nrc  hifspeed=20000000 spi_bus_num=0 spi_cs_num=0
spi_gpio_irq=5 spi_polling_interval=0 fw_name=uni_s1g.bin power_save=0
auto_ba=1 sw_enc=1 listen_interval=1000 credit_ac_be=40*
*[4] Set tx power*
*Tx power : 17                   Calibration_use : ON*
*OK*
*Board Data use                   : off*
*OK*
*[5] Set guard interval*
*guard interval : long*
*OK*
*[6] Set cal_use*
*Calibration_use : on                    Country  : US*
*OK*
*[*] Start DHCPCD and DNSMASQ*
*[7] Start wpa_supplicant on wlan0*
*Successfully initialized wpa_supplicant*
*wlan0: CTRL-EVENT-REGDOM-CHANGE init=USER type=COUNTRY alpha2=US*
*[8] Start NAT*
*Waiting for IP*
*Waiting for IP*
*Waiting for IP*
*Waiting for IP*
*Waiting for IP*

what can i do???
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

newracom, 2022-10-06:
Dear @spereg53.

I close this.
Please retry with the latest SW package and open a new issue if you meet any problem.

Best regards,
Newracom

closed 2022-10-06 by newracom

spereg53, 2022-10-11:
i am working with modules from sparkfun,i want to get evk from
newraom,where can i get it???
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
***@***.***
***@***.***

magicone2571, 2024-04-22:
> Dear @spereg53,
> 
> Please try with following commands.
> 
> [image]
> 
> Best regards, Newracom

Do I put those commands in the mesh.py file? Or do I run those in that order from the CLI?