newracom/nrc7292_sw_pkg #41: Cannot install nrc.ko in raspberry-linux 5.15.44-v8+. Please help, thank you in advance.

TaylorIsAGoodboy, 2022-07-07:
Environment: `uname -a`
```
Linux rakpios 5.15.44-v8+ #1 SMP PREEMPT Wed Jun 8 15:22:35 CST 2022 aarch64 GNU/Linux
```
I download the lastest raspberry linux kernel, and compile the nrc.ko in 64bit mode.  But when I install the driver I built, it failed.

Result of `sudo insmod nrc.ko`
```
insmod: ERROR: could not insert module nrc.ko: Invalid parameters
```

`dmesg` log:
```
[ 1750.623465] nrc: no symbol version for module_layout
[ 1750.623545] nrc: loading out-of-tree module taints kernel.
[ 1750.626372] nrc: disagrees about version of symbol ieee80211_stop_tx_ba_session
[ 1750.626389] nrc: Unknown symbol ieee80211_stop_tx_ba_session (err -22)
[ 1750.626467] nrc: disagrees about version of symbol ieee80211_start_tx_ba_cb_irqsafe
[ 1750.626479] nrc: Unknown symbol ieee80211_start_tx_ba_cb_irqsafe (err -22)
[ 1750.626524] nrc: disagrees about version of symbol ieee80211_connection_loss
[ 1750.626535] nrc: Unknown symbol ieee80211_connection_loss (err -22)
[ 1750.626614] nrc: Unknown symbol __ll_sc_atomic64_or (err -2)
[ 1750.626655] nrc: disagrees about version of symbol ieee80211_iterate_interfaces
[ 1750.626667] nrc: Unknown symbol ieee80211_iterate_interfaces (err -22)
[ 1750.626721] nrc: disagrees about version of symbol ieee80211_chswitch_done
[ 1750.626732] nrc: Unknown symbol ieee80211_chswitch_done (err -22)
[ 1750.626842] nrc: Unknown symbol __stack_chk_guard (err -2)
[ 1750.626892] nrc: Unknown symbol vfs_statx (err -2)
[ 1750.626940] nrc: disagrees about version of symbol ieee80211_free_hw
[ 1750.626951] nrc: Unknown symbol ieee80211_free_hw (err -22)
[ 1750.627133] nrc: disagrees about version of symbol ieee80211_iterate_stations_atomic
[ 1750.627147] nrc: Unknown symbol ieee80211_iterate_stations_atomic (err -22)
[ 1750.627208] nrc: Unknown symbol nla_strlcpy (err -2)
[ 1750.627266] nrc: disagrees about version of symbol ieee80211_start_tx_ba_session
[ 1750.627277] nrc: Unknown symbol ieee80211_start_tx_ba_session (err -22)
[ 1750.627319] nrc: disagrees about version of symbol ieee80211_beacon_loss
[ 1750.627330] nrc: Unknown symbol ieee80211_beacon_loss (err -22)
[ 1750.627378] nrc: disagrees about version of symbol ieee80211_nullfunc_get
[ 1750.627389] nrc: Unknown symbol ieee80211_nullfunc_get (err -22)
[ 1750.627442] nrc: disagrees about version of symbol ieee80211_register_hw
[ 1750.627452] nrc: Unknown symbol ieee80211_register_hw (err -22)
[ 1750.627498] nrc: disagrees about version of symbol ieee80211_restart_hw
[ 1750.627508] nrc: Unknown symbol ieee80211_restart_hw (err -22)
[ 1750.627569] nrc: Unknown symbol cfg80211_find_ie_match (err -2)
[ 1750.627636] nrc: disagrees about version of symbol ieee80211_find_sta
[ 1750.627646] nrc: Unknown symbol ieee80211_find_sta (err -22)
[ 1750.627696] nrc: Unknown symbol __ll_sc_atomic64_andnot (err -2)
[ 1750.627744] nrc: disagrees about version of symbol ieee80211_tx_status_irqsafe
[ 1750.627755] nrc: Unknown symbol ieee80211_tx_status_irqsafe (err -22)
[ 1750.627818] nrc: disagrees about version of symbol __cfg80211_alloc_reply_skb
[ 1750.627829] nrc: Unknown symbol __cfg80211_alloc_reply_skb (err -22)
[ 1750.627868] nrc: disagrees about version of symbol __cfg80211_alloc_event_skb
[ 1750.627879] nrc: Unknown symbol __cfg80211_alloc_event_skb (err -22)
[ 1750.627927] nrc: Unknown symbol usleep_range (err -2)
[ 1750.628003] nrc: disagrees about version of symbol ieee80211_stop_tx_ba_cb_irqsafe
[ 1750.628014] nrc: Unknown symbol ieee80211_stop_tx_ba_cb_irqsafe (err -22)
[ 1750.628057] nrc: disagrees about version of symbol wiphy_to_ieee80211_hw
[ 1750.628067] nrc: Unknown symbol wiphy_to_ieee80211_hw (err -22)
[ 1750.628107] nrc: disagrees about version of symbol __cfg80211_send_event_skb
[ 1750.628118] nrc: Unknown symbol __cfg80211_send_event_skb (err -22)
[ 1750.628176] nrc: disagrees about version of symbol ieee80211_queue_delayed_work
[ 1750.628187] nrc: Unknown symbol ieee80211_queue_delayed_work (err -22)
[ 1750.628285] nrc: disagrees about version of symbol wiphy_apply_custom_regulatory
[ 1750.628296] nrc: Unknown symbol wiphy_apply_custom_regulatory (err -22)
[ 1750.628346] nrc: disagrees about version of symbol ieee80211_remain_on_channel_expired
[ 1750.628357] nrc: Unknown symbol ieee80211_remain_on_channel_expired (err -22)
[ 1750.628421] nrc: disagrees about version of symbol ieee80211_wake_queues
[ 1750.628432] nrc: Unknown symbol ieee80211_wake_queues (err -22)
[ 1750.628477] nrc: disagrees about version of symbol ieee80211_tx_prepare_skb
[ 1750.628488] nrc: Unknown symbol ieee80211_tx_prepare_skb (err -22)
[ 1750.628525] nrc: disagrees about version of symbol ieee80211_beacon_get_template
[ 1750.628536] nrc: Unknown symbol ieee80211_beacon_get_template (err -22)
[ 1750.628583] nrc: disagrees about version of symbol ieee80211_free_txskb
[ 1750.628594] nrc: Unknown symbol ieee80211_free_txskb (err -22)
[ 1750.628646] nrc: Unknown symbol printk (err -2)
[ 1750.628691] nrc: disagrees about version of symbol ieee80211_alloc_hw_nm
[ 1750.628701] nrc: Unknown symbol ieee80211_alloc_hw_nm (err -22)
[ 1750.628753] nrc: disagrees about version of symbol ieee80211_tx_dequeue
[ 1750.628764] nrc: Unknown symbol ieee80211_tx_dequeue (err -22)
[ 1750.628814] nrc: disagrees about version of symbol cfg80211_vendor_cmd_reply
[ 1750.628825] nrc: Unknown symbol cfg80211_vendor_cmd_reply (err -22)
[ 1750.628886] nrc: disagrees about version of symbol ieee80211_stop_queue
[ 1750.628897] nrc: Unknown symbol ieee80211_stop_queue (err -22)
[ 1750.628953] nrc: Unknown symbol __ll_sc_atomic_add (err -2)
[ 1750.628991] nrc: disagrees about version of symbol ieee80211_ready_on_channel
[ 1750.629001] nrc: Unknown symbol ieee80211_ready_on_channel (err -22)
[ 1750.629045] nrc: disagrees about version of symbol wdev_to_ieee80211_vif
[ 1750.629056] nrc: Unknown symbol wdev_to_ieee80211_vif (err -22)
[ 1750.629095] nrc: disagrees about version of symbol ieee80211_stop_queues
[ 1750.629106] nrc: Unknown symbol ieee80211_stop_queues (err -22)
[ 1750.629174] nrc: disagrees about version of symbol ieee80211_vif_to_wdev
[ 1750.629185] nrc: Unknown symbol ieee80211_vif_to_wdev (err -22)
[ 1750.629227] nrc: disagrees about version of symbol ieee80211_scan_completed
[ 1750.629238] nrc: Unknown symbol ieee80211_scan_completed (err -22)
[ 1750.629313] nrc: Unknown symbol __ll_sc_atomic_sub (err -2)
[ 1750.629354] nrc: disagrees about version of symbol ieee80211_unregister_hw
[ 1750.629365] nrc: Unknown symbol ieee80211_unregister_hw (err -22)
[ 1750.629499] nrc: Unknown symbol dev_printk (err -2)
[ 1750.629565] nrc: Unknown symbol __alloc_workqueue_key (err -2)
[ 1750.629648] nrc: Unknown symbol cfg80211_find_vendor_ie (err -2)
[ 1750.629692] nrc: Unknown symbol _mcount (err -2)
[ 1750.629748] nrc: disagrees about version of symbol ieee80211_csa_finish
[ 1750.629759] nrc: Unknown symbol ieee80211_csa_finish (err -22)
[ 1750.629798] nrc: disagrees about version of symbol ieee80211_rx_irqsafe
[ 1750.629809] nrc: Unknown symbol ieee80211_rx_irqsafe (err -22)
```

R0uter, 2022-09-04:
Hi there, I m reopening this issue since I encountered this error too. Can I ask how did you solve this problem?
I got  `insmod: ERROR: could not insert module /home/pi/nrc_pkg/sw/driver/nrc.ko: Unknown symbol in module` all of a sudden after a reboot of Pi, now I have no idea about this. The error persists even after I rebuild the .ko file. 🤕

kc2rxo, 2022-09-15:
Chiming in that I also am experiencing the `unknown symbol in module` issue using `v1.3.4_rev08`.

`uname -a`:

```
Linux raspberrypi 5.15.61-v7+ #1579 SMP Fri Aug 26 11:10:59 BST 2022 armv7l GNU/Linux
```

synaption, 2022-11-18:
I get this error often when I compile change something about the kernel so it detects that the kernel is different from the one that the driver is compiled against.  YMMV