newracom/nrc7292_sw_pkg #42: Cannot insmod the driver with rapspberry pi cm4 and 64 bit OS.

TaylorIsAGoodboy, 2022-07-11:
I bought AHPI7292S modules, but has not received yet. So I compile the driver before, when i run start script,  segment default happenned. 

Is the reason of no hardware devive?

`uname -a`
```
Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```
`cat /boot/config.txt`

[image]

Log is as followings:
```
pi@raspberrypi:~/nrc_pkg/script $ ./start.py 1 0 US
Done.
Done.
------------------------------
Model            : 7292
STA Type         : AP
Country          : US
Security Mode    : OPEN
CAL. USE         : ON
AMPDU            : ON
Download FW      : uni_s1g.bin
Max TX Power     : 24
BSS MAX IDLE     : 180
------------------------------
NRC AP setting for HaLow...
[*] Set Max CPU Clock on RPi
1400000
1400000
1400000
1400000
Done
[0] Clear
hostapd: no process found
wireshark-gtk: no process found
rmmod: ERROR: Module nrc is not currently loaded
rm: cannot remove '/home/pi/nrc_pkg/script/conf/temp_self_config.conf': No such file or directory
rm: cannot remove '/home/pi/nrc_pkg/script/conf/temp_hostapd_config.conf': No such file or directory
Failed to stop dnsmasq.service: Unit dnsmasq.service not loaded.
[1] Copy and Set Module Parameters
total 672
drwxr-xr-x 2 pi   pi     4096 Jul 11 05:08 .
drwxr-xr-x 4 pi   pi     4096 Jul 11 05:08 ..
-rwxrwxrwx 1 pi   pi      281 Jul 11 05:08 copy
-rwxr-xr-x 1 pi   pi     1980 Jul 11 05:08 nrc7292_bd.dat
-rwxr-xr-x 1 pi   pi   331996 Jul 11 05:08 nrc7292_cspi.bin
-rwxr-xr-x 1 root root 331996 Jul 11 05:08 uni_s1g.bin
-rwxr-xr-x 1 root root 331996 Jul 11 05:08 /lib/firmware/uni_s1g.bin
====================================
 AP INTERFACE     : wlan0
 AP STATIC IP     : 192.168.200.1 
 NET MASK NUM     : 24 
====================================
Config for AP is done!
IP and DHCP config done
[2] Set Country
[3] Loading module
sudo insmod /home/pi/nrc_pkg/sw/driver/nrc.ko  hifspeed=20000000 spi_bus_num=0 spi_cs_num=0 spi_gpio_irq=5 spi_polling_interval=0 fw_name=uni_s1g.bin bss_max_idle=180 ndp_preq=1 auto_ba=1 listen_interval=1000
Segmentation fault

Message from syslogd@raspberrypi at Jul 11 05:08:32 ...
 kernel:[ 3773.740823] Internal error: Oops: 96000005 [#1] PREEMPT SMP

Message from syslogd@raspberrypi at Jul 11 05:08:32 ...
 kernel:[ 3773.747832] Code: 35000780 531e7421 52814402 d2800000 (f9403f57) 

Message from syslogd@raspberrypi at Jul 11 05:08:32 ...
 kernel:[ 3773.747841] Internal error: Oops: 96000005 [#2] PREEMPT SMP

Message from syslogd@raspberrypi at Jul 11 05:08:32 ...
 kernel:[ 3773.748359] Code: b9000abf b9414a60 7100001f 540003ac (b942cec0) 
wlan0: ERROR while getting interface flags: No such device
rmmod: ERROR: Module nrc is in use

```

TaylorIsAGoodboy, 2022-07-11:
`sudo insmod nrc.ko`
[image]

enganguish, 2022-07-11:
Hi @TaylorIsAGoodboy ,

Yes, the host driver can be run successfully when only your module is connected physically and correctly.

Please try again with your module after received.

newracom, 2022-10-06:
Dear @TaylorIsAGoodboy,

I close this because there's no update from you.
I expect you already resolved this issue with the help of comment above.

Best regards,
Newracom

closed 2022-10-06 by newracom