newracom/nrc7292_sw_pkg #44: STA cannot connect AP.

enric-lee, 2022-07-20:
I met a issue with raspberry pi  OS(2022-04-04).
MY AP ruan ok, bu STA run error.

```
pi@raspberrypi:~/nrc_pkg/script $ sudo ./start.py 0 0 US
Done.
Done.
------------------------------
Model            : 7292
STA Type         : STA
Country          : US
Security Mode    : OPEN
CAL. USE         : ON
AMPDU            : ON
CQM              : ON
Download FW      : uni_s1g.bin
TX Power         : 17
Power Save Type  : Always On
Listen Interval  : 1000
------------------------------
NRC STA setting for HaLow...
[*] Set Max CPU Clock on RPi
1400000
1400000
1400000
1400000
Done
[0] Clear
wpa_supplicant: no process found
hostapd: no process found
wireshark-gtk: no process found
rmmod: ERROR: Module nrc is not currently loaded
rm: cannot remove '/home/pi/nrc_pkg/script/conf/temp_self_config.conf': No such file or directory
rm: cannot remove '/home/pi/nrc_pkg/script/conf/temp_hostapd_config.conf': No such file or directory
[1] Copy
total 336
drwxr-xr-x 2 pi pi   4096 Jul 20 00:43 .
drwxr-xr-x 4 pi pi   4096 Jul 20 00:43 ..
-rwxr-xr-x 1 pi pi    362 May 25 01:40 copy
-rwxr-xr-x 1 pi pi    562 May 25 01:40 nrc7292_bd.dat
-rwxr-xr-x 1 pi pi 325628 May 25 01:40 nrc7292_cspi.bin
-rwxr-xr-x 1 root root 325628 May 25 01:40 /lib/firmware/uni_s1g.bin
====================================
 STA INTERFACE    : wlan0
 USE DHCP Client 
====================================
Config for STA is done!
IP and DHCP config done
[2] Set Module Parameters
[3] Loading module
sudo modprobe nrc  hifspeed=20000000 spi_bus_num=0 spi_cs_num=0 spi_gpio_irq=5 spi_polling_interval=0 fw_name=uni_s1g.bin power_save=0 auto_ba=1 listen_interval=1000 credit_ac_be=40
[4] Set tx power
family ID request : receive error
FAIL
family ID request : receive error
FAIL
[5] Set guard interval
family ID request : receive error
FAIL
[6] Set cal_use
family ID request : receive error
FAIL
[*] Start DHCPCD and DNSMASQ
[6] Start wpa_supplicant on wlan0
Successfully initialized wpa_supplicant
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
nl80211: kernel reports: Match already configured
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
[7] Connect and DHCP
Waiting for IP
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Waiting for IP
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Waiting for IP
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Waiting for IP
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Waiting for IP
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Waiting for IP
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Waiting for IP
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Waiting for IP
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
wlan0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
```

newracom, 2022-07-20:
Dear @enric-lee,

Please try again with "./start.py 0 0 US" (no need to use sudo command)
And could you provide the result of following commands?
- uname -a
- cat /etc/os-release
- wpa_supplicant -v

Best regards,
Newracom

enric-lee, 2022-07-21:
For removing `sudo`, there is still the same error.  BTW, I installed driver according to https://docs.alfa.com.tw/Product/AHPI7292S/20_Getting_Started_New/, and with the image of raspberry pi [2022-04-04](https://docs.alfa.com.tw/Product/AHPI7292S/20_Getting_Started_New/2022-04-04-raspios-bullseye-armhf_halow_nrc7292_1.3.4_2022-04-18.img.xz).

For `STA`, the results of commands are as follow:
[image]

For `AP`,  the results of commands are as follow:
[image]

enganguish, 2022-07-21:
Hi @enric-lee ,

It seems that there might be a problem for loading nrc driver module while running start.py.
By the way, did you modify the start.py file? The original start.py doesn't use 'modprobe'.

Please try again with the original start.py.

enric-lee, 2022-07-25:
I did not do any other things only refer to https://docs.alfa.com.tw/Product/AHPI7292S/20_Getting_Started_New/  step by step.
I will do them again to find if there is the same issue.

newracom, 2022-10-06:
Dear @enric-lee,

I close this because there's no update from you.
Please open a new issue for further question.

Best regards,
Newracom

closed 2022-10-06 by newracom