newracom/nrc7292_sw_pkg #47: BD file is invalid..Exit!!

kkcloudy, 2022-08-02:
Hi, I try to deloy this project with raspberry pi CM4, my test module is AHPI7292S. I follow the documentation 'UG-7292-018-Raspberry_Pi_setup.pdf', set everything up , then build the driver module 'nrc.ko' from source code. but when I start an AP module with 'start.py' script, there are some error message in the kernel. I check the source code, find it is caused by the file 'nrc7292_bd.dat',
[image]

I know the board data file can be dsiabled, when I disabled it, the driver works well. but I am not sure it is a good choice to disable it, 
is there any potential risk or issue? If I want to enable it, what should I do?  is there any way to generate the matched board data file?
look forward to your replay.

tumist, 2022-08-03:
There seems to be no documentation on this file and its purpose. I too this this issue with the supplied 2128 byte board file, but the 562 byte board file from the nrc7292_openwrt repository works, at least combined with that openwrt driver.

Is it possible to disable the board file check using a module parameter? Commenting out some lines in nrc-mac80211.c also does the trick, but cumbersome.

newracom, 2023-02-01:
Dear @kkcloudy,

There's target version number at the BD file and this should be configured to be aligned with the target board.
I'm afraid AHPI7292S has a different value from NRC7292 EVK. So, I updated that value of BD file.
Please try with this: https://newratek-my.sharepoint.com/:u:/g/personal/jehun_lee_newracom_com/Eas32q7PLc5CjvxLlYVHYcgBOV0ZJ3X_ZQrevveEvxQccQ?e=GI3Uo3

Best regards,
Newracom

govindsi, 2023-10-01:
I am wondering if [https://newratek-my.sharepoint.com/:u:/g/personal/jehun_lee_newracom_com/Eas32q7PLc5CjvxLlYVHYcgBOV0ZJ3X_ZQrevveEvxQccQ?e=GI3Uo3] is fully compatible with latest release nrc7292_sw_pkg-1.4.1. It works, but i am afraid if there are some optimization missing with old bd file.

jr1221, 2023-10-02:
Is there any way to get the ALFA sources used to make the bd file?  I am having the same problem I want the latest newracom package.

psmitty7373, 2023-10-02:
If you go through the old commits there is a tool to generate your own board files.

https://github.com/newracom/nrc7292_sw_pkg/blob/71cc30047ce2980d49d2c3f55070b3b5a04b2554/package/host/tools/BoardDataEditor/BoardDataEditor.exe

jr1221, 2023-10-27:
Too bad they seemed to have updated the schema since then.  The reason for removal: `Remove BoardDataEditor tool due to liability issue`

psmitty7373, 2023-10-27:
I use custom board files in all my builds as generated with that tool.

jr1221, 2023-10-27:
On the latest sw version?  I couldn't load any of the old or new bd files so I figured it didn't work.  Good to know, thanks!