newracom/nrc7292_sw_pkg #48: imx8mq no spi activity

synaption, 2022-08-02:
Due to the relative unavailability of raspberry pies I am trying to get the nrc7292 to work with an google coral dev board which uses an imx8mq cpu.  Running the nrc_simple sanity check produces no output according to an oscilloscope.  I'm new to imx, so I don't know if this requires use of spi_imx.c in addition to the regular spi.c code from the linux kernal.  Spi1.0 seems to exist, and spidev does not.
 Below is the output from the commands if that helps.

'''
$ sudo insmod /lib/modules/$(uname -r)/extra/nrc_simple.ko spi_bus_num=1 spi_cs_num=0 hifspeed=10000000
[  660.919948] ### [nrc_simple] Value of paramters ###
[  660.924872] - bus_num: 1
[  660.927476] - chip_select: 0
[  660.930399] - max_speed_hz: 20000000
[  660.934321] [nrc_hspi_probe,L166]
[  660.937912] ------------[ cut here ]------------
[  660.942539] WARNING: CPU: 2 PID: 217441 at ../newah_simple-custom/nrc_hspi_simple_driver.c:151 c_spi_read_regs.constprop.0+0x138/0x178 [nrc_simple]
[  660.955771] Modules linked in: nrc_simple(O+) fsl_jr_uio caam_jr caamkeyblob_desc caamhash_desc caamalg_desc crypto_engine rng_core authenc libdes imx8mq_interconnect imx_interconnect crct10dif_ce imx8m_ddrc governor_userspace imx_bus caam secvio error fuse [last unloaded: nrc_simple]
[  660.981014] CPU: 2 PID: 217441 Comm: insmod Tainted: G        W  O      5.15.32 #1
[  660.988586] Hardware name: Google i.MX8MQ Phanbell (DT)
[  660.993811] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  661.000774] pc : c_spi_read_regs.constprop.0+0x138/0x178 [nrc_simple]
[  661.007223] lr : c_spi_read_regs.constprop.0+0x124/0x178 [nrc_simple]
[  661.013671] sp : ffff80000c37b5b0
[  661.016984] x29: ffff80000c37b5b0 x28: 0000000000000013 x27: 0000000000000100
[  661.024126] x26: ffff80000131b140 x25: ffff800008133d70 x24: 0000000000000021
[  661.031266] x23: 0000000000000026 x22: ffff800009d710e0 x21: ffff80000131b050
[  661.038406] x20: ffff80000c37b8c8 x19: ffff0000058ab800 x18: 0000000000000030
[  661.045546] x17: 0000000000000180 x16: ffff80000131a030 x15: 0000000000000000
[  661.052686] x14: 00000000000000eb x13: 0000000000000024 x12: 0000000000000000
[  661.059827] x11: ffff00003fd9afec x10: 0000000000000110 x9 : 0000000000000000
[  661.066966] x8 : ffff00003fdafec0 x7 : 0000000000000000 x6 : 0000000000000000
[  661.074107] x5 : 0000000000000000 x4 : 0000000000000000 x3 : ffff0000045d6380
[  661.081246] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000000
[  661.088387] Call trace:
[  661.090834]  c_spi_read_regs.constprop.0+0x138/0x178 [nrc_simple]
[  661.096936]  nrc_hspi_probe+0x50/0xc4 [nrc_simple]
[  661.101734]  spi_probe+0x88/0xf0
[  661.104970]  really_probe.part.0+0x9c/0x310
[  661.109159]  __driver_probe_device+0x98/0x150
[  661.113520]  driver_probe_device+0xc8/0x160
[  661.117707]  __driver_attach+0xfc/0x190
[  661.121547]  bus_for_each_dev+0x70/0xd0
[  661.125386]  driver_attach+0x28/0x40
[  661.128965]  bus_add_driver+0x108/0x1f0
[  661.132805]  driver_register+0x7c/0x130
[  661.136645]  __spi_register_driver+0xcc/0xe0
[  661.140920]  nrc_init+0xd8/0x1000 [nrc_simple]
[  661.145373]  do_one_initcall+0x50/0x1b0
[  661.149214]  do_init_module+0x44/0x240
[  661.152968]  load_module+0x1f70/0x2680
[  661.156721]  __do_sys_finit_module+0xb8/0x100
[  661.161081]  __arm64_sys_finit_module+0x24/0x30
[  661.165615]  invoke_syscall+0x48/0x120
[  661.169368]  el0_svc_common.constprop.0+0xcc/0xf0
[  661.174075]  do_el0_svc+0x28/0x90
[  661.177393]  el0_svc+0x20/0x60
[  661.180451]  el0t_64_sync_handler+0x1a8/0x1b0
[  661.184811]  el0t_64_sync+0x1a0/0x1a4
[  661.188476] ---[ end trace 8eec052da0c4e7a5 ]---
[  661.193344] [Error] failed to read register(0x0).
[  661.198075] nrc-hspi-simple: probe of spi1.0 failed with error -5
[  661.204344] done successfully.
$ cat /sys/class/spi_master/spi1/spi1.0/modalias 
spi:nrc-hspi-simple
$ ls /dev/s*
/dev/stderr  /dev/stdin   /dev/stdout
```

synaption, 2022-08-03:
looks like some more work on the dts fixed it.

closed 2022-08-03 by synaption

mountain33r, 2023-11-14:
> looks like some more work on the dts fixed it.

Hi there! Any chance you could share your dts file? I've been struggling to get the 7292 to work on the coral board and I think the dts/dtbo is the issue.