newracom/nrc7292_sw_pkg #49: start.py 1 0 US not working with onboard WiFi & BT enabled

Ankush-Pathak, 2022-09-13:
I am trying to start the NRC7292 radio in AP mode with onboard WiFi enabled, but the `ifconfig wlan0 up` command in `start.py` script fails with 
```
SIOCSIFFLAGS: Operation not permitted
```
`dmesg`

```
[   45.189114] v3d fec00000.v3d: MMU error from client CLE (4) at 0x9241000, pte invalid
[   45.292364] nrc: loading out-of-tree module taints kernel.
[   45.313051] nrc80211 nrc80211: [spi_suspend] drv_state:-1 wowlan_enabled:0 power_save:0
[   45.313068] start FW 332
[   45.746537] end FW
[   48.795723] nrc80211 nrc80211: size of bd file is 2128
[   48.795779] nrc80211 nrc80211: Major 01 Minor 04 Total len 0840 Num_Data_Groups 0009 Checksum 38AB
[   48.795787] nrc80211 nrc80211: target version is not matched(2 : 0)
```

Last few lines of `/boot/config.txt` (see `dtoverlay=disable-spidev-overlay`)
```
# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1
dtoverlay=disable-spidev-overlay
[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
enable_uart=1
```

Rename `wlan0` to `wlan2`
```
$ cat /etc/udev/rules.d/70-persistent-net.rules
SUBSYSTEM=="net", ACTION=="add", DRIVERS=="?*", ATTR{address}=="e4:5f:01:1e:4c:a6", ATTR{dev_id}=="0x0", ATTR{type}=="1", NAME="wlan2"
```
`/etc/wpa_supplicant`
```
$ ls -l /etc/wpa_supplicant/
total 52
-rwxr-xr-x 1 root root   937 Feb 25  2021 action_wpa.sh
-rw-r--r-- 1 root root 25569 Feb 25  2021 functions.sh
-rwxr-xr-x 1 root root  4696 Feb 25  2021 ifupdown.sh
-rw-r--r-- 1 root root    83 Apr 11 19:11 unsued_wpa_supplicant.conf
-rw-r--r-- 1 root root   159 Sep 12 18:15 unsued_wpa_supplicant-wlan0.conf
-rw-r--r-- 1 root root   159 Sep 12 19:32 wpa_supplicant-wlan2.conf
```

`raspi-blacklist.conf` is empty.

Please let me know if you any further information.

newracom, 2022-09-16:
Dear @Ankush-Pathak,

Please check this user guide first: https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/UG-7292-018-Raspberry_Pi_setup.pdf

Best regards,
Newracom

Ankush-Pathak, 2022-09-16:
Dear @newracom,
Thanks for your response. I have gone through and setup my devices according to the guide numerous times. Based on my testing so far, even with onboard WiFi and BT disabled, I get the same error.
I enabled debug logging and based on that looks like the error I get from the `start.py` script is a symptom of the issue described [here](https://github.com/newracom/nrc7292_sw_pkg/issues/47). I see the same messages mentioned in the linked issue in my `dmesg` output.

Thanks,
Ankush

newracom, 2022-09-16:
Dear @Ankush-Pathak,

What kind of EVK are you using?

Best regards,
Newracom

newracom, 2022-09-16:
Dear @Ankush-Pathak,

If you're not using this, https://newracom.com/products/nrc7292-evk, 
I'm afraid I need more detailed information of your board.
Could you provide me the link or detailed description?
And please let me know your kernel version of RPi4.

Best regards,
Newracom

Ankush-Pathak, 2022-09-16:
@newracom 

I will have to look for a link, but meanwhile any information I can get for you from the actual board to help you identify it?
```
$ uname -a
Linux raspberrypi 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux
```

Ankush-Pathak, 2022-09-20:
@newracom [This ](https://www.cnx-software.com/2022/04/03/add-wifi-halow-to-raspberry-pi-with-alfa-network-ahpi7292s-hat/)is the device I have.

newracom, 2022-09-21:
Dear @Ankush-Pathak,

Thank you for your information.
Unfortunately, NRC7292 EVK doesn't support kernel 5.15.61 at the moment.
So, I have no choice but to ask you to contact Alfa Network.

Best regards,
Newracom

Ankush-Pathak, 2022-09-21:
@newracom What is the latest kernel version that you support?

newracom, 2022-09-22:
Dear @Ankush-Pathak,

Officially it is version 5.10.63.

Best regards,
Newracom

closed 2022-09-22 by Ankush-Pathak