newracom/nrc7292_sw_pkg #50: AMD Architecture Driver Support

teo-tsou, 2022-09-14:
Hello!
This is definitely not an issue, but I would like to know if I can connect the AHMB7292S with AMD devices like intel UP Board. I am totally new to this newracom's driver but I know that is ARM based so the possible answer is no. But is there any chance to have AMD compatibility?
I would really appreciate your help thank you.

newracom, 2022-09-17:
Dear @teo-tsou,

From NRC7292 EVK's perspective, we have no chance to verify its compatibility with AMD host devices.
So, we can't tell about the AMD compatibility of AHMB7292S.
Please contact the AHMB7292S vendor.

Best regards,
Newracom

newracom, 2022-09-22:
Dear @teo-tsou,

Thank you for your question.
Unfortunately, there's no plan to support x86 architecture host device for now.

Best regards,
Newracom

closed 2022-10-06 by newracom