newracom/nrc7292_sw_pkg #52: Unknown symbol in module when insmod nrc.ko

tomcattw, 2022-10-02:
I follow the whole procedure, then run 
 ./start.py 1 0 EU
in Loading module step, error occured as following 

sudo insmod /home/pi/nrc_pkg/sw/driver/nrc.ko  hifspeed=20000000 spi_bus_num=0 spi_cs_num=0 spi_gpio_irq=5 spi_polling_interval=0 fw_name=uni_s1g.bin bss_max_idle=180 ndp_preq=1 auto_ba=1 listen_interval=1000
insmod: ERROR: could not insert module /home/pi/nrc_pkg/sw/driver/nrc.ko: Unknown symbol in module
I install module manually, the same erro ... 
[image]

How to solve " Unknown symbol in module"
Thanks.

enganguish, 2022-10-07:
How did you create nrc.ko? Was it built on you RPi?

If the kernel version of your RPi is not 4.14.70, you should build it again based on your environment.
the nrc.ko in nrc7292_sw_pkg has been created for 4.14.70 by default.

tumist, 2022-11-13:
Make sure the kernel module mac80211 is loaded

newracom, 2023-02-01:
I'd like to close this issue. Please open a new one if there's any further issue.

closed 2023-02-01 by newracom