newracom/nrc7292_sw_pkg #60: Happen SIOCSIFFLAGS: Operation not permitted

mingtsann, 2022-11-18:
Hi Sirs,
Please help ! Thank you so much
We use: 
AHPI7292S REV 0C 2021-01-22 + Raspberry PI 4 
OS: Linux raspberrypi 5.10.63-v7l+
Have Done: setup follow UG-7292-018-Raspberry_Pi_setup.pdf 
kernel headers be installed manually using the kernel source
make nrc.ko OK and copy this one to target
sudo ./start.py 1 0 TW
Happen:
sudo insmod /home/pi/nrc_pkg/sw/driver/nrc.ko  hifspeed=20000000 spi_bus_num=0 spi_cs_num=0 spi_gpio_irq=5 spi_polling_interval=0 fw_name=uni_s1g.bin bss_max_idle=180 ndp_preq=1 auto_ba=1 listen_interval=1000
SIOCSIFFLAGS: Operation not permitted

dmesg:
[  619.743905] start FW 333
[  620.184203] end FW
[  623.208490] nrc80211 nrc80211: size of bd file is 2128
[  623.208540] nrc80211 nrc80211: Major 01 Minor 04 Total len 0840 Num_Data_Groups 0009 Checksum 38AB
[  623.208550] nrc80211 nrc80211: target version is not matched(1 : 0)
pi@raspberrypi:~/nrc_pkg/script $ 

************************************************************************************
pi@raspberrypi:~/nrc_pkg/script $ sudo ./start.py 1 0 TW
Done.
Done.
------------------------------
Model            : 7292
STA Type         : AP
Country          : TW
Security Mode    : OPEN
CAL. USE         : ON
AMPDU            : ON
Download FW      : uni_s1g.bin
Max TX Power     : 24
BSS MAX IDLE     : 180
------------------------------
NRC AP setting for HaLow...
[*] Set Max CPU Clock on RPi
1400000
1400000
1400000
1400000
Done
[0] Clear
sudo: iptables: command not found
sudo: iptables: command not found
[1] Copy and Set Module Parameters
total 672
drwxr-xr-x 2 pi   pi     4096 11月  9 13:29 .
drwxr-xr-x 4 pi   pi     4096 11月  7 16:53 ..
-rwxrwxrwx 1 pi   pi      281 11月  7 16:53 copy
-rwxr-xr-x 1 pi   pi     2128 11月  7 16:53 nrc7292_bd.dat
-rwxr-xr-x 1 pi   pi   335356 11月  7 16:53 nrc7292_cspi.bin
-rwxr-xr-x 1 root root 335356 11月 18 09:37 uni_s1g.bin
-rwxr-xr-x 1 root root 335356 11月 18 09:37 /lib/firmware/uni_s1g.bin
====================================
 AP INTERFACE     : wlan0
 AP STATIC IP     : 192.168.200.1 
 NET MASK NUM     : 24 
====================================
Config for AP is done!
IP and DHCP config done
[2] Set Initial Country
[3] Loading module
sudo insmod /home/pi/nrc_pkg/sw/driver/nrc.ko  hifspeed=20000000 spi_bus_num=0 spi_cs_num=0 spi_gpio_irq=5 spi_polling_interval=0 fw_name=uni_s1g.bin bss_max_idle=180 ndp_preq=1 auto_ba=1 listen_interval=1000
SIOCSIFFLAGS: Operation not permitted
pi@raspberrypi:~/nrc_pkg/script $ uname -a
Linux raspberrypi 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux
pi@raspberrypi:~/nrc_pkg/script $
**********************************************************************
[... 423 lines cut ...]
[    8.324968] vc4-drm gpu: [drm] fb0: vc4drmfb frame buffer device
[    9.721225] systemd-journald[130]: File /var/log/journal/6863a4a704a449d0b3397f55bec3e7ce/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[   10.480750] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   10.481619] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   10.687283] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   10.688725] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   10.692112] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   11.756031] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   11.756060] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   12.237056] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   12.237983] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   12.252788] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   12.253137] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   12.253444] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   13.199596] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.eth0.retrans_time - use net.ipv6.neigh.eth0.retrans_time_ms instead
[   32.142583] v3d fec00000.v3d: MMU error from client L2T (0) at 0xba1000, pte invalid
[  619.712808] nrc: loading out-of-tree module taints kernel.
[  619.743896] nrc80211 nrc80211: [spi_suspend] drv_state:-1 wowlan_enabled:0 power_save:0

[  619.743905] start FW 333
[  620.184203] end FW
[  623.208490] nrc80211 nrc80211: size of bd file is 2128
[  623.208540] nrc80211 nrc80211: Major 01 Minor 04 Total len 0840 Num_Data_Groups 0009 Checksum 38AB
[  623.208550] nrc80211 nrc80211: target version is not matched(1 : 0)
pi@raspberrypi:~/nrc_pkg/script $

mingtsann, 2022-12-12:
Skip the bugs

closed 2022-12-12 by mingtsann

silos96, 2023-01-31:
hi, i have the same problem.
Can you tell to me how you fix it?

many thanks

silos96, 2023-01-31:
i'm using the last  revision 1.3.4 rev 14

and raspberry pi4 

thanks

mingtsann, 2023-01-31:
It is newracom's driver problem. Suggest just skip it and waits for
newracom fix it.

Have a nice day !

Ed

silos96, 2023-01-31:
i'm sorry, can you tell me some more information about the fix? 

what i have to skip? 

i know that the problem is related to a nrc.ko driver. but what i have to do in order to solve it?

newracom, 2023-01-31:
Dear @silos96,

Could you send me this? 
**nrc_pkg/sw/firmware/nrc7292_bd.dat**

BTW, which version of SW package are you using?

Best regards,
Newracom

silos96, 2023-02-01:
hi, i'm using a legacy version of raspberrypi OS, 22-09-2022, Buster. 
the kernel is 5.10 version. i follow your guide to install all in raspberry pi 4.

the nrc7292_sw_pkg is the last one 1.3.4 rev 14

i attach the log of start.py:

 ~/nrc_pkg/script/start.py 1 0 EU
Done.
Done.
------------------------------
Model            : 7292
STA Type         : AP
Country          : EU
Security Mode    : OPEN
AMPDU            : ON
Download FW      : uni_s1g.bin
Max TX Power     : 24
BSS MAX IDLE     : 180
------------------------------
NRC AP setting for HaLow...
[*] Set Max CPU Clock on RPi
1400000
1400000
1400000
1400000
Done
[0] Clear
[1] Copy and Set Module Parameters
total 680
drwxr-xr-x 2 pi   pi     4096 Feb  1 09:46 .
drwxr-xr-x 4 pi   pi     4096 Feb  1 09:42 ..
-rwxrwxrwx 1 pi   pi      281 Feb  1 09:42 copy
-rwxr-xr-x 1 pi   pi     2128 Feb  1 09:42 nrc7292_bd.dat
-rwxr-xr-x 1 pi   pi   337820 Feb  1 09:42 nrc7292_cspi.bin
-rwxr-xr-x 1 root root 337820 Feb  1 09:56 uni_s1g.bin
-rwxr-xr-x 1 root root 337820 Feb  1 09:56 /lib/firmware/uni_s1g.bin
====================================
 AP INTERFACE     : wlan0
 AP STATIC IP     : 192.168.200.1
 NET MASK NUM     : 24
====================================
Config for AP is done!
IP and DHCP config done
[2] Set Initial Country
[3] Loading module
sudo insmod /home/pi/nrc_pkg/sw/driver/nrc.ko  hifspeed=20000000 spi_bus_num=0 spi_cs_num=0 spi_gpio_irq=5 spi_polling_interval=0 fw_name=uni_s1g.bin bss_max_idle=180 ndp_preq=1 auto_ba=1 listen_interval=1000
SIOCSIFFLAGS: Operation not permitted

i cannot attach the .dat file here. i attach a link of my onedrive where you can download it.

https://unipvit-my.sharepoint.com/:f:/g/personal/francesco_silino01_universitadipavia_it/Emi-8RsC6XBGiLbpPeyqwrUBQIaM9J4mKAB_oQivCnWTsA?e=XMXDXZ

many thanks

silos96, 2023-02-03:
many thanks, it's work! 
i'm sorry but this stuff is not indicated in the guide. can you tell to me where i have to modify the .c file and which number i have to put? 

best regards
francesco

newracom, 2023-02-03:
Dear @silos96,

Basically, the BD file should be maintained by the EVK manufacturer. And users can't change it.
In this case, Alfa should provide the proper BD file to the customers, however, I supported just once for your convenience.

Best regards,
Newracom

silos96, 2023-02-03:
ah ok, thanks. the problem was that the alfa repository is not updated. the sdk present in their repository is 1.3.4 rev.02, and with this one we have some problems. now that i updated to 1.3.4 rev 14 the problems was gone. 

anyway, many thanks 
Francesco