newracom/nrc7292_sw_pkg #63: overrun, tx size:151508

synaption, 2022-12-05:
Under heavy or moderate use, the mesh point portal is showing the message ```overrun, tx size:151508```  many times.  Sometimes this causes the network to stall.  Sometimes it is ok.  It seems like some devices may be worse than other which could point to a hardware issue.  I am wondering if there is a way to solve this.  Perhaps by increasing the size of whatever is overrunning.

newracom, 2022-12-08:
Dear @synaption,

Thank you for your questions.
The following feature was added on v1.3.4 rev13, https://github.com/newracom/nrc7292_sw_pkg/releases/tag/v1.3.4_rev13

- Suppress overrun log for the host platform with a lower clock speed CPU

Could you try with the latest package and share the result?

Best regards,
Newracom

synaption, 2023-01-11:
So I got around to looking at this again.  Suppressing the log does not really solve the problem.  This log info appears when the network is not functioning.  Currently None of the noes will connect.  The CPU seems to be set at my devices max frequency by default.  
```
$ cat//sys/devices/system/cpu/cpu0/cpufreq/scaling_available_frequencies 
1000000 1500000 
$ cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq 
1500000
$ cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq 
1500000
```