newracom/nrc7292_sw_pkg #66: start.py failing to enable wlan0

Jake-Howell, 2023-02-01:
I'm trying to run the start.py script after following: [UG-7292-018-Raspberry_Pi_setup ](https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/host/doc/UG-7292-018-Raspberry_Pi_setup.pdf)
however, i'm finding that it fails when attempting to enable the wlan0 network interface.
The script stops and prints:
SIOCSIFFLAGS: Operation not permitted.

The only notable differance between what is specified in the setup manual, is that my Pi is running linux kernal 5.15.84-v7+. I have compiled the nrc driver against the newer kernel so insmod doesn't complain (as sudgested in a previous issue).

I've added a few extra print statments to the start script so there is more detail of what's happening. 
The shell output is in the following doc:
[start-py-output.txt](https://github.com/newracom/nrc7292_sw_pkg/files/10561432/nrc_start_script_debug.info.txt)

If more infomation is needed please let me know.
many thanks!

newracom, 2023-02-01:
Dear @Jake-Howell,

Please share following information.
- dmesg log
- HaLow board information

Best regards,
Newracom

Jake-Howell, 2023-02-02:
Hi I've updated my original post with the info you asked for.

Thanks alot :)

newracom, 2023-02-02:
Dear @Jake-Howell ,

There's target version number at the BD file and this should be configured to be aligned with the target board.
I'm afraid AHPI7292S has a different value from NRC7292 EVK. So, I updated that value of BD file.
Please try with this: https://newratek-my.sharepoint.com/:u:/g/personal/jehun_lee_newracom_com/Eas32q7PLc5CjvxLlYVHYcgBOV0ZJ3X_ZQrevveEvxQccQ?e=GI3Uo3 (please replace nrc_pkg/sw/firmware/nrc7292_bd.dat with this)

Best regards,
Newracom

Jake-Howell, 2023-02-11:
Hi, thanks for getting back to me.
Unfortunatly i've just tried this however i still get the same problem.

newracom, 2023-02-13:
Dear @Jake-Howell,

If you saw this, "ERROR: could not insert module nrc.ko: unknown symbol in module", it's because of the discrepancy between the linux-header version and kernel version. Could you share the results of following commands?

- ls /usr/src
- uname -a

Best,
Aaron

Jake-Howell, 2023-02-13:
@newracom sorry, i forgot to remove that from the log. I did run into that however i have already fixed that as I've noticed someone else had the same issue here before.

The error im currently getting is 
SIOCSIFFLAGS: Operation not permitted

after running:
./start.py 1 0 EU

The script always fails when it attempts to enable the wlan0 intrface.

newracom, 2023-02-13:
Dear @Jake-Howell,

It seems that this is the BD version issue. Please try with following BD file after renaming it to nrc7292_bd.dat.
https://newratek-my.sharepoint.com/:u:/g/personal/jehun_lee_newracom_com/EW3fKeUZjQlJtBi3O9D5TFMBZ4-Vnqw8Xm9GOlFabXkZsQ?e=YfacqZ
As I mentioned above, the BD file for the board should be provided by the board manufacturer.
If you fail with this BD file, please contact Alfa for the further assistance.

Best regards,
Newracom

Jake-Howell, 2023-02-13:
Dear @newracom,

That file seems to have done the job, the start script now completes. 
Now to attempt to connect another device to it!

Thanks alot :)

closed 2023-02-13 by newracom

Avamander, 2023-07-14:
For anyone stumbling upon this issue. I saw a similar error with the AHPI7292S HAT from Alfa Networks, replacing the `nrc7292_bd.dat` in `/lib/firmware` worked.

```
nrc80211 spi0.0: target version is not matched(1 : 0)
nrc80211 spi0.0: BD file is invalid! Stop loading FW
Succeed to register spi driver(nrc80211).
nrc80211 spi0.0: @@@@@@@@@ bd(0) is invalid. Stop MAC START @@@@@@@
nrc80211 spi0.0: stop()
nrc_netlink_exit
```