newracom/nrc7292_sw_pkg #72: cannot build driver on Pi CM4 with linux kernel 6.1

teinterence, 2023-05-08:
board: CM4 with 8G ram and 32G emmc
module: AHMC7292SU
`pi@raspberrypi:~/nrc7292_sw_pkg/package/host/src/nrc $ make

make[1]: Entering directory '/usr/src/linux-headers-6.1.21-v8+'
  CC [M]  /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.o
In file included from /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-hif.h:22,
                 from /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:18:
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c: In function ‘nrc_assoc_h_basic’:
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:662:7: error: ‘struct ieee80211_bss_conf’ has no member named ‘aid’
  662 |   info->aid, info->bssid);
      |       ^~
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-debug.h:80:59: note: in definition of macro ‘nrc_mac_dbg’
   80 | #define nrc_mac_dbg(fmt, ...) nrc_dbg(NRC_DBG_MAC, fmt, ##__VA_ARGS__)
      |                                                           ^~~~~~~~~~~
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:664:16: error: ‘struct ieee80211_bss_conf’ has no member named ‘aid’
  664 |  nw->aid = info->aid;
      |                ^~
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:669:51: error: ‘struct ieee80211_bss_conf’ has no member named ‘aid’
  669 |  nrc_wim_skb_add_tlv(skb, WIM_TLV_AID, sizeof(info->aid), &info->aid);
      |                                                   ^~
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:669:64: error: ‘struct ieee80211_bss_conf’ has no member named ‘aid’
  669 |  nrc_wim_skb_add_tlv(skb, WIM_TLV_AID, sizeof(info->aid), &info->aid);
      |                                                                ^~
In file included from ./include/linux/rculist.h:11,
                 from ./include/linux/dcache.h:8,
                 from ./include/linux/fs.h:8,
                 from ./include/linux/highmem.h:5,
                 from ./include/linux/bvec.h:10,
                 from ./include/linux/skbuff.h:17,
                 from ./include/linux/if_ether.h:19,
                 from ./include/linux/etherdevice.h:20,
                 from /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc.h:21,
                 from /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.h:20,
                 from /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:17:
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:674:28: error: ‘struct ieee80211_vif’ has no member named ‘chanctx_conf’
  674 |  conf = rcu_dereference(vif->chanctx_conf);
      |                            ^~
./include/linux/rcupdate.h:436:10: note: in definition of macro ‘__rcu_dereference_check’
  436 |  typeof(*p) *local = (typeof(*p) *__force)READ_ONCE(p); \
      |          ^
./include/linux/rcupdate.h:656:28: note: in expansion of macro ‘rcu_dereference_check’
  656 | #define rcu_dereference(p) rcu_dereference_check(p, 0)
      |                            ^~~~~~~~~~~~~~~~~~~~~
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:674:9: note: in expansion of macro ‘rcu_dereference’
  674 |  conf = rcu_dereference(vif->chanctx_conf);
      |         ^~~~~~~~~~~~~~~
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:674:28: error: ‘struct ieee80211_vif’ has no member named ‘chanctx_conf’
  674 |  conf = rcu_dereference(vif->chanctx_conf);
      |                            ^~
./include/linux/rcupdate.h:436:31: note: in definition of macro ‘__rcu_dereference_check’
  436 |  typeof(*p) *local = (typeof(*p) *__force)READ_ONCE(p); \
      |                               ^
./include/linux/rcupdate.h:656:28: note: in expansion of macro ‘rcu_dereference_check’
  656 | #define rcu_dereference(p) rcu_dereference_check(p, 0)
      |                            ^~~~~~~~~~~~~~~~~~~~~
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:674:9: note: in expansion of macro ‘rcu_dereference’
  674 |  conf = rcu_dereference(vif->chanctx_conf);
      |         ^~~~~~~~~~~~~~~
In file included from <command-line>:
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:674:28: error: ‘struct ieee80211_vif’ has no member named ‘chanctx_conf’
  674 |  conf = rcu_dereference(vif->chanctx_conf);
      |                            ^~
././include/linux/compiler_types.h:337:9: note: in definition of macro ‘__compiletime_assert’
  337 |   if (!(condition))     \
      |         ^~~~~~~~~
././include/linux/compiler_types.h:357:2: note: in expansion of macro ‘_compiletime_assert’
  357 |  _compiletime_assert(condition, msg, __compiletime_assert_, __COUNTER__)
      |  ^~~~~~~~~~~~~~~~~~~
./include/asm-generic/rwonce.h:36:2: note: in expansion of macro ‘compiletime_assert’
[... 284 lines cut ...]
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c: In function ‘nrc_mac_channel_switch_beacon’:
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:2693:6: error: too few arguments to function ‘ieee80211_beacon_get_template’
 2693 |  b = ieee80211_beacon_get_template(hw, vif, NULL);
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc.h:28,
                 from /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.h:20,
                 from /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:17:
./include/net/mac80211.h:5196:1: note: declared here
 5196 | ieee80211_beacon_get_template(struct ieee80211_hw *hw,
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c: At top level:
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:2871:22: error: initialization of ‘void (*)(struct ieee80211_hw *, struct ieee80211_vif *, struct ieee80211_bss_conf *, u64)’ {aka ‘void (*)(struct ieee80211_hw *, struct ieee80211_vif *, struct ieee80211_bss_conf *, long long unsigned int)’} from incompatible pointer type ‘void (*)(struct ieee80211_hw *, struct ieee80211_vif *, struct ieee80211_bss_conf *, u32)’ {aka ‘void (*)(struct ieee80211_hw *, struct ieee80211_vif *, struct ieee80211_bss_conf *, unsigned int)’} [-Werror=incompatible-pointer-types]
 2871 |  .bss_info_changed = nrc_mac_bss_info_changed,
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:2871:22: note: (near initialization for ‘nrc_mac80211_ops.bss_info_changed’)
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:2894:13: error: initialization of ‘int (*)(struct ieee80211_hw *, struct ieee80211_vif *, unsigned int,  u16,  const struct ieee80211_tx_queue_params *)’ {aka ‘int (*)(struct ieee80211_hw *, struct ieee80211_vif *, unsigned int,  short unsigned int,  const struct ieee80211_tx_queue_params *)’} from incompatible pointer type ‘int (*)(struct ieee80211_hw *, struct ieee80211_vif *, u16,  const struct ieee80211_tx_queue_params *)’ {aka ‘int (*)(struct ieee80211_hw *, struct ieee80211_vif *, short unsigned int,  const struct ieee80211_tx_queue_params *)’} [-Werror=incompatible-pointer-types]
 2894 |  .conf_tx = nrc_mac_conf_tx,
      |             ^~~~~~~~~~~~~~~
/home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.c:2894:13: note: (near initialization for ‘nrc_mac80211_ops.conf_tx’)
cc1: all warnings being treated as errors
make[2]: *** [scripts/Makefile.build:250: /home/pi/nrc7292_sw_pkg/package/host/src/nrc/nrc-mac80211.o] Error 1
make[1]: *** [Makefile:2012: /home/pi/nrc7292_sw_pkg/package/host/src/nrc] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.21-v8+'
make: *** [Makefile:62: modules] Error 2
`

bienxanh1901, 2023-06-19:
Got the same issue! Does anyone found a solution?

teledatics, 2023-07-04:
Please see out Pull Request for a solution:

https://github.com/newracom/nrc7292_sw_pkg/pull/67

You can also use our nrc.ko driver fork with these changes at:

https://github.com/teledatics/nrc7292_sw_pkg/tree/teledatics_spi_crash_fix

closed 2024-01-23 by newracom