newracom/nrc7292_sw_pkg #73: The NRC7292 Standalone Firmware Downloader for Windows meets "Pinging 2nd bootloader failed"

edmingChen, 2023-06-08:
* Follow the document "NRC7292Evaluation Kit User Guide (Standalone) Ultra-low power & Long-range Wi-Fi Ver 2.3 Feb. 28, 2023" to do.
* NRC7292 EVB: Raspberry Pi 3B, NRC7292 module+NRC7292 adapter board (V0.4_7292B_TW12; NRM7292 Test Board V0.4 2020.06.17)
* No use Raspberry Pi3 B host for standalone operation.
* Setup S/W build environment is OK (follow the document's say to do) 
* Download SDK: https://github.com/newracom/nrc7292_sdk.git (use the version [NRC7292_Standalone_SDK_v1.4])
* Build sample application program "hello_world" is OK (as attached)
[image]
[image]

*  UART connection between PC and EVB is OK (as attached)
* Change the DIP Switch mode to DOWNLOAD MODE, ‘HHHLLH’ in advance. 
* Launch the NRC7292 Standalone Firmware flash tool and select the correct serial port. (under Window 11 environment)
* Press the ‘SET’ button to launch the file selector. (as attached)
* Push the "Start" button, then respond error message "Pinging 2nd bootloader failed" 
* Question:  the document say "MAC addresses (for WLAN0 and WLAN1) can be read from the flash." , it means MAC addresses will auto read from flash or we should key in the MAC address. If it is the later one, how to get the MAC addresses.
* Please advise what happen it is and how to fix it. Thank you very much

[image]
[image]

edmingChen, 2023-06-08:
Use the flashtool
[image]

corytodd, 2023-11-06:
On my boards which are both v1 (no other version info on silkscreen?), the dips are backwards and upside down in reference to the FlashTool image, PDF docs, and RTD docs. The correct download mode should look like this.

[image]

Use the same rotation for standalone and host modes.

As far I can tell, the MAC addresses are only read after a flash update.

closed 2024-01-23 by newracom