newracom/nrc7292_sw_pkg #75: DKMS support

Avamander, 2023-07-02:
It would be really nice if the official distribution used DKMS. The primary benefit would be that the kernel module is (re)built automatically necessary. But it would also improve integration with for example Secure Boot.

teledatics, 2023-07-04:
We created a DKMS module for the nrc.ko driver. Source code is available at:

https://github.com/teledatics/nrc-dkms

Please see our docs site at https://teledatics.com/docs for a walk-through on how to add our repo and install the module using apt-get. Our DKMS ncr.ko implementation is designed to work with our DKMS spi_ft232h.ko USB-SPI driver and TD-XPAH EVK hardware. It may need to be modified for hardware with a direct SPI connection.

Avamander, 2023-07-04:
Thanks! Will try and use that once my hardware arrives. Are there any other differences one should keep in mind?

teledatics, 2023-07-04:
If you are using a device with an FTDI USB-SPI translation chip, such as our TD-XPAH EVK or the Gateworks mini-PCIe card, then it will 'just work'. If your hardware is directly connected over SPI, you may need to manually configure the SPI bus number and module loading. 

We created a utility to find the SPI bus and load the nrc.ko module automatically named nrc_load_module.sh that is installed in /usr/local/bin. You may need to edit this script in order to load the nrc.ko module with the correct parameters for a directly connected nrc7292 device.