newracom/nrc7292_sw_pkg #77: FT232H configuration

Avamander, 2023-10-02:
I built the ft232 module and enabled `ft232h_usb_spi` in `start.py`, but dmesg doesn't say anything besides `Succeed to register spi driver(nrc80211).` .

I suspected a mismatch between the FT232H wiring expected by the `ft232h-usb-spi` module and the pins labeled for SPI on the [Adafruit FT232H Breakout board](https://www.adafruit.com/product/2264), but I can't figure out where.

On the Adafruit board the pinout is as follows:
```
D0 - SCK/SCL
D1 - MOSI
D2 - MISO
D3 - SPICS 
```

From the FT232 documentation in MPSSE mode these should be equivalent to:
```
ADBUS0 -  TCK/SK
ADBUS1 - TDI/DO
ADBUS2 - TDO/DI
ADBUS3 - TMS/CS
```
From [AN-7292-008-FT232H_USB_SPI.pdf](https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/doc/AN-7292-008-FT232H_USB_SPI.pdf):
```
MPSSE_GPIOL0 - 500
MPSSE_GPIOL1 - 501
MPSSE_GPIOL2 - 502
MPSSE_GPIOL3 - 503 
```

So this seems correct, but after running the following commands, I can't see a new wireless interface or any dmesg output related to the module except `Succeed to register spi driver(nrc80211).`.
```
sudo insmod ./nrc7292_sw_pkg/package/src/ft232h-usb-spi/spi-ft232h.ko latency=1 spi_bus_num=3 gpio_base_num=500
sudo insmod ./nrc7292_sw_pkg/package/src/nrc/nrc.ko fw_name=nrc7292_cspi.bin bd_name=nrc7292_bd.dat listen_interval=1000 debug_level_all=1 spi_bus_num=3 spi_polling_interval=5 spi_cs_num=503
```

newracom, 2023-10-10:
Dear @Avamander,

Thank you for your inquiry.

Could you try with following insmod command?

sudo insmod ./nrc7292_sw_pkg/package/src/nrc/nrc.ko fw_name=nrc7292_cspi.bin bd_name=nrc7292_bd.dat listen_interval=1000 debug_level_all=1 **hifspeed=15000000** spi_bus_num=3 **spi_cs_num=0 spi_gpio_irq=503** spi_polling_interval=50

As evident from this command, hifspeed should be configured as 15MHz for FT232H. 
Additionally, spi_cs_num should be set to 0, and spi_gpio_irq should be either 500 or 503.

Best regards,
Newracom

Avamander, 2023-10-10:
@newracom 

> Could you try with following insmod command?

I tried it but that provided no difference in dmesg, still just:

```
Succeed to register spi driver(nrc80211).
```

I also attached a logic analyzer to D0, D1, D2, D3. Even though D0 (CLK) and D1 (MOSI) are logic level low, I see no actual activity.

> Additionally, spi_cs_num should be set to 0, 

Just to be sure, `spi_cs_num=0` corresponds to which pin on the FT232?

> and spi_gpio_irq should be either 500 or 503.

 I did not connect IRQ because polling was supported and it was unclear which pin on the FT232 breakout was going to be utilized. Though wouldn't 500 be ADBUS0/SCK/CLK and 503 ADBUS3/SPICS/CS?

Avamander, 2023-10-12:
@newracom 
It seems like the `nrc` module will not initialize itself if there isn't an overlay for it. There's also no way to use the non-DT initialization on newer kernels. I also can't figure out how would one even write an overlay to use the FT232H module.

What's the recommended approach on new kernels on regular PCs? (That don't utilize device trees(?))

newracom, 2023-10-12:
Dear @Avamander,

Please refer to following table.
[image]
Please update the spi_gpio_irq parameter value to match your GPIO pin for HSPI_EIRQ. 
- GPIOL0 : 500
- GPIOL1 : 501
- GPIOL2 : 502
- GPIOL3 : 503
- If no using HSPI_EIRQ: -1

If there are any additional logs after "Succeed to register spi driver(nrc80211)," kindly inform us.

Best regards,
Newracom

Avamander, 2023-10-13:
@newracom 
> Please update the spi_gpio_irq parameter value to match your GPIO pin for HSPI_EIRQ.

Thank you for this image, it seems to have no effect. In general the module parameters seem to have no effect at all. The driver gets registered but nothing besides that.

> If there are any additional logs after "Succeed to register spi driver(nrc80211)," kindly inform us.

If I apply the overlay that this repository provides then the module actually tries to initialize the radio (but using the incorrect SPI bus 0, even if defined otherwise, it seems).

Avamander, 2023-10-29:
Yeah, no success in getting the kernel module to actually do anything with the FT232H SPI bus, if there's an overlay for using a physical SPI bus the module actually loads on the same machine. Trying to specify a bus using module arguments does not work with an overlay in effect.

I'd really like to get it working in order to work on native wpa_supplicant and hostapd support.

Avamander, 2024-03-30:
I tried to revisit this using the latest driver, but the best I can see is still `usbcore: registered new interface driver spi_ft232h` and `Succeed to register spi driver(nrc80211)`. But this practically means that the driver is not actually loaded (no `nrc`-specific log lines). When I replug the FT232 module I also see ` spi_ft232h 1-1:1.0: gpiochip: label=ftdi-mpsse-gpio.0 base=500 ngpio=12`, so at least that part seems to interface fine.

Does the driver actually work without an overlay?

Avamander, 2024-04-06:
@newracom It seems that on newer Linux kernels the module is always built with `CONFIG_SPI_USE_DT` which makes it not behave correctly without a warning.

I've resorted to force-disabling the flag and reimplementing `spi_busnum_to_master`. Then the module loads and seems to basically work. Ideally these changes are also made for your driver.