newracom/nrc7292_sw_pkg #80: [Jetson Nano / Kernel 4.9.337]: wiphy_register+0x3e0/0x778 [cfg80211]

wizath, 2023-11-01:
Tried to use new v1.5 driver, but received kernel panic during loading new driver, removed ENABLE_DYNAMIC_PS that helped to load the driver properly when using v1.3.x, but no success. Any ideas?

```
[  436.270957] spi spi0.0: SPI Device Created (bus_num:0, cs_num:0, irq_num:246, max_speed:16000000
[  436.271042] tegra-gpio 6000d000.gpio: Failed to set pinctrl input direction of GPIO 246: -517
[  436.280103] nrc80211 spi0.0: probe()
[  436.330425] nrc80211 spi0.0: SPI probing. chip_id:7292 modem_id:00000001 status:1
[  436.330431] nrc80211 spi0.0: Newracom IEEE802.11 C-SPI: chipid=7292, sw_id=1020716, board_id=DEB09757
[  436.330434] nrc80211 spi0.0: Boot loader
[  436.331146] nrc80211 spi0.0: credit[ 0] :  4
[  436.331150] nrc80211 spi0.0: credit[ 1] : 80
[  436.331153] nrc80211 spi0.0: credit[ 2] :  8
[  436.331156] nrc80211 spi0.0: credit[ 3] :  8
[  436.331159] nrc80211 spi0.0: credit[ 4] :  0
[  436.331162] nrc80211 spi0.0: credit[ 5] : 40
[  436.331165] nrc80211 spi0.0: credit[ 6] :  4
[  436.331168] nrc80211 spi0.0: credit[ 7] : 80
[  436.331171] nrc80211 spi0.0: credit[ 8] :  8
[  436.331174] nrc80211 spi0.0: credit[ 9] :  8
[  436.331177] nrc80211 spi0.0: credit[10] :  0
[  436.331180] nrc80211 spi0.0: credit[11] :  0
[  436.337962] nrc80211 spi0.0: CQM is enabled
[  436.337968] nrc80211 spi0.0: Configuration of H/W Dependent Setting : 7292
[  436.337971] nrc80211 spi0.0: - HW_QUEUES: 6
[  436.337974] nrc80211 spi0.0: - WoWLAN Pattern num: 1
[  436.337977] nrc80211 spi0.0: [spi_check_fw++]
[  436.338100] nrc80211 spi0.0: [spi_check_fw, 1525] sys.sw_id = 0x1020716
[  436.338103] nrc80211 spi0.0: [spi_check_fw--]
[  436.341336] nrc80211 spi0.0: [nrc_check_fw_file, 205] Checking firmware... (nrc7292_cspi.bin)
[  436.341341] nrc80211 spi0.0: [nrc_check_fw_file, 214] OK...(000000009f70ced8, 0x53840)
[  436.341344] nrc80211 spi0.0: FW download....nrc7292_cspi.bin
[  436.341347] nrc80211 spi0.0: [nrc_download_fw] priv->
[  436.341350] nrc80211 spi0.0: - fw->data:000000009f70ced8
[  436.341352] nrc80211 spi0.0: - fw->size:342080
[  436.341355] nrc80211 spi0.0: - fw_data_pos:000000009f70ced8
[  436.341357] nrc80211 spi0.0: - remain_bytes:342080
[  436.341360] start FW 340
[  437.049330] end FW
[  437.051887] nrc80211 spi0.0: [spi_check_ready++]
[  437.052178] nrc80211 spi0.0: [spi_check_ready] chip_id:7292 build_chip_id:0102 target_sw_id:1814 host_sw_id:1  status:1
[  437.152199] nrc80211 spi0.0: [spi_check_ready++]
[  437.152369] nrc80211 spi0.0: [spi_check_ready] chip_id:7292 build_chip_id:0102 target_sw_id:1814 host_sw_id:1  status:1
[  437.252383] nrc80211 spi0.0: [spi_check_ready++]
[  437.252508] nrc80211 spi0.0: [spi_check_ready] chip_id:7292 build_chip_id:0102 target_sw_id:1814 host_sw_id:1  status:1
[  437.352521] nrc80211 spi0.0: [spi_check_ready++]
[  437.352888] nrc80211 spi0.0: [spi_check_ready] chip_id:7292 build_chip_id:7292 target_sw_id:1 host_sw_id:1  status:1
[  437.352892] nrc80211 spi0.0: start()
[  437.366485] nrc80211 spi0.0: system ready
[  437.366488] nrc80211 spi0.0: ready
[  437.366492] nrc80211 spi0.0:   -- type: 19
[  437.366494] nrc80211 spi0.0:   -- length: 67
[  437.366497] nrc80211 spi0.0:   -- version: 0x00000002
[  437.366500] nrc80211 spi0.0:   -- tx_head_size: 12
[  437.366502] nrc80211 spi0.0:   -- rx_head_size: 36
[  437.366505] nrc80211 spi0.0:   -- payload_align: 0
[  437.366508] nrc80211 spi0.0:   -- hw_version: 65535
[  437.366510] nrc80211 spi0.0:   -- cap_mask: 0x39
[  437.366513] nrc80211 spi0.0:   -- cap_li: 1000, 1000
[  437.366516] nrc80211 spi0.0:   -- cap_idle: 1800, 0
[  437.366520] nrc80211 spi0.0:   -- mac_addr[0]: 84:25:3f:9c:09:c4
[  437.366522] nrc80211 spi0.0:   -- mac_addr[1]: 86:25:3f:9c:09:c4
[  437.366525] nrc80211 spi0.0:   -- cap_mask[0]: 0x2
[  437.366528] nrc80211 spi0.0:   -- cap_mask[1]: 0x2
[  437.366609] ------------[ cut here ]------------
[  437.371274] WARNING: CPU: 3 PID: 7847 at /dvs/git/dirty/git-master_linux/kernel/kernel-4.9/net/wireless/core.c:579 wiphy_register+0x3e0/0x778 [cfg80211]
[  437.384892] Modules linked in: nrc(O+) mac80211 cfg80211 fuse xt_conntrack ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_conntrack_netlink nfnetlink xt_addrtype iptable_filter iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack br_netfilter zram userspace_alert overlay binfmt_misc nvgpu ip_tables x_tables

[  437.384941] CPU: 3 PID: 7847 Comm: insmod Tainted: G        W  O    4.9.337-tegra #1
[  437.384943] Hardware name: NVIDIA Jetson Nano 2GB Developer Kit (DT)
[  437.384946] task: 0000000058d38af5 task.stack: 0000000090de0ab5
[  437.384995] PC is at wiphy_register+0x3e0/0x778 [cfg80211]
[  437.385068] LR is at ieee80211_register_hw+0x3e8/0xa90 [mac80211]
[  437.385071] pc : [<ffffff80011c0970>] lr : [<ffffff800127ae78>] pstate: 00400045
[  437.385073] sp : ffffffc0f40eb8c0
[  437.385075] x29: ffffffc0f40eb8c0 x28: 0000000000000018 
[  437.385080] x27: 0000000000000035 x26: 0000000000000000 
[  437.385085] x25: ffffffc0aaee11e0 x24: ffffffc0aaee15b8 
[  437.385090] x23: 0000000000000001 x22: 000000000000002f 
[  437.385095] x21: 0000000000000003 x20: 00000000100a1891 
[  437.385100] x19: ffffffc0aaee02a0 x18: 0000000000000020 
[  437.385104] x17: 0000000000000000 x16: 0000000000000000 
[  437.385109] x15: ffffff800134f960 x14: 0000000c0000000e 
[  437.385114] x13: 0000000000001162 x12: ffffff800134f990 
[  437.385118] x11: 0000000800000027 x10: 0000000000000050 
[  437.385123] x9 : 0000000000000002 x8 : 0000000000000002 
[  437.385128] x7 : 000000000000075e x6 : 000000000000008c 
[  437.385132] x5 : ffffff8001346c84 x4 : 0000000000000002 
[  437.385137] x3 : ffffff8001346c84 x2 : ffffff8001346808 
[  437.385141] x1 : 000000000000008c x0 : ffffff80013467f0 

[  437.385148] ---[ end trace fe1b96fb439afab9 ]---
[  437.389752] Call trace:
[  437.389803] [<00000000f0bca983>] wiphy_register+0x3e0/0x778 [cfg80211]
[  437.389871] [<00000000882fc6a5>] ieee80211_register_hw+0x3e8/0xa90 [mac80211]
[  437.389923] [<00000000137bf930>] nrc_register_hw+0x3b4/0x500 [nrc]
[  437.389975] [<0000000072af85a1>] nrc_nw_start+0x14c/0x2b0 [nrc]
[  437.390030] [<000000002a961d02>] nrc_cspi_probe+0x210/0x2e8 [nrc]
[  437.390039] [<000000008aa426b6>] spi_drv_probe+0x80/0xd0
[  437.390045] [<0000000040ad9c00>] driver_probe_device+0x298/0x448
[  437.390050] [<00000000b1ad9eae>] __driver_attach+0x110/0x138
[  437.390054] [<00000000c8649431>] bus_for_each_dev+0x5c/0xa8
[  437.390058] [<00000000dc3ded88>] driver_attach+0x30/0x40
[  437.390061] [<000000002d40348a>] bus_add_driver+0x20c/0x2a8
[  437.390065] [<000000007eb08dc4>] driver_register+0x6c/0x110
[  437.390068] [<00000000cf2b26af>] __spi_register_driver+0x6c/0x78
[  437.390110] [<00000000a64c23b8>] nrc_cspi_init+0xe8/0x190 [nrc]
[  437.390114] [<00000000a53bf0ed>] do_one_initcall+0x44/0x130
[  437.390119] [<00000000a1cf8ff8>] do_init_module+0x64/0x1b0
[  437.390123] [<000000001a5fa72e>] load_module+0x10e4/0x1318
[  437.390126] [<000000007f4c33c1>] SyS_finit_module+0xd8/0xf0
[  437.390129] [<00000000830fef3b>] el0_svc_naked+0x34/0x38
[  437.390178] nrc80211 spi0.0: ieee80211_register_hw failed (-22)
[  437.390197] nrc80211 spi0.0: Failed to nrc_register_hw
[  437.395350] nrc_netlink_exit
[  437.421949] nrc80211 spi0.0: Failed to nrc_nw_start (-22)
[  437.427601] nrc80211: probe of spi0.0 failed with error -22
```

wizath, 2023-11-06:
It turns out that `DCONFIG_MAC80211_MESH` caused this error. Disabling it solves it

pietertolsma, 2023-12-14:
@wizath how did you load the device trees for the Jetson? What device tree did you use?

wizath, 2024-02-20:
Sorry for late reply, here is pinmux configration for header SPI
```
exp-header-pinmux {
        phandle = <0x122>;
        linux,phandle = <0x122>;
        
        hdr40-pin26 {
                nvidia,enable-input = <0x1>;
                nvidia,tristate = <0x0>;
                nvidia,pull = <0x2>;
                nvidia,function = "spi1";
                nvidia,pins = "spi1_cs1_pc4";
        };
        
        hdr40-pin24 {
                nvidia,enable-input = <0x1>;
                nvidia,tristate = <0x0>;
                nvidia,pull = <0x2>;
                nvidia,function = "spi1";
                nvidia,pins = "spi1_cs0_pc3";
        };
        
        hdr40-pin23 {
                nvidia,enable-input = <0x1>;
                nvidia,tristate = <0x0>;
                nvidia,pull = <0x1>;
                nvidia,function = "spi1";
                nvidia,pins = "spi1_sck_pc2";
        };
        
        hdr40-pin21 {
                nvidia,enable-input = <0x1>;
                nvidia,tristate = <0x0>;
                nvidia,pull = <0x1>;
                nvidia,function = "spi1";
                nvidia,pins = "spi1_miso_pc1";
        };
        
        hdr40-pin19 {
                nvidia,enable-input = <0x1>;
                nvidia,tristate = <0x0>;
                nvidia,pull = <0x1>;
                nvidia,function = "spi1";
                nvidia,pins = "spi1_mosi_pc0";
        };
};`

puffie, 2024-02-23:
CONFIG_MAC80211_MESH should be defined while buildl kernel to have mesh support enabled.
It doesn't look like CONFIG_MAC80211_MESH is not defined while building 4.9 kernel.
It is correct that you will have to remove "-DCONFIG_MAC80211_MESH".

Since the nrc driver building Makefile already includes kernel config by setting below line.
"include $(KDIR)/.config"

-DCONFIG_MAC80211_MESH should be removed in nrc Makefile.