newracom/nrc7292_sw_pkg #83: Driver breaks on apt update of RPI CM4 6.1.21-v7l+

remcolouter, 2024-01-23:
I am currently testing with "AHPI7292S" from Alfa Networks. I have used their prebuilt image and it works as long as there is no apt update/upgrade, only with this particular image I could make it work.  With the latest apt upgrade it breaks the driver, since it switches to 64-bit. In the /boot/config.txt, I have added "arm_64bit=0" and it boots again the 32-bit version. I am now at 6.1.21-v7l+ but unfortunately it does not work anymore. I have also tried to rebuild the driver within the new kernel, there were no error messages, but unfortunately it never worked again. When I apply the start.py script, I get:  

insmod: ERROR: could not insert module /home/pi/nrc_pkg/sw/driver/nrc.ko: Unknown symbol in module
wlan0: ERROR while getting interface flags: No such device
rmmod: ERROR: Module nrc is not currently loaded

I tried to build the driver on the 64-bit kernel as well, it did the build successfully, but alas the same error message.

Is there anything I can do to make it work with the latest kernel?

remcolouter, 2024-01-23:
Added dmesg output:

[   12.422668] nrc: loading out-of-tree module taints kernel.
[   12.425566] nrc: Unknown symbol ieee80211_stop_tx_ba_session (err -2)
[   12.425670] nrc: Unknown symbol ieee80211_start_tx_ba_cb_irqsafe (err -2)
[   12.425738] nrc: Unknown symbol ieee80211_connection_loss (err -2)
[   12.425833] nrc: Unknown symbol __ieee80211_schedule_txq (err -2)
[   12.425899] nrc: Unknown symbol ieee80211_iterate_interfaces (err -2)
[   12.425979] nrc: Unknown symbol rate_control_set_rates (err -2)
[   12.426048] nrc: Unknown symbol ieee80211_chswitch_done (err -2)
[   12.426223] nrc: Unknown symbol ieee80211_free_hw (err -2)
[   12.426288] nrc: Unknown symbol ieee80211_txq_schedule_start (err -2)
[   12.426361] nrc: Unknown symbol ieee80211_iterate_stations_atomic (err -2)
[   12.426460] nrc: Unknown symbol ieee80211_start_tx_ba_session (err -2)
[   12.426530] nrc: Unknown symbol ieee80211_beacon_loss (err -2)
[   12.426602] nrc: Unknown symbol ieee80211_nullfunc_get (err -2)
[   12.426703] nrc: Unknown symbol ieee80211_register_hw (err -2)
[   12.426796] nrc: Unknown symbol ieee80211_restart_hw (err -2)
[   12.426961] nrc: Unknown symbol ieee80211_find_sta (err -2)
[   12.427055] nrc: Unknown symbol ieee80211_tx_status_irqsafe (err -2)
[   12.427166] nrc: Unknown symbol ieee80211_stop_tx_ba_cb_irqsafe (err -2)
[   12.427191] nrc: Unknown symbol wiphy_to_ieee80211_hw (err -2)
[   12.427225] nrc: Unknown symbol ieee80211_queue_delayed_work (err -2)
[   12.427268] nrc: Unknown symbol ieee80211_next_txq (err -2)
[   12.427293] nrc: Unknown symbol ieee80211_remain_on_channel_expired (err -2)
[   12.427319] nrc: Unknown symbol ieee80211_wake_queues (err -2)
[   12.427336] nrc: Unknown symbol ieee80211_tx_prepare_skb (err -2)
[   12.427348] nrc: Unknown symbol ieee80211_beacon_get_template (err -2)
[   12.427371] nrc: Unknown symbol ieee80211_free_txskb (err -2)
[   12.427388] nrc: Unknown symbol ieee80211_alloc_hw_nm (err -2)
[   12.427406] nrc: Unknown symbol ieee80211_tx_dequeue (err -2)
[   12.427460] nrc: Unknown symbol ieee80211_ready_on_channel (err -2)
[   12.427476] nrc: Unknown symbol wdev_to_ieee80211_vif (err -2)
[   12.427490] nrc: Unknown symbol ieee80211_stop_queues (err -2)
[   12.427521] nrc: Unknown symbol ieee80211_vif_to_wdev (err -2)
[   12.427538] nrc: Unknown symbol ieee80211_scan_completed (err -2)
[   12.427581] nrc: Unknown symbol ieee80211_unregister_hw (err -2)
[   12.427729] nrc: Unknown symbol ieee80211_csa_finish (err -2)
[   12.427741] nrc: Unknown symbol ieee80211_rx_irqsafe (err -2)
[   23.434708] bcmgenet fd580000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   23.434740] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   63.166209] nrc: Unknown symbol ieee80211_stop_tx_ba_session (err -2)
[   63.166251] nrc: Unknown symbol ieee80211_start_tx_ba_cb_irqsafe (err -2)
[   63.166267] nrc: Unknown symbol ieee80211_connection_loss (err -2)
[   63.166297] nrc: Unknown symbol __ieee80211_schedule_txq (err -2)
[   63.166312] nrc: Unknown symbol ieee80211_iterate_interfaces (err -2)
[   63.166335] nrc: Unknown symbol rate_control_set_rates (err -2)
[   63.166349] nrc: Unknown symbol ieee80211_chswitch_done (err -2)
[   63.166397] nrc: Unknown symbol ieee80211_free_hw (err -2)
[   63.166410] nrc: Unknown symbol ieee80211_txq_schedule_start (err -2)
[   63.166425] nrc: Unknown symbol ieee80211_iterate_stations_atomic (err -2)
[   63.166448] nrc: Unknown symbol ieee80211_start_tx_ba_session (err -2)
[   63.166462] nrc: Unknown symbol ieee80211_beacon_loss (err -2)
[   63.166477] nrc: Unknown symbol ieee80211_nullfunc_get (err -2)
[   63.166499] nrc: Unknown symbol ieee80211_register_hw (err -2)
[   63.166520] nrc: Unknown symbol ieee80211_restart_hw (err -2)
[   63.166553] nrc: Unknown symbol ieee80211_find_sta (err -2)
[   63.166572] nrc: Unknown symbol ieee80211_tx_status_irqsafe (err -2)
[   63.166626] nrc: Unknown symbol ieee80211_stop_tx_ba_cb_irqsafe (err -2)
[   63.166653] nrc: Unknown symbol wiphy_to_ieee80211_hw (err -2)
[   63.166690] nrc: Unknown symbol ieee80211_queue_delayed_work (err -2)
[   63.166737] nrc: Unknown symbol ieee80211_next_txq (err -2)
[   63.166763] nrc: Unknown symbol ieee80211_remain_on_channel_expired (err -2)
[   63.166791] nrc: Unknown symbol ieee80211_wake_queues (err -2)
[   63.166809] nrc: Unknown symbol ieee80211_tx_prepare_skb (err -2)
[   63.166821] nrc: Unknown symbol ieee80211_beacon_get_template (err -2)
[   63.166846] nrc: Unknown symbol ieee80211_free_txskb (err -2)
[   63.166864] nrc: Unknown symbol ieee80211_alloc_hw_nm (err -2)
[   63.166883] nrc: Unknown symbol ieee80211_tx_dequeue (err -2)
[   63.166940] nrc: Unknown symbol ieee80211_ready_on_channel (err -2)
[   63.166957] nrc: Unknown symbol wdev_to_ieee80211_vif (err -2)
[   63.166972] nrc: Unknown symbol ieee80211_stop_queues (err -2)
[   63.167005] nrc: Unknown symbol ieee80211_vif_to_wdev (err -2)
[   63.167023] nrc: Unknown symbol ieee80211_scan_completed (err -2)
[   63.167068] nrc: Unknown symbol ieee80211_unregister_hw (err -2)
[   63.167227] nrc: Unknown symbol ieee80211_csa_finish (err -2)
[   63.167240] nrc: Unknown symbol ieee80211_rx_irqsafe (err -2)

newracom, 2024-01-23:
Dear @remcolouter, 

Please reach out to Alfa Networks for assistance. 
However, the issue may stem from the improper use of an incorrect DTS file. 
Please check yours by refering to section 3.4 of https://github.com/newracom/nrc7292_sw_pkg/blob/master/package/doc/UG-7292-018-Raspberry_Pi_setup.pdf and https://github.com/newracom/nrc7292_sw_pkg/blob/master/dts/newracom_for_5.16_or_later.dts.

Regards,
Newracom

remcolouter, 2024-01-24:
Thanks for the reply. Will do. 

I did see the item about creating this overlay for kernels 5.16+ and did it like this. Unfortunately that did not work either, however I have tried many things, so possibly I did some things in the wrong order. If I have a solution, I will revert to this issue. 

Kind regards,

Remco

jr1221, 2024-02-12:
Just use the install instructions from this repo on a fresh debian bullseye release, but change out the board data file with the one linked in previous issues by newracom (it is easy to find).