newracom/nrc7292_sw_pkg #91: Kernel module does not compile on 6.6

timothyb89, 2024-04-09:
Attempting to build the kernel module fails on a fresh install of Raspberry Pi OS (bookworm) running the 6.6.20 kernel on a Pi 4:
```
pi@piradio5:~/nrc7292_sw_pkg/package/src/nrc $ uname -a
Linux piradio5 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux
pi@piradio5:~/nrc7292_sw_pkg/package/src/nrc $ make

make[1]: Entering directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
  CC [M]  /home/pi/nrc7292_sw_pkg/package/src/nrc/nrc-netlink.o
/home/pi/nrc7292_sw_pkg/package/src/nrc/nrc-netlink.c:129:27: error: initialization of ‘int (*)(const struct genl_split_ops *, struct sk_buff *, struct genl_info *)’ from incompatible pointer type ‘int (*)(const struct genl_ops *, struct sk_buff *, struct genl_info *)’ [-Werror=incompatible-pointer-types]
  129 |         .pre_doit       = nrc_nl_pre_doit,
      |                           ^~~~~~~~~~~~~~~
/home/pi/nrc7292_sw_pkg/package/src/nrc/nrc-netlink.c:129:27: note: (near initialization for ‘nrc_nl_fam.pre_doit’)
/home/pi/nrc7292_sw_pkg/package/src/nrc/nrc-netlink.c:130:27: error: initialization of ‘void (*)(const struct genl_split_ops *, struct sk_buff *, struct genl_info *)’ from incompatible pointer type ‘void (*)(const struct genl_ops *, struct sk_buff *, struct genl_info *)’ [-Werror=incompatible-pointer-types]
  130 |         .post_doit      = nrc_nl_post_doit,
      |                           ^~~~~~~~~~~~~~~~
/home/pi/nrc7292_sw_pkg/package/src/nrc/nrc-netlink.c:130:27: note: (near initialization for ‘nrc_nl_fam.post_doit’)
cc1: all warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/scripts/Makefile.build:248: /home/pi/nrc7292_sw_pkg/package/src/nrc/nrc-netlink.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:1938: /home/pi/nrc7292_sw_pkg/package/src/nrc] Error 2
make[1]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:246: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
make: *** [Makefile:50: modules] Error 2
```

EugeneN, 2024-04-09:
It is a warning being treated as an error (see `cc1: all warnings being treated as errors` in the compiler output). 

This warning can be disabled by adding `-Wno-incompatible-pointer-types` to the compiler flags. After doing that the kernel module seems to compile and work ok. Generally it is not a good idea to disable warnings, and the right way would be to fix the pointer type in the code, but it would enable you to proceed right away with your project until the code will have been fixed and updated on Github.

You can disable the warning by editing `~/nrc7292_sw_pkg/package/src/nrc/Makefile` file and adding the mentioned compiler flag to the `ccflags-y` variable on the line 42, so that it would look like this:

```
ccflags-y += -Wall -Werror -Wno-incompatible-pointer-types
```

Hope this helps until the code is updated.

serious-01, 2024-04-09:
Compiled without issue, wouldn't install.  Fresh install of raspbian bookwork 64 bit on a raspberry pi 5

insmod: ERROR: could not insert module /home/pi/nrc_pkg/sw/driver/nrc.ko: Unknown symbol in module

dmesg:
[  949.775693] nrc: Unknown symbol ieee80211_stop_tx_ba_session (err -2)
[  949.775715] nrc: Unknown symbol ieee80211_start_tx_ba_cb_irqsafe (err -2)
[  949.775723] nrc: Unknown symbol ieee80211_connection_loss (err -2)
[  949.775741] nrc: Unknown symbol __ieee80211_schedule_txq (err -2)
[  949.775748] nrc: Unknown symbol ieee80211_iterate_interfaces (err -2)
[  949.775761] nrc: Unknown symbol rate_control_set_rates (err -2)
[  949.775768] nrc: Unknown symbol ieee80211_chswitch_done (err -2)
[  949.775788] nrc: Unknown symbol ieee80211_free_hw (err -2)
[  949.775795] nrc: Unknown symbol ieee80211_txq_schedule_start (err -2)
[  949.775802] nrc: Unknown symbol ieee80211_iterate_stations_atomic (err -2)
[  949.775813] nrc: Unknown symbol ieee80211_start_tx_ba_session (err -2)
[  949.775822] nrc: Unknown symbol ieee80211_beacon_loss (err -2)
[  949.775829] nrc: Unknown symbol ieee80211_nullfunc_get (err -2)
[  949.775840] nrc: Unknown symbol ieee80211_register_hw (err -2)
[  949.775850] nrc: Unknown symbol ieee80211_restart_hw (err -2)
[  949.775867] nrc: Unknown symbol ieee80211_find_sta (err -2)
[  949.775876] nrc: Unknown symbol ieee80211_tx_status_irqsafe (err -2)
[  949.775910] nrc: Unknown symbol ieee80211_stop_tx_ba_cb_irqsafe (err -2)
[  949.775924] nrc: Unknown symbol wiphy_to_ieee80211_hw (err -2)
[  949.775939] nrc: Unknown symbol ieee80211_queue_delayed_work (err -2)
[  949.775959] nrc: Unknown symbol ieee80211_next_txq (err -2)
[  949.775973] nrc: Unknown symbol ieee80211_remain_on_channel_expired (err -2)
[  949.775985] nrc: Unknown symbol ieee80211_wake_queues (err -2)
[  949.775993] nrc: Unknown symbol ieee80211_tx_prepare_skb (err -2)
[  949.775999] nrc: Unknown symbol ieee80211_beacon_get_template (err -2)
[  949.776007] nrc: Unknown symbol ieee80211_free_txskb (err -2)
[  949.776015] nrc: Unknown symbol ieee80211_alloc_hw_nm (err -2)
[  949.776024] nrc: Unknown symbol ieee80211_tx_dequeue (err -2)
[  949.776049] nrc: Unknown symbol ieee80211_ready_on_channel (err -2)
[  949.776057] nrc: Unknown symbol wdev_to_ieee80211_vif (err -2)
[  949.776063] nrc: Unknown symbol ieee80211_stop_queues (err -2)
[  949.776079] nrc: Unknown symbol ieee80211_vif_to_wdev (err -2)
[  949.776087] nrc: Unknown symbol ieee80211_scan_completed (err -2)
[  949.776106] nrc: Unknown symbol ieee80211_unregister_hw (err -2)
[  949.776169] nrc: Unknown symbol ieee80211_csa_finish (err -2)
[  949.776175] nrc: Unknown symbol ieee80211_rx_irqsafe (err `-2)

mclendon99, 2024-04-14:
What I did is to change the declarations in nrc-netlink.c to use the "correct" type for the argument as follows:

#ifdef CONFIG_SUPPORT_NEW_NETLINK
static int nrc_nl_pre_doit(const struct genl_split_ops *ops,
         struct sk_buff *skb, struct genl_info *info)
#else

Both functions simply return 0, so it's a fairly low risk change.

k3rn3Lp4n1cK, 2024-05-25:
I compiled on 6.6 and works with @mclendon99 fix.
insmod is failing
`sudo insmod /home/pi/nrc_pkg/sw/driver/nrc.ko fw_name=uni_s1g.bin bss_max_idle=1800 ndp_preq=1 listen_interval=1000 bd_name=nrc7292_bd.dat`
`insmod: ERROR:  could not insert module /home/pi/nrc_pkg/sw/driver/nrc.ko: Invalid module format`

mclendon99, 2024-05-25:
> I compiled on 6.6 and works with @mclendon99 fix. Although: insmod is failing
> 
> ```
> sudo insmod /home/pi/nrc_pkg/sw/driver/nrc.ko fw_name=uni_s1g.bin bss_max_idle=1800 ndp_preq=1 listen_interval=1000 bd_name=nrc7292_bd.dat
> insmod: ERROR:  could not insert module /home/pi/nrc_pkg/sw/driver/nrc.ko: Invalid module format
> ```

You're not using the right kernel headers. Once you upgrade to 6.6, you must reboot, recompile nrc.ko with the correct kernel headers in /lib/modules/$(uname -r), and then insmod it. I'm on 6.6.28 now with no issues.
(STA-3)~>uname -r
6.6.28+rpt-rpi-v8

(STA-3)~>lsmod
Module                  Size  Used by
sch_codel              16384  1
nrc                   200704  0
mac80211             1040384  1 nrc
libarc4                12288  1 mac80211
pps_ldisc              12288  2
cmac                   12288  3
algif_hash             12288  1
aes_arm64              12288  3
aes_generic            32768  1 aes_arm64
algif_skcipher         12288  1
af_alg                 28672  6 algif_hash,algif_skcipher
bnep                   24576  2
vc4                   372736  4
brcmfmac_wcc           12288  0
hci_uart               49152  0
btbcm                  24576  1 hci_uart
bluetooth             606208  26 hci_uart,btbcm,bnep
brcmfmac              348160  1 brcmfmac_wcc
binfmt_misc            16384  1
brcmutil               24576  1 brcmfmac
v3d                    90112  0
snd_soc_hdmi_codec     20480  2
cfg80211              995328  3 nrc,brcmfmac,mac80211
gpu_sched              53248  1 v3d
drm_display_helper     16384  1 vc4
cec                    53248  1 vc4
**** snip snip ***
(STA-3)~>cli_app
===================================================
    Newracom Command Line Application (2.21.5)
===================================================

NRC> show config
---------------------------------------------------
[MAC Configuration]
Device Mode                      : STA
MAC Address                      : 84:25:3f:eb:41:04
Country                          : US
Bandwidth                        : 2M
Frequency                        : 9090 (14)
MAC80211_freq                    : 5765 (153)
Rate Control                     : ON
 -MCS                            : 6
 -bw                             : 2 Mhz (NRC Auto)
Guard Interval                   : AUTO
Security                         : OFF
RTS                              : OFF
RTS threshold                    : 0
Format                           : S1G
Preamble type                    : S1G_SHORT
Promiscuous Mode                 : OFF
color                            : 0x0
Auto CFO Cal                     : ON
BSSID                            : 84:25:3f:ab:17:8b
AID                              : 2

[PHY Configuration]
TX_Gain                          : AUTO
Base RX_Gain                     : 82
Compensated RX_Gain              : 82
Tx Power Type                    : ~30 [AUTO]
---------------------------------------------------
OK

k3rn3Lp4n1cK, 2024-05-25:
Thanks, I just realized apt updated kernel to 6.6.31-v8+, but there are no kernel-headers in /usr/src.
Want to use ubuntu 22.04, but Raspi 5 doesn't officially support it because of the change in the physical header on the board.
I am using raspi 5 with raspbian 32bit OS

Thanks tho, I'll keep working it.