newracom/nrc7292_sw_pkg #100: SIOCSIFFLAGS: Operation not permitted

tugce-bbasoglu, 2024-07-18:
I'm working with NRC7292 AND RASPBEERY 3.I got such an error in the Host Mode document, how can I fix it?

pi@raspberrypi:~ $ ~/nrc_pkg/script/start.py 1 0 US
Done.
Done.
------------------------------
Model            : 7292
STA Type         : AP
Country          : US
Security Mode    : OPEN
BD Name          : nrc7292_bd.dat
AMPDU            : AUTO
Download FW      : uni_s1g.bin
MAX TX Power     : 24 dBm
BSS MAX IDLE     : 1800
------------------------------
NRC AP setting for HaLow...
[*] Set Max CPU Clock on RPi
1400000
1400000
1400000
1400000
Done
[0] Clear
[1] Copy and Set Module Parameters
total 696
drwxr-xr-x 2 pi   pi     4096 Jul 18 13:07 .
drwxr-xr-x 4 pi   pi     4096 Jul 18 12:38 ..
-rwxrwxrwx 1 pi   pi      271 Jul 18 12:38 copy
-rwxr-xr-x 1 pi   pi     2104 Jul 18 12:38 nrc7292_bd.dat
-rwxr-xr-x 1 pi   pi   344920 Jul 18 12:38 nrc7292_cspi.bin
-rwxr-xr-x 1 root root 344920 Jul 18 13:37 uni_s1g.bin
-rwxr-xr-x 1 root root 344920 Jul 18 13:37 /lib/firmware/uni_s1g.bin
====================================
 AP INTERFACE     : wlan0
 AP STATIC IP     : 192.168.200.1
 NET MASK NUM     : 24
====================================
Config for AP is done!
IP and DHCP config done
[2] Set Initial Country
[3] Loading module
sudo insmod /home/pi/nrc_pkg/sw/driver/nrc.ko  fw_name=uni_s1g.bin bss_max_idle=1800 ndp_preq=1 listen_interval=1000 bd_name=nrc7292_bd.dat
SIOCSIFFLAGS: Operation not permitted

vinc3yz, 2024-11-05:
check that theres no conflict on SPI bus and interrupt pin, and check that your disable spidev overlay for newracom is set up correctly. The driver should load without this error

frobinette, 2024-11-29:
Hi,

I have the same error with a PI4 Compute module with AlfaNetwork newracom7292 HAT using bookworm. When I use the image provided by Alfanetwork based on bullseye, it works. My goals is to get it working with the latest version for better support.

dmesg shows:
```
[ 1471.429846] nrc80211 spi0.0: SPI probing. chip_id:7292 modem_id:00000001 status:1
[ 1471.430119] nrc80211 spi0.0: Configuration of H/W Dependent Setting : 7292
[ 1471.430127] nrc80211 spi0.0: - HW_QUEUES: 6
[ 1471.430132] nrc80211 spi0.0: - WoWLAN Pattern num: 1
[ 1471.430643] start FW 343
[ 1471.865177] end FW
[ 1471.865245] nrc80211 spi0.0: [spi_check_ready] chip_id:7292 build_chip_id:0102 target_sw_id:1814 host_sw_id:1  status:1
[ 1471.965285] nrc80211 spi0.0: [spi_check_ready] chip_id:7292 build_chip_id:0102 target_sw_id:1814 host_sw_id:1  status:1
[ 1472.065320] nrc80211 spi0.0: [spi_check_ready] chip_id:7292 build_chip_id:0102 target_sw_id:1814 host_sw_id:1  status:1
[ 1472.165376] nrc80211 spi0.0: [spi_check_ready] chip_id:7292 build_chip_id:7292 target_sw_id:1 host_sw_id:1  status:1
[ 1472.178469] nrc80211 spi0.0: target version is not matched(1 : 0)
[ 1472.179359] Succeed to register spi driver(nrc80211).
[ 1485.722296] [Error] the target device cannot respond while deep sleep.
``` 

iwconfig shows:
```
lo        no wireless extensions.

eth0      no wireless extensions.

wlan0     IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=0 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
``` 

cli_app shows:
```
./cli_app show config
Error while receiving reply from kernel: NACK Received
FAIL   
``` 

but sudo ifconfig wlan0 up give the SIOCSIFFLAGS error.

Any help would be appriciated since I'm running out of ideas.

Thanks,

vinc3yz, 2024-11-29:
Hello @frobinette check my issue : 
#102 

Or go to the issue #60  

You need to download the Board Data File from issue 60, this is a file that Alfa Networks is supposed to provide but they haven't ! Change your board data file and it should work without issue

frobinette, 2024-11-29:
Hi @vinc3yz! it worked!!

Thanks a lot!