newracom/nrc7292_sw_pkg #105: Pi 5 STA doesn't get IP and gets disconnected

vinc3yz, 2025-01-16:
Hello i have been able to almost adapt to RPI 5 on 6.6.51 kernel version. 

So driver loads on both system because it was compiled for this version : 
AP : I feel like it's not setting up the gateway properly but wlan0 gets instanciated. 

STA : block at waiting for IP keeps disconnecting and reconnecting to halow_demo ssid

[image]

[image]

[image]

I get authentiacted assion with an id generated on AP, but the STA doesn't get IPv4 adressed, or IPv6 and IPv4 gateway fixed 

dhcpd has been modified to udhcpd on Bookworm, can it be the issue ? 

[image]

I get Ignore event : ignore connect when using userspace SME, is it beause i'm on 64bit and i absolutely need a pi with 32 bit userspace ? Thank you very much

HookdomPonix, 2025-02-08:
Semi-offtopic but I'm trying to get this working on the CM5 system.  You wouldnt be able to share you dts file you made would you?

As for your issue - you need to load the wireless cfg80211 modules with the country code specified.   It is available as a kernel module for the most part.  The CC error will then disappear and you can then set the channels properly.  
`
modprobe cfg80211 ieee80211_regdom=US`

HookdomPonix, 2025-02-08:
I am now in this position as well.  I got past the situation with the dtbo.  

i’m using the CM5 and am getting the same buffer misalignment issue for spi. 

[image]

vinc3yz, 2025-02-10:
Hello @HookdomPonix  thank you for your answer, i will make a few more test in the week to try to make it working, but i'm getting crazy with this problem. I rebuilt the driver with 1.5.2 and still same problem, but not the same errors in the log except for the Ignore Connect event with SME.  Same errors on both with 2 Pi 5 and  2 Pi 0 2W on 6.6.y 

I've been trying to build on Pi 0 2W and Pi 5 and now i'm trying with an older kernel to reproduce Alfa Network image with a 5.15 Linux Kernel (in case a behavior has changed between 5.1 and after 5.1). 

I can share my Pi 0 2W and Pi 5 DTS if that's what you want ?  I will check about cfg80211 but i thought  i was loaded on boot for this one. As in the official Alfa Image they don't use modprobe to load this.

HookdomPonix, 2025-02-10:
I wouldn't mind seeing the Pi5 DTS - that said, I got around the issue by changing the &spi to &spi0 and deleting the status and such from the second spi0 entry.   

I am now stuck at the same place you are.  As far as I can tell, what is happening is EVERYTHING works until you start sending TCP/IP traffic on the Pi5/CM5 platform.  I believe what's happening is the driver switches its regular SPI bus functionality out for the  DMAC IP stack.  When it does so the DMAC IP stack doesn't work right, probably due to differences in usage of the RP1 chip on the Pi5.  

I have tried many different SPI clock speeds and a complete second unit (WisGate Connect mainboard) to see if this is addressable.  Teledatics is also looking at it at this time.  

Also, are you using the stock Pi image?   If so, you can set the regulatory code with raspi-config.   Kernel 6.x handles the regulatory code differently if you aren't using raspbian and this option must be set in /etc/modprobe.d/cft80211_regdomain.conf

`options cfg80211 ieee80211_regdom=US`

HookdomPonix, 2025-02-10:
More data in this. Added the spi alias back to the bcm2712 dtb so the newracom dts would load again without mods. 

It did not help.

HookdomPonix, 2025-02-11:
Thats very promising - I was concerned that some of the issues with the RP1 chip and SPI might be conflicting with the driver.   This seems unlikely to be a long term show-stopper as the MM chipset works.  

I was also able to get around the dts issue and able to use the factory file by adding the alias for spi (no number added) back to the CM5 modules dtb main device tree.  But it just points to spi0, which means changing spi to spi0 works just as well.  

Here is what I believe so far - everything works for me (and you) until TCP/IP traffic is sent using the DMAC IP stack.  DMAC reports that the buffer is misaligned and then we start failing to transfer the queue.  

I have teledatics looking at this too.  

So to summarize, regular comms with the board for hostapd or wpa_supplicant work fine.   We associate the station with the other system in ah mode, and everything seems to work.   Then we explode in the nrc_hif_work thread as we try to use DMA to move the actual traffic over the network link.

vinc3yz, 2025-02-11:
Do you get IP assigned ? Like do you have an IP assigned if both AP and STA mode when connecting to the AP ? 
Because i got this error, but the failing to connect because SME Userspace error makes the STA WiFi module disconnect from the AP and try to reconnect, but IP is never assigned in my case. Were you able to get an IP ? And then only traffic isn't working ? 

How did you built your kernel to recompile the driver ?

HookdomPonix, 2025-02-11:
I’m using the stock kernel for the Pi5.  

I use bridging and mesh mode - so i don’t DHCP an IP so i’m don’t believe i would get one. if I tried. 

What i believe is happening is ieee80211* traffic is all working right, but the instant you try using any TCP/IP traffic the driver enqueues the traffic stream with the Linux Kernels interface to the Analog Devices DMA (dma02) device.   This is when it all explodes.  

Setting different SPi bus speeds appears to have no impact.

vinc3yz, 2025-02-11:
Okay thank you for your informations. It gives me help, so i assume you're also on 6.6 kernel. I will post my updates soon

jr1221, 2025-05-20:
Any luck?  Ill be tackling this soon with a Pi 5 and kernel 6.12, and will let yall know if I get it working.

jr1221, 2025-05-24:
WIth pi 5 as the AP I avoided the DMA issue with this patch (remove the DMA check).  I have not done any regression testing but seems to work.

```
diff --git a/drivers/dma/dw-axi-dmac/dw-axi-dmac-platform.c b/drivers/dma/dw-axi-dmac/dw-axi-dmac-platform.c
index acd0833b7bd4..e31527bda399 100644
--- a/drivers/dma/dw-axi-dmac/dw-axi-dmac-platform.c
+++ b/drivers/dma/dw-axi-dmac/dw-axi-dmac-platform.c
@@ -702,11 +702,6 @@ static int dw_axi_dma_set_hw_desc(struct axi_dma_chan *chan,
 
 	mem_width = __ffs(data_width | mem_addr | len);
 
-	if (!IS_ALIGNED(mem_addr, 4)) {
-		dev_err(chan->chip->dev, "invalid buffer alignment\n");
-		return -EINVAL;
-	}
-
 	/* Use a reasonable upper limit otherwise residue reporting granularity grows large */
 	mem_burst_msize = axi_dma_encode_msize(16);
 
```

jr1221, 2025-05-24:
May require latest raspberrypi/kernel 6.6.y ( I used bba53a117a4a5c29da892962332ff1605990e17a)